micro-ROS/micro_ros_stm32cubemx_utils #1: HardFault on rclc_support_init

Eembedder, 2021-03-22:
Hi. I try to repeat the steps and on your **sample_main** application got an **HardFault** when call **_rclc_support_init(&support, 0, NULL, &allocator);_** Also I added **-specs=nosys.specs** to Makefile, because it doesn't want to compile a static library. The setup is pretty simple stm32f407vgt6 micro with usart1 initialized.

pablogs9, 2021-03-23:
Could you share you `.ioc` so we can replicate you scenario?

Eembedder, 2021-03-23:
#MicroXplorer Configuration settings - do not modify
FREERTOS.IPParameters=Tasks01
FREERTOS.Tasks01=defaultTask,24,128,StartDefaultTask,Default,NULL,Dynamic,NULL,NULL
File.Version=6
KeepUserPlacement=false
Mcu.Family=STM32F4
Mcu.IP0=FREERTOS
Mcu.IP1=NVIC
Mcu.IP2=RCC
Mcu.IP3=SYS
Mcu.IP4=USART1
Mcu.IPNb=5
Mcu.Name=STM32F407V(E-G)Tx
Mcu.Package=LQFP100
Mcu.Pin0=PA9
Mcu.Pin1=PA10
Mcu.Pin2=VP_FREERTOS_VS_CMSIS_V2
Mcu.Pin3=VP_SYS_VS_Systick
Mcu.PinsNb=4
Mcu.ThirdPartyNb=0
Mcu.UserConstants=
Mcu.UserName=STM32F407VGTx
MxCube.Version=6.1.1
MxDb.Version=DB.6.0.10
NVIC.BusFault_IRQn=true\:0\:0\:false\:false\:true\:false\:false\:false
NVIC.DebugMonitor_IRQn=true\:0\:0\:false\:false\:true\:false\:false\:false
NVIC.ForceEnableDMAVector=true
NVIC.HardFault_IRQn=true\:0\:0\:false\:false\:true\:false\:false\:false
NVIC.MemoryManagement_IRQn=true\:0\:0\:false\:false\:true\:false\:false\:false
NVIC.NonMaskableInt_IRQn=true\:0\:0\:false\:false\:true\:false\:false\:false
NVIC.PendSV_IRQn=true\:15\:0\:false\:false\:false\:true\:false\:false
NVIC.PriorityGroup=NVIC_PRIORITYGROUP_4
NVIC.SVCall_IRQn=true\:0\:0\:false\:false\:false\:false\:false\:false
NVIC.SavedPendsvIrqHandlerGenerated=true
NVIC.SavedSvcallIrqHandlerGenerated=true
NVIC.SavedSystickIrqHandlerGenerated=true
NVIC.SysTick_IRQn=true\:15\:0\:false\:false\:true\:true\:false\:true
NVIC.UsageFault_IRQn=true\:0\:0\:false\:false\:true\:false\:false\:false
PA10.Mode=Asynchronous
PA10.Signal=USART1_RX
PA9.Mode=Asynchronous
PA9.Signal=USART1_TX
PinOutPanel.RotationAngle=0
ProjectManager.AskForMigrate=true
ProjectManager.BackupPrevious=false
ProjectManager.CompilerOptimize=6
ProjectManager.ComputerToolchain=false
ProjectManager.CoupleFile=false
ProjectManager.CustomerFirmwarePackage=
ProjectManager.DefaultFWLocation=true
ProjectManager.DeletePrevious=true
ProjectManager.DeviceId=STM32F407VGTx
ProjectManager.FirmwarePackage=STM32Cube FW_F4 V1.25.2
ProjectManager.FreePins=false
ProjectManager.HalAssertFull=false
ProjectManager.HeapSize=0x800
ProjectManager.KeepUserCode=true
ProjectManager.LastFirmware=true
ProjectManager.LibraryCopy=1
ProjectManager.MainLocation=Core/Src
ProjectManager.NoMain=false
ProjectManager.PreviousToolchain=
ProjectManager.ProjectBuild=false
ProjectManager.ProjectFileName=rlx.ioc
ProjectManager.ProjectName=rlx
ProjectManager.RegisterCallBack=
ProjectManager.StackSize=0x800
ProjectManager.TargetToolchain=STM32CubeIDE
ProjectManager.ToolChainLocation=
ProjectManager.UnderRoot=true
ProjectManager.functionlistsort=1-MX_GPIO_Init-GPIO-false-HAL-true,2-SystemClock_Config-RCC-false-HAL-false,3-MX_USART1_UART_Init-USART1-false-HAL-true
RCC.AHBFreq_Value=16000000
RCC.APB1Freq_Value=16000000
RCC.APB2Freq_Value=16000000
RCC.CortexFreq_Value=16000000
RCC.FamilyName=M
RCC.HSE_VALUE=25000000
RCC.HSI_VALUE=16000000
RCC.I2SClocksFreq_Value=96000000
RCC.IPParameters=AHBFreq_Value,APB1Freq_Value,APB2Freq_Value,CortexFreq_Value,FamilyName,HSE_VALUE,HSI_VALUE,I2SClocksFreq_Value,LSE_VALUE,LSI_VALUE,PLLCLKFreq_Value,PLLQCLKFreq_Value,RTCFreq_Value,RTCHSEDivFreq_Value,SYSCLKFreq_VALUE,VCOI2SOutputFreq_Value,VCOInputFreq_Value,VCOOutputFreq_Value,VcooutputI2S
RCC.LSE_VALUE=32768
RCC.LSI_VALUE=32000
RCC.PLLCLKFreq_Value=96000000
RCC.PLLQCLKFreq_Value=48000000
RCC.RTCFreq_Value=32000
RCC.RTCHSEDivFreq_Value=12500000
RCC.SYSCLKFreq_VALUE=16000000
RCC.VCOI2SOutputFreq_Value=192000000
RCC.VCOInputFreq_Value=1000000
RCC.VCOOutputFreq_Value=192000000
RCC.VcooutputI2S=96000000
USART1.IPParameters=VirtualMode
USART1.VirtualMode=VM_ASYNC
VP_FREERTOS_VS_CMSIS_V2.Mode=CMSIS_V2
VP_FREERTOS_VS_CMSIS_V2.Signal=FREERTOS_VS_CMSIS_V2
VP_SYS_VS_Systick.Mode=SysTick
VP_SYS_VS_Systick.Signal=SYS_VS_Systick
board=custom
isbadioc=false

pablogs9, 2021-03-23:
Try to increase the `defaultTask` size to 2500 Words:

[image]

Let me know if this solves your issue.

pablogs9, 2021-03-23:
Regarding the `-specs=nosys.specs`, I have been able to build it without adding it.

Eembedder, 2021-03-23:
I already increased it 3000 and still the same HardFault. Regarding the -specs=nosys.specs, now it is ok to build without it.

ffffffe1:   Failed to execute MI command:
          -data-disassemble -s 4294967265 -e 4294967421 -- 3
          Error message from debugger back end:
          Cannot access memory at address 0xfffffffe

Eembedder, 2021-03-23:
I put all sources here: https://github.com/Eembedder/rlx

pablogs9, 2021-03-23:
I have updated transports and instructions, check 6c04106ece3baf17157c91138ae764485fb2f034.

Read the new README because I have explained how to configure the serial transports DMA and priorities and FreeRTOS minimum stack size.

Let me know if this works for you.

pablogs9, 2021-03-23:
BTW, I have been testing it with a Nucleo F446ZE

Eembedder, 2021-03-23:
@pablogs9 Thanks for help! Now everything is working fine.

pablogs9, 2021-03-23:
That's cool!

Could you share with us some details of your implementation? What are you using micro-ROS for?

**Important:** Let us know every performance/error/bug that you find in micro-ROS, we are always looking for improvements.

closed 2021-03-23 by pablogs9

Eembedder, 2021-03-23:
This is a robotic platform prototype. STM32 will collect sensor data and send it to the high-performance node for processing and receive commands to control some peripheral.

Eembedder, 2021-03-23:
Also observed that only 43 msg delivered and then it stops sending messages. Try to debug why this happens. It is still with provided example (no extra code).

Eembedder, 2021-03-23:
My bad with DMA configuration. Everything is working well!