micro-ROS/micro_ros_stm32cubemx_utils #3: Ran a test of the

keshavchintamani, 2021-03-26:
Hi Guys, 

Just wanted to report my  experience trying to get these utils working for my STM32F407VETx board. 

My tests are based on c408c275a613709412b9f051aff8b258331cb285 here. 

Before running the docker ros build, I've followed the steps as:

sourced  /opt/ros/foxy/setup.bash
pulled latest on microros_setup, followed intrsructions to 

- rosdep and colcon build microros_setup
- ros2 run micro_ros_setup create_agent_ws.sh
- ros2 run micro_ros_setup build_agent.sh [to build the xrce agent]

sourced the micro_ros_setup_workspace

then I followed these instructions and cd into microros_component and do the docker build. 

But seems like some dependency is missing. 
```
/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c: In function 'rclc_make_node_a_lifecycle_node':
/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:42:3: error: unknown type name 'rcl_lifecycle_state_machine_options_t'; did you mean 'rcl_lifecycle_state_machine_t'?
   42 |   rcl_lifecycle_state_machine_options_t state_machine_options =
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   rcl_lifecycle_state_machine_t
/uros_ws/firmware/mcu_ws/uros/rclc/rclc_lifecycle/src/rclc_lifecycle/rclc_lifecycle.c:43:5: warning: implicit declaration of function 'rcl_lifecycle_get_default_state_machine_options'; did you mean 'rcl_lifecycle_init_default_state_machine'? [-Wimplicit-function-declaration]
   43 |     rcl_lifecycle_get_default_state_machine_options();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     rcl_lifecycle_init_default_state_machine

```
Any ideas?

pablogs9, 2021-03-29:
Hello, @keshavchintamani we have had some issues with this package last week. 
Can you retry the procedure and tell us if https://github.com/ros2/rclc/pull/66 have solved this?
If not, I'll replicate you scenario to check what is happening.

Thanks!

keshavchintamani, 2021-03-29:
@pablogs9 [ros2/rclc#66](https://github.com/ros2/rclc/pull/66) seems to have fixed this. We are able to sucessfully use microROS on the STM32F407VETX. The procedure offered in this repository is correct. We'll try to work on a PR to automate the transition from MakeFile project to STM32CubeIDE. Seems to be a large community of STM32 users how only use the STM32CubeIDE. Thanks again.

pablogs9, 2021-03-30:
Thanks for reporting.

Regarding the IDE we provide some instructions [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils#using-this-package-with-stm32cubeide) but it would be nice if someone contributes a more automated approach.

closed 2021-03-30 by pablogs9

pablogs9, 2021-04-08:
Hello @keshavchintamani check the new instructions added [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/pull/4) for STM32CubeIDE automation.