micro-ROS/micro_ros_stm32cubemx_utils #5: Using the FreeRTOS examples with the IDE

Rob-le-Houe, 2021-04-26:
Hi, I am looking for advice on how I should go about uploading and running the provided FreeRTOS examples through the STM32cubeIDE.

I have successfully finished uploading the example code using a debug session and have then connected a micro-dds agent and used 'ros2 topic echo /cubemx_publisher' to see the integer being incremented and published. I have also used the bash scripts method to upload the code from the ping pong example. However, when I do this the led on the board (Olimex-E407) flashes in the correct timing but the node or topic are not visible.

I made my own .ioc configuration so that I could use the USARTs of my choice for seeing the topics and I know that this works so I would like to continue using this as I like the debug interface. I'm quite new to FreeRTOS so it's not immediately obvious to me where certain parts of the code should go. I can see how the code is structured for the single startDefaultTask() and the similarities to appMain() but it's mainly where I should place the functions called inside of appMain() that I don't know where to place in the IDE style code.

Any help would be greatly appreciated.
Rob

pablogs9, 2021-04-27:
Hello @Rob-le-Houe, could you share your STM32CubeIDE project so we can take a look?

Rob-le-Houe, 2021-04-27:
Hello @pablogs9 thank you for your response. I have attached my STM32cubeIDE  project  folder so you can take a look. I have had to delete the debug folder to keep below the 10Mb limit. 

[Olimex-STM32-E407-FreeRTOS-NO_SD_CARD.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/6382854/Olimex-STM32-E407-FreeRTOS-NO_SD_CARD.zip)

pablogs9, 2021-04-28:
Hello @Rob-le-Houe, I have had time to test you project. I can see that it build correctly, does it works properly in the board?

In general a project generated with CubeIDE will have a `startDefaultTask`, in the CubeIDE we just put the micro-ROS functionality there to offer a basic demo. But for example in the other micro-ROS approaches, the `startDefaultTask` launch another task that runs the `appMain` function.

I'm not sure what was your question here.
Hope that helps.

Rob-le-Houe, 2021-04-28:
Hello @pablogs9 thank you for taking the time for to look into this.

The project works correctly on the board. The information you have given is exactly what I was asking, apologies if it wasn't very clear.

Which part of the code inside of the `startDefaultTask()` launches the `appMain` function?

Thank you! Gracias!
Rob

pablogs9, 2021-04-28:
This one: https://github.com/micro-ROS/freertos_apps/blob/e8e995135b9d337a818c4e03fd29721b3c696fdd/microros_olimex_e407_extensions/Src/main.c#L470

closed 2021-04-28 by Rob-le-Houe