micro-ROS/micro_ros_stm32cubemx_utils #7: Error using this package with STM32CubeIDE

bacurau, 2021-05-22:
Hi, I tried to follow the instructions in "Using this package with STM32CubeIDE" to use the micro-ros with the STM32Cube MX. However I had the following error in the IDE when trying to build the project: "cannot find -lmicroros"

Follow the complete error showed in the console:

arm-none-eabi-gcc -o "micro_ROS_test1.elf" @ "objects.list" -lmicroros -mcpu = cortex-m7 -T "/ media / bacurau / Data / OneDrive - Universidade Estadual de Campinas / Research / SAMSUNG Project / ARM Firmware / micro_ROS_test1 / STM32H753ZITX_FLASH.ld "--specs = nosys.specs -Wl, -Map =" micro_ROS_test1.map "-Wl, - gc-sections -static -Wl, - start-group -lmicroros -Wl, --end-group -Lmicro_ros_stm32cubemx_utils / microros_static_library_ide / libmicroros --specs = nano.specs -mfpu = fpv5-d16 -mfloat-abi = hard -mthumb -Wl, - start-group -lc -lm -Wl, - end -group

/opt/st/stm32cubeide_1.6.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/. ./lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lmicroros

/opt/st/stm32cubeide_1.6.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/. ./lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lmicroros
collect2: error: ld returned 1 exit status

make [1]: *** [makefile: 70: micro_ROS_test1.elf] Error 1
make: *** [makefile: 63: all] Error 2
"make -j12 all" terminated with exit code 2. Build might be incomplete.

I can find those directories in my PC (/opt/st/stm32cubeide_1.6.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/. ./lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld) but, in fact, there is no lmicroros file. I tried to update the STM32Cube IDE, but it has no updates.

Could someone help me solve this problem?

Best Regards,

Rodrigo Bacurau

benmarsan, 2021-05-24:
Do you have the micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a file? If not, building the static library in Docker could've failed.
If that file is there, then something is probably wrong with this part: "-Lmicro_ros_stm32cubemx_utils / microros_static_library_ide / libmicroros". On my project, this part has an absolute path to the directory and not a relative one, so something might not be right with step 4 of the setup.

pablogs9, 2021-05-24:
Yes, first of all check if `micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a` exists.

Enkuushka, 2021-05-24:
`-Lmicro_ros_stm32cubemx_utils / microros_static_library_ide / libmicroros`
instead of above maybe try giving the absolute path to the directory. In fact _-lmicroros_ searches for the file named _libmicroros.a_.
I think the directory containing the _libmicroros.a_ is set faulty.

pablogs9, 2021-05-24:
@Enkuushka could you contribute a PR with this fix in the README.md instructions?

bacurau, 2021-05-24:
Hi,

Thank you very much to all of you! The file libmicroros.a had been generated correctly, the problem was in the path "micro_ros_stm32cubemx_utils / microros_static_library_ide / libmicroros". I replaced this path with the absolute one (I redone step 4 of the setup and it worked). Another thing I had to do, both in step 4 and in step 2, was to put the paths between " ", because my directories contain names with spaces. I think that in the tutorial it would be interesting to include these " ". 

Thank you all. 

Rodrigo Bacurau

closed 2021-05-27 by bacurau