micro-ROS/micro_ros_stm32cubemx_utils #8: Undefined reference to `mkdir` in function `rcutils_mkdir`

Enkuushka, 2021-05-24:
Hello,
I have ran **microros_static_library_ide/library_generation.sh** and built the *libmicroros*. 
It compiled with following logs. But _static library files were created_.
```
Summary: 62 packages finished [1min 43s]
  39 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces composition_interfaces control_msgs diagnostic_msgs example_interfaces geometry_msgs libyaml_vendor lifecycle_msgs micro_ros_msgs microxrcedds_client nav_msgs rcl rcl_action rcl_interfaces rcl_lifecycle rcl_logging_noop rclc rclc_lifecycle rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs tf2_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
ls: cannot access '*.o': No such file or directory
```
Then I configured the build path in stm32CubeIDE. 
And when I build the project it is giving the following error.
```
arm-none-eabi-gcc -o "uROS-test-2.elf" @"objects.list"  -lmicroros -mcpu=cortex-m4 -T"/home/codchin/STM32CubeIDE/workspace_1.6.1/uROS-test-2/STM32F446RETX_FLASH.ld" --specs=nosys.specs -Wl,-Map="uROS-test-2.map" -Wl,--gc-sections -static -Wl,--start-group -lmicroros -Wl,--end-group -L/home/codchin/test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb -Wl,--start-group -lc -lm -Wl,--end-group
/opt/st/stm32cubeide_1.6.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/codchin/test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `unlock_memory':
atomic_64bits.c:(.text+0x98): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.6.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/codchin/test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_load_8':
atomic_64bits.c:(.text+0x114): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.6.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/codchin/test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_store_8':
atomic_64bits.c:(.text+0x190): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.6.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/codchin/test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_exchange_8':
atomic_64bits.c:(.text+0x20c): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.6.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/codchin/test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_fetch_add_8':
atomic_64bits.c:(.text+0x298): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.6.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.linux64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/codchin/test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-filesystem.c.obj): in function `rcutils_mkdir':
filesystem.c:(.text+0x100): undefined reference to `mkdir'
collect2: error: ld returned 1 exit status
make: *** [makefile:71: uROS-test-2.elf] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

16:47:08 Build Failed. 13 errors, 0 warnings. (took 3s.722ms)
```

Thank you for your time and help.

pablogs9, 2021-05-24:
Can you share your STM32CubeIDE project configuration so we can replicate this?

Enkuushka, 2021-05-24:
[uROS-test-2.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/6531653/uROS-test-2.zip)

It is made for F446RE.

pablogs9, 2021-05-25:
I have been able to compile your project along with micro-ROS using [this instructions](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils#using-this-package-with-stm32cubeide). Let me know if you can replicate it.

Here you have my project: 
[uROS-test-2.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/6536819/uROS-test-2.zip)

Enkuushka, 2021-05-25:
Still having same error.
Here I noticed that you added the docker script in the "Pre-build step". But is it necessary to build the _libmicroros.a_ before each build?

pablogs9, 2021-05-26:
The docker will compile the library only when necessary, so yes, it's correct to add it to prebuild steps.
In fact, is the script is prepared to work inside the docker because it automatically retrieves the compiler flags [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/5463b56b850987dbd29bccfc71a22c74e10784a7/microros_static_library_ide/library_generation/library_generation.sh#L13). So please try to compile the library using the provided instructions.

Also, check if you are using `-ffunction-sections -fdata-sections` in your building flags.

Enkuushka, 2021-05-26:
I set the docker to run as a regular user without 'sudo'. I was lazy on setting docker to run as a regular user and just compiled the libs separately. But it was compulsory to use the docker script before the build. I don't know the reason. But I think we should add this to the document too :smile:

Enkuushka, 2021-05-26:
And it worked fine with the docker command.

closed 2021-05-26 by Enkuushka

pablogs9, 2021-05-26:
Could you please open a PR with your proposals for improving this documentation?