micro-ROS/micro_ros_stm32cubemx_utils #13: DDS error on rclc_node_init_default

maximiliankir, 2021-07-14:
Hello,
I don't know if this is the right repository for my issue, but maybe someone here can help me. I'm using the cubemx utils in the CubeIDE to compile Micro-Ros for a custom board with a STM32L4 on it. In the Micro Ros Task I just copied the sample code. Only the Uart port was changed: 
```C
void StartMicroRos(void *argument)
{
	/* USER CODE BEGIN StartMicroRos */
	// micro-ROS configuration
	rmw_uros_set_custom_transport(
	true, (void*) &huart1, cubemx_transport_open, cubemx_transport_close,
			cubemx_transport_write, cubemx_transport_read);

	rcl_allocator_t freeRTOS_allocator =
			rcutils_get_zero_initialized_allocator();
	freeRTOS_allocator.allocate = microros_allocate;
	freeRTOS_allocator.deallocate = microros_deallocate;
	freeRTOS_allocator.reallocate = microros_reallocate;
	freeRTOS_allocator.zero_allocate = microros_zero_allocate;

	if (!rcutils_set_default_allocator(&freeRTOS_allocator))
	{
		printf("Error on default allocators (line %d)\n", __LINE__);
	}

	// micro-ROS app

	rcl_publisher_t publisher;
	std_msgs__msg__Int32 msg;
	rclc_support_t support;
	rcl_allocator_t allocator;
	rcl_node_t node;

	allocator = rcl_get_default_allocator();

	//create init_options
	rcl_ret_t ret = rclc_support_init(&support, 0, NULL, &allocator);

	// create node
	ret = rclc_node_init_default(&node, "cubemx_node", "", &support);

	// create publisher
	ret = rclc_publisher_init_default(&publisher, &node,
			ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
			"cubemx_publisher");

	msg.data = 0;

	for (;;)
	{
		ret = rcl_publish(&publisher, &msg, NULL);
		if (ret != RCL_RET_OK)
		{
			printf("Error publishing (line %d)\n", __LINE__);
		}

		msg.data++;
		osDelay(10);
	}
	/* USER CODE END StartMicroRos */
}
```

The code fully compiles. However, no messages arrive on the PC side. On the PC side I use the micro-ros-agent which I installed via snap. Here I get the following error:

```
maximilian@swift3:~$ micro-ros-agent serial  -D /dev/ttyUSB0 -v6 
[1626278380.941748] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 4
[1626278380.942016] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1626278386.437285] info     | Root.cpp           | create_client            | create                 | client_key: 0x1060D750, session_id: 0x81
[1626278386.437470] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x274782032, address: 0
[1626278386.437768] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1060D750, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1626278386.452495] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1060D750, len: 44, data: 
0000: 81 80 00 00 01 07 22 00 00 0A 00 01 01 03 00 00 14 00 00 00 00 01 A5 A5 0C 00 00 00 63 75 62 65
0020: 6D 78 5F 6E 6F 64 65 00 00 00 00 00
[1626278386.452871] debug    | ProxyClient.cpp    | create_participant       | DDS error              | client_key: 0x1060D750, participant_id: 0x000(1)
[1626278386.453244] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1060D750, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 80 00
[1626278386.453310] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x1060D750, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1626278386.468461] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x1060D750, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
```

By debugging I could find out that the DDS error is triggered by `rclc_node_init_default`. The return value of `rclc_node_init_default` is 1. Of course the whole code doesn't work after that.

I'm not even sure where the error comes from, the PC side or the Micro-Ros side.
I hope someone can help me. Thank you for your time.

pablogs9, 2021-07-15:
Hello, @maximiliankir sorry for that but our snap image of the micro-ROS Agent is outdated. We are working on it.

Is possible for you to use dockerized versions of the agent? They are up to date. Just:

```bash
docker run -it --rm --net=host -v /dev/shm:/dev/shm -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev [YOUR SERIAL DEVICE] -v6
```

Please, let me know if this works for you.

maximiliankir, 2021-07-15:
Thank you very much. With the docker version of the agent everything is working as intended

closed 2021-07-15 by maximiliankir

Mu-Raza, 2023-10-03:
can you please explain how to use docker version?

Mu-Raza, 2023-10-03:
i am using ro2 humble and micro-ros on esp32.