micro-ROS/micro_ros_stm32cubemx_utils #14: Adding custom messages and services

maximiliankir, 2021-07-15:
Hello,
I have another issue. I'm using the utils in the CubeIDE. I would like to add custom messages and services to project. I added a package with a custom message type to the `micro_ros_static_library_ide/library_generation/extra_packages` directory. But it is not included, when I build the project.
And secondly is it somehow possible to rebuild the micro-ros library, when custom packages have been added, without the apt install part, because this part of the building process takes very long.

Thank you for your time and your help.

pablogs9, 2021-07-15:
The build script should copy the folders that you add to `extra_sources` and import the repos that you add to `extra_packages.repos`. Check [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/ab3673ef65b25c630f68bc29c9fc8a56e8c1cb74/microros_static_library_ide/library_generation/library_generation.sh#L47).

Could you please @Acuadros95 check this and let us know if something is not working correctly?

maximiliankir, 2021-07-15:
So I should add the the package to `extra_sources` and not to `extra_packages`? I'll try that.

pablogs9, 2021-07-15:
Sorry my fault, `extra_packages` is where it should work. I'm going to edit my previous comment.

maximiliankir, 2021-07-15:
I have found the error. I did not copy the package to `extra_packages`, but only created a symbolic link. This is the reason why the content of the package is not found by the build script.

But can you tell me something about my second question. Do I have to rebuild the entire library every time I add or change packages? This takes quite a long time, mainly because of the apt downloads. Is there a way to speed up the build script so that it only looks for the new or changed packages and rebuilds incrementally?

pablogs9, 2021-07-15:
Using this docker approach is not possible to speed up the build using the incremental build, sorry...

We will put this in our roadmap to improve the build time. Thanks for the feedback.

If you don't mind I'm closing since it is solved. Do not hesitate to reopen if you have further problems.

closed 2021-07-15 by pablogs9