micro-ROS/micro_ros_stm32cubemx_utils #17: [Here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/galactic/sample_main.c) you have some demo code. Just use it in your main application.

elgarbe, 2021-08-09:
[Here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/galactic/sample_main.c) you have some demo code. Just use it in your main application.

Do not hesitate to reopen if you have any problem.

_Originally posted by @pablogs9 in https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/16#issuecomment-894968443_

Hi, I can't reopen this issue.
Ok for uC code. But, what about agent part? do I need to create/modify app-colcon.meta file?
I mean, this tutorial: https://micro.ros.org//docs/tutorials/core/first_application_rtos/freertos/ talks about app.c ans app-colcon.meta. But I'm nos sure how to combine those steps with cubemxIDE.

pablogs9, 2021-08-10:
Those steps are not required for `micro_ros_stm32cubemx_utils`. The steps are for [micro-ROS build system](https://micro.ros.org/docs/concepts/build_system/) and you are using a [external build system](https://micro.ros.org/docs/concepts/build_system/external_build_systems/). Just build and flash the CubeIDE project and run a micro-ROS Agent using, for example, the dockerized version:

```bash
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:galactic serial --dev [YOUR BOARD PORT] -v6
```

or 

```bash
docker run -it --rm --net=host microros/micro-ros-agent:galactic udp4 --port 8888 -v6
```

elgarbe, 2021-08-10:
Ok, thank, understand now.

closed 2021-08-10 by elgarbe