micro-ROS/micro_ros_stm32cubemx_utils #18: Problem creating more than 11 Publishers

bacurau, 2021-08-11:
Hi. I'm having difficulties with my micro ros project on STM32CubeIDE using the
micro_ros_stm32cubemx_utils. In my project I created 20 publishers, however, in fact, only the first 11 publishers are created. I found that in /micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include/rmw_microxrcedds_c/config.h file there is "#define RMW_UXRCE_MAX_PUBLISHERS 10+1". I changed the "10+1" to "20+1", cleaned up the project, and rebuilt it, but still only the first 11 publishers are created. Then I did another test, lower the number of publishers to "2+1", and still the first 11 publishers were created. It appears that the RMW_UXRCE_MAX_PUBLISHERS variable has no effect. Do you know what I can do to solve my problem (create more than 11 publishers)? I imagine I have re recompile something else. I thank you for your help!

pablogs9, 2021-08-12:
You have to configure your [colcon.meta](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/ab3673ef65b25c630f68bc29c9fc8a56e8c1cb74/microros_static_library_ide/library_generation/colcon.meta#L44) in order to configure the RMW. Modify this file and rebuild the library from scratch.

closed 2021-08-12 by pablogs9

bacurau, 2021-08-12:
Dear pablogs9, thank you very much for your quick reply. As you directed, I edited the field "-DRMW_UXRCE_MAX_PUBLISHERS=25" of the colcon.meta file and rebuilt the project (I initially rexecuted a "Project Clean" and then a "Project Build", in STM32CubeIDE). However, unfortunately, the number of publishers created is still limited to 11. Do I need to run any different commands to rebuild the project? Do you have any suggestions for anything else I can do? Thank you very much

pablogs9, 2021-08-13:
You need to remove manually the generated files in order to rebuild the library. A Project clean will not clean the micro-ROS related files.

bacurau, 2021-08-13:
Now it worked. Thank you very much!