micro-ROS/micro_ros_stm32cubemx_utils #23: How to set ROS_DOMAIN_ID in micro-ROS on STM32

bacurau, 2021-09-11:
Hi. In our project, we need to set the ROS_DOMAIN_ID to run the robot in an isolated logical network. However, to make the micro-ROS nodes visible in the computer embedded in the robot, that runs ROS 2, we cannot have the ROS_DOMAIN_ID set. There is any way to set the ROS_DOMAIN_ID in micro-ROS and make the micro-ROS nodes visible in ROS 2 with the ROS_DOMAIN_ID set? I tried to include the code presented in https://github.com/micro-ROS/micro_ros_arduino/issues/21#issuecomment-724558682, but the rcl_node_options_t does not contain the field domain_id. Regards!

pablogs9, 2021-09-13:
This API has changed from foxy to galactic.

Here a [galactic example](https://github.com/micro-ROS/micro-ROS-demos/blob/a7fd95132afe9208de0a69c0e5897f9d9db46e7e/rclc/configuration_example/configured_publisher/main.c#L49) and here a [foxy example](https://github.com/micro-ROS/micro-ROS-demos/blob/c67c38174b33e7fcf29ab81ea39cac0186ac40d6/rclc/configuration_example/configured_publisher/main.c#L52)

closed 2021-09-13 by pablogs9

bacurau, 2021-09-15:
Hi. I test the codes you suggested with both galatic and foxy APIs, it "worked", since now I can subscribe to the topics published by micro-ROS in the ROS2 with the ROS_DOMAIN_ID set. However, with both APIs, I can no longer view the nodes with the command "ros2 node list" nor the node details with the command "ros2 node info <node>". You know what's going on and how I can solve this problem? Regards!

pablogs9, 2021-09-16:
How are you building the micro-ROS Agent? Be sure that you have a recent version of it.

Also, make sure that you have the same `ROS_DOMAIN_ID` where you make the `ros2 node info`

bacurau, 2021-09-16:
Hi pablogs9. Thank you again for your support. I recompiled the agent following the tutorial in https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/ and it worked was expected. You were right, probably the problem was that I was using an old version of the agent. Best Regards!