micro-ROS/micro_ros_stm32cubemx_utils #24: How to sync the micro-ROS timestamp with the ROS2 timestamp (from the agent)

bacurau, 2021-09-15:
I would like to know what to do to  sync the micro-ROS timestamp with the ROS2 timestamp (from the agent). I saw in https://github.com/micro-ROS/rmw-microxrcedds/issues/116 that I can use the function "rmw_uros_sync_session". I tryed to use this funcion with the stm32cubemx_utils, both with galaxy and foxy APIs, and it did not work (the timestamp was not sync but I did not receive an error from the function "rmw_uros_sync_session"). I realized that the function rmw_uros_sync_session seems diferente in my system than the one presented in https://github.com/micro-ROS/rmw-microxrcedds/issues/116, since it has only one parameter "rmw_ret_t rmw_uros_sync_session( const int timeout_ms);", while in the example, it receive two parameters "rmw_ret_t rmw_uros_sync_session( int64_t * result, const int timeout_ms);". How can I solve my problem? Regards

Acuadros95, 2021-09-16:
The `rmw_uros_sync_session` function is supposed to only have one parameter, it was modified from the one present on that PR. 
You can check this example: [link](https://github.com/micro-ROS/micro-ROS-demos/blob/galactic/rclc/epoch_synchronization/main.c), and the available documentation: [link](https://micro.ros.org/docs/tutorials/programming_rcl_rclc/micro-ROS/#time-sync), [link](https://micro.ros.org/docs/api/rmw/#function-rmw_uros_epoch_synchronized).

Also, if `rmw_uros_sync_session` returned without errors, then the `rmw_uros_epoch` functions should return the actual epoch time.
How are you checking the timestamp?

bacurau, 2021-09-16:
Hi Acuadros95, thank you for your support. I did a test getting the time with rmw_uros_epoch_nanos(); and it worked as expected. So I discovered that my problem was that I was getting the time with clock_gettime(CLOCK_REALTIME, &ts);, I just substitute it by rmw_uros_epoch_nanos(); and it worked. Thank you very much!

Acuadros95, 2021-09-16:
Glad to help!

flux0-0, 2024-11-29:
hi @Acuadros95 , I use platform io and ros2 humble, when I run example code " https://github.com/micro-ROS/micro-ROS-demos/blob/galactic/rclc/epoch_synchronization/main.c " , I have error " collect2: error: ld returned 1 exit status
*** [.pio/build/esp32doit-devkit-v1/firmware.elf] Error 1 ". How can I fix it, this is my .ini file " [env:esp32doit-devkit-v1]
platform = espressif32@6.9.0
board = esp32doit-devkit-v1
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_port = /dev/ttyUSB0

board_microros_distro = humble
board_microros_transport = wifi
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
   "