micro-ROS/micro_ros_stm32cubemx_utils #40: Easier instructions for using STM32CubeIDE

pnn16004, 2021-11-20:
## Issue template

- Hardware description: STM32F303VC
- RTOS: FreeRTOS
- Installation type: STM32CubeIDE
- Version or commit hash: foxy

#### Steps to reproduce the issue
Download and install STM32CubeIDE from https://www.st.com/en/development-tools/stm32cubeide.html
Open up STM32CubeIDE and create a new project, choosing STM32F303VC as the board.
Follow the steps at https://github.com/micro-ROS/micro_ros_stm32cubemx_utils#using-this-package-with-stm32cubeide

#### Expected behavior
Successfully build the project.

#### Actual behavior
`
21:06:18 **** Incremental Build of configuration Debug for project microROS_test ****
make -j4 all 
docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v :/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:foxy
foxy: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:33463a1857e63af0edf46178ddb2aeb9bb880a73a4e10fb03745a79751135198
Status: Image is up to date for microros/micro_ros_static_library_builder:foxy
docker.io/microros/micro_ros_static_library_builder:foxy
/entrypoint.sh: line 9: /project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/library_generation.sh: No such file or directory
make[1]: *** [makefile:91: pre-build] Error 127
make: *** [makefile:59: all] Error 2
"make -j4 all" terminated with exit code 2. Build might be incomplete.

21:06:21 Build Failed. 2 errors, 0 warnings. (took 3s.296ms)
`

#### Additional information
I have no prior knowledge of STM32CubeIDE, so I have probably made some mistake when following the instructions.

Could someone please breakdown the instructions for someone who has never worked with a STM32 before? I have worked a lot with ESP32 and micro-ROS, which only needed the command line, not an IDE.

pnn16004, 2021-11-21:
So it seems like the docker script can't find the utils map. I placed the utils folder inside the root of my project, and I even tested placing it inside a folder named project, but the docker still won't work.

With the STM32CubeMX steps I am able to use the docker, however, when I try to build the sample_main.c example, the builder can't find all the functions such as rcutils_get_zero_initialized_allocator, rcutils_set_default_allocator, etc.

I placed the microros_static_library inside Core->Inc, because before when I had it at the root folder, it wouldn't even find the includes.

pablogs9, 2021-11-22:
Your docker command is not being generated well. Check the `-v :/project` it is not filling correctly the `-v ${workspace_loc:/${ProjName}}:/project`. Which version are you using?

pnn16004, 2021-11-22:
Not sure if this is what you meant, but when I run `docker -v :/project` it says:

> Docker version 20.10.11, build dea9396

pablogs9, 2021-11-23:
which version of STM32CubeIDE are you using?

pnn16004, 2021-11-23:
It's STM32CubeIDE-DEB 1.7.0, the latest one.

pablogs9, 2021-11-23:
This is the first line in the console when I click on the build button: 

```
docker pull microros/micro_ros_static_library_builder:galactic && docker run --rm -v /home/pgarrido/STM32CubeIDE/workspace_1.7.0/test:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:galactic
```

As you can see, it is just the same we write in pre-steps, but with some "environment variables" correctly set:

```
docker pull microros/micro_ros_static_library_builder:galactic && docker run --rm -v ${workspace_loc:/${ProjName}}:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:galactic
```

In your case, for some reason `${workspace_loc:/${ProjName}}` is not being converted to the project path. Maybe you can to set it manually to the path where your micro-ROS project is.

pnn16004, 2021-11-23:
It successfully downloaded all the dependencies now, thank you!

I did however bump into a new problem, I get this error now:

```
20:09:13 **** Incremental Build of configuration Debug for project microROS_test ****
make -j4 all 
docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v /home/peter/STM32CubeIDE/workspace_1.7.0/microROS_test:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:foxy
foxy: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:33463a1857e63af0edf46178ddb2aeb9bb880a73a4e10fb03745a79751135198
Status: Image is up to date for microros/micro_ros_static_library_builder:foxy
docker.io/microros/micro_ros_static_library_builder:foxy
micro-ROS library found. Skipping...
Delete micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/ for rebuild.
 
arm-none-eabi-gcc "../Core/Src/custom_memory_manager.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/custom_memory_manager.d" -MT"Core/Src/custom_memory_manager.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/custom_memory_manager.o"
arm-none-eabi-gcc "../Core/Src/dma_transport.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/dma_transport.d" -MT"Core/Src/dma_transport.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/dma_transport.o"
arm-none-eabi-gcc "../Core/Src/main.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/main.d" -MT"Core/Src/main.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/main.o"
arm-none-eabi-gcc "../Core/Src/microros_allocators.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/microros_allocators.d" -MT"Core/Src/microros_allocators.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/microros_allocators.o"
../Core/Src/dma_transport.c:1:10: fatal error: uxr/client/transport.h: No such file or directory
    1 | #include <uxr/client/transport.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:46: Core/Src/dma_transport.o] Error 1
make[1]: *** Waiting for unfinished jobs....
../Core/Src/microros_allocators.c:3:10: fatal error: cmsis_os.h: No such file or directory
    3 | #include "cmsis_os.h"
      |          ^~~~~~~~~~~~
../Core/Src/custom_memory_manager.c:13:10: fatal error: FreeRTOS.h: No such file or directory
   13 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:46: Core/Src/custom_memory_manager.o] Error 1
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:46: Core/Src/microros_allocators.o] Error 1
../Core/Src/main.c:25:10: fatal error: rcl/rcl.h: No such file or directory
   25 | #include <rcl/rcl.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:46: Core/Src/main.o] Error 1
make: *** [makefile:59: all] Error 2
"make -j4 all" terminated with exit code 2. Build might be incomplete.

20:09:19 Build Failed. 10 errors, 0 warnings. (took 5s.119ms)
```

I'm assuming it's something with step 3, and I've triple checked that I put the correct path at the correct place.

I also tried changing it to the absolute path, and adding "", or using ../micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include like the other include paths have, but no luck.

I also checked the include map that everything is there.

The #include "FreeRTOS.h" is probably a separate problem, because in another project I made there is a Middleware map with FreeRTOS in it, and that can include it with no problem, so I should be able to solve that.

Any ideas about the other problems though?

pnn16004, 2021-11-23:
Okay, I solved FreeRTOS.h and cmsis_os.h by clicking rebuild when the .ioc file was in focus. In FREERTOS->Advanced settings I changed USE_NEWLIB_REENTRANT to Enabled, and changed SYS->Timebase Source to TIM6, not sure if you're supposed to do this.

I also get some new problems this time:
```
20:30:48 **** Incremental Build of configuration Debug for project microROS_test ****
make -j4 all 
docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v /home/peter/STM32CubeIDE/workspace_1.7.0/microROS_test:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:foxy
foxy: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:33463a1857e63af0edf46178ddb2aeb9bb880a73a4e10fb03745a79751135198
Status: Image is up to date for microros/micro_ros_static_library_builder:foxy
docker.io/microros/micro_ros_static_library_builder:foxy
micro-ROS library found. Skipping...
Delete micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/ for rebuild.
 
arm-none-eabi-gcc "../Core/Src/dma_transport.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/dma_transport.d" -MT"Core/Src/dma_transport.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/dma_transport.o"
arm-none-eabi-gcc "../Core/Src/main.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/main.d" -MT"Core/Src/main.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/main.o"
arm-none-eabi-gcc "../Core/Src/microros_allocators.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/microros_allocators.d" -MT"Core/Src/microros_allocators.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/microros_allocators.o"
arm-none-eabi-gcc "../Core/Src/microros_time.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/microros_time.d" -MT"Core/Src/microros_time.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/microros_time.o"
../Core/Src/dma_transport.c:1:10: fatal error: uxr/client/transport.h: No such file or directory
    1 | #include <uxr/client/transport.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:52: Core/Src/dma_transport.o] Error 1
make[1]: *** Waiting for unfinished jobs....
../Core/Src/microros_time.c: In function 'clock_gettime':
../Core/Src/microros_time.c:32:5: error: unknown type name 'TimeOut_t'
   32 |     TimeOut_t xCurrentTime = { 0 };
      |     ^~~~~~~~~
../Core/Src/microros_time.c:44:5: warning: implicit declaration of function 'vTaskSetTimeOutState' [-Wimplicit-function-declaration]
   44 |     vTaskSetTimeOutState( &xCurrentTime );
      |     ^~~~~~~~~~~~~~~~~~~~
../Core/Src/microros_time.c:48:47: error: request for member 'xOverflowCount' in something not a structure or union
   48 |     ullTickCount = ( uint64_t ) ( xCurrentTime.xOverflowCount ) << ( sizeof( TickType_t ) * 8 );
      |                                               ^
../Core/Src/microros_time.c:51:33: error: request for member 'xTimeOnEntering' in something not a structure or union
   51 |     ullTickCount += xCurrentTime.xTimeOnEntering;
      |                                 ^
make[1]: *** [Core/Src/subdir.mk:52: Core/Src/microros_time.o] Error 1
../Core/Src/main.c:26:10: fatal error: rcl/rcl.h: No such file or directory
   26 | #include <rcl/rcl.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:52: Core/Src/main.o] Error 1
make: *** [makefile:63: all] Error 2
"make -j4 all" terminated with exit code 2. Build might be incomplete.

20:30:52 Build Failed. 10 errors, 1 warnings. (took 4s.163ms)
```

The IDE is able to find these files when I ctrl + click them, so I'm not sure what's going on.

pnn16004, 2021-11-23:
I now put in all the important things from sample_main.c into my code and I get these errors now:

```
20:51:48 **** Incremental Build of configuration Debug for project microROS_test ****
make -j4 all 
docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v /home/peter/STM32CubeIDE/workspace_1.7.0/microROS_test:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:foxy
foxy: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:33463a1857e63af0edf46178ddb2aeb9bb880a73a4e10fb03745a79751135198
Status: Image is up to date for microros/micro_ros_static_library_builder:foxy
docker.io/microros/micro_ros_static_library_builder:foxy
micro-ROS library found. Skipping...
Delete micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/ for rebuild.
 
arm-none-eabi-gcc "../Core/Src/dma_transport.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/dma_transport.d" -MT"Core/Src/dma_transport.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/dma_transport.o"
arm-none-eabi-gcc "../Core/Src/main.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/main.d" -MT"Core/Src/main.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/main.o"
arm-none-eabi-gcc "../Core/Src/microros_time.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/microros_time.d" -MT"Core/Src/microros_time.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/microros_time.o"
arm-none-eabi-gcc "../Core/Src/syscalls.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F303xC -c -Imicro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Core/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Drivers/STM32F3xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/syscalls.d" -MT"Core/Src/syscalls.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/syscalls.o"
../Core/Src/dma_transport.c:1:10: fatal error: uxr/client/transport.h: No such file or directory
    1 | #include <uxr/client/transport.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:52: Core/Src/dma_transport.o] Error 1
make[1]: *** Waiting for unfinished jobs....
../Core/Src/microros_time.c: In function 'clock_gettime':
../Core/Src/microros_time.c:32:5: error: unknown type name 'TimeOut_t'
   32 |     TimeOut_t xCurrentTime = { 0 };
      |     ^~~~~~~~~
../Core/Src/microros_time.c:44:5: warning: implicit declaration of function 'vTaskSetTimeOutState' [-Wimplicit-function-declaration]
   44 |     vTaskSetTimeOutState( &xCurrentTime );
      |     ^~~~~~~~~~~~~~~~~~~~
../Core/Src/microros_time.c:48:47: error: request for member 'xOverflowCount' in something not a structure or union
   48 |     ullTickCount = ( uint64_t ) ( xCurrentTime.xOverflowCount ) << ( sizeof( TickType_t ) * 8 );
      |                                               ^
../Core/Src/microros_time.c:51:33: error: request for member 'xTimeOnEntering' in something not a structure or union
   51 |     ullTickCount += xCurrentTime.xTimeOnEntering;
      |                                 ^
make[1]: *** [Core/Src/subdir.mk:52: Core/Src/microros_time.o] Error 1
../Core/Src/main.c:26:10: fatal error: rcl/rcl.h: No such file or directory
   26 | #include <rcl/rcl.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:52: Core/Src/main.o] Error 1
make: *** [makefile:63: all] Error 2
"make -j4 all" terminated with exit code 2. Build might be incomplete.

20:51:53 Build Failed. 10 errors, 1 warnings. (took 5s.310ms)
```

I attached my project if you want to try it.

[microROS_test.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/7590700/microROS_test.zip)

pablogs9, 2021-11-24:
It seems that you dont have `/microROS_test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include/` in your include path and thats why for example `rcl/rcl.h` is not being found

pnn16004, 2021-11-24:
Neither `/microROS_test/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include/` or `/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include/` worked, but now when I tried `../micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include`, it works!

For some reason I had to manually include task.h in microros_time.c.

After this I got some other errors from the sample_main.c code, but after some googling I realized I had to turn on Connectivity->USART3. I picked Asynchronous.

Now it builds and I can flash it, but it won't connect to the micro-ROS Agent.

I guessing from sample_main.c that System Core->DMA->DMA1->USART3_RX and USART3_TX are turned on. So I did that as well with no problems.

From sample_main.c I also see the line `PCD_HandleTypeDef hpcd_USB_OTG_FS`, so I'm guessing it's Connectivity->USB, but when I turn that on, I get problems with the Clock Configuration that can't be resolved.

Thus I didn't turn it on and went ahead and built and flashed the project onto my board, but still no luck connecting to the micro-ROS Agent.

How do I solve the USB thing, or is it necessary? I attached my project again if you want to try.

[microROS_test.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/7598327/microROS_test.zip)

pnn16004, 2021-11-24:
Okay, I was able to turn on Connectivity->USB by first turning on System Core->RCC->HSE->BYPASS Clock Source, and then clicking Resolve Clock Issues.

I now get `PCD_HandleTypeDef hpcd_USB_FS`, which is similar but not the same, not sure if it matters.

Built it and flashed it again, still won't work with the micro-ROS Agent.

I connected to the micro-ROS Agent using:

> docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0670FF545150898367145119-if02 -v6

pnn16004, 2021-11-28:
I have now created a new project, using the Board Selector (STM32F3DISCOVERY) instead of the MCU/MPU Selector.

I did steps 1-5 as usual, but at step 6 I can't get the stack size to 3000, only 768, so I set it to 700. I also edited the colcon.meta file to only use 1 node, 1 publisher, 1 history, and the others 0.

I forgot to look at step 7 in my previous project, so I followed it now, using U(S)ART with DMA.

After building and running I still can't connect to the Agent, so I debugged it to see what's wrong. I see that rclc_support_init takes a very long time to complete, and also returns a 1, meaning it failed. I'm not able to step into the function, so I can't figure out what's wrong.

Any ideas @pablogs9?

[utils_test.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/7613121/utils_test.zip)

pablogs9, 2021-11-29:
Is your code writing correctly to the serial port? Can you see data being sent using `minicom` or `miniterm`?

pnn16004, 2021-11-29:
No I can't see anything, tried both `minicom` and `miniterm`. The flashed code is using `printf`, and I have set U(S)ART according to the instructions, with baud rate 38400.

`miniterm` correctly lists
> 1: /dev/ttyACM0         'STM32 STLink - ST-Link VCP Ctrl'
But nothing is printed even when I set the correct baud rate.

`tty` gives me `/dev/pts/0`, and `stty` gives me
> speed 38400 baud; line = 0;
> -brkint -imaxbel iutf8

I tried `minicom` with both `/dev/ttyACM0` and `/dev/ttyS0`, but nothing is shown.

pablogs9, 2021-11-30:
So the point is that your UART is not working properly. micro-ROS needs a well-configured UART:

pnn16004, 2021-11-30:
Oh okay, was hoping Board Selector would fix that for me. I'll try to get a `printf` working then before testing with micro-ROS again.

pnn16004, 2021-12-11:
Hi @pablogs9,

After a lot of struggling I've finally gotten `printf` to work, I just had to switch to USART1. I get a clean output from miniterm now.

The problem now is that it gets stuck on `rclc_support_init`. If I debug it and pause, I see that I'm stuck inside task.c.

After pausing the execution a few times, I see that it jumps between the functions `static portTASK_FUNCTION`, `static void prvCheckTasksWaitingTermination`, `void vTaskSwitchContext` and `void vPortExitCritical`.

Any idea what it might be this time?

pablogs9, 2021-12-13:
How much stack have you configured for the micro-ROS task?

pnn16004, 2021-12-13:
700, max is 768. I also tried with 720, I don't think I can go above that. When I put the stack at too low, like 100, I get stuck in ErrorDefaultHandler.

pablogs9, 2021-12-13:
Could you share your latest code?

pnn16004, 2021-12-13:
This one is made with the MCU Selector: [micro_test.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/7707485/micro_test.zip)

I am able to go up to 710 stack, but when I go up to 720 I get stuck at `osKernelStart` instead of `rclc_support_init`.

I made a new one today with Board Selector, it actually gets through `rclc_support_init`, but returns 1. When it's on the `rclc_support_init` function it prints out:

> ~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��~�XRCE5ԁ��

I have USART1 baud rate at 38400. This is the new project: [full_test.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/7707490/full_test.zip)

The max stack I could set for the task is 738 before the heap complains, but when I have it that high it just gets stuck at `osKernelStart`.

pablogs9, 2021-12-14:
Where are you receiving this data? This is supposed to be received by the agent, if you have a minicom session, the agent won't receive it.

pnn16004, 2021-12-14:
Yes that makes sense. I received it with the Command Shell Console in the IDE.

I tried it again now not receiving the data and used LEDs instead to know when `rclc_support_init` and the rest of the micro-ROS functions finished. Still not working.

I've tried putting lower stack sizes like 600 and 500 for FreeRTOS and not getting stuck, but agent still won't connect.

I now put in `while (RMW_RET_OK != rmw_uros_ping_agent(1000, 1));`, but it doesn't respond to me starting the agent. Might the problem be there? That it doesn't see the agent at all.

I start the agent with
> docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6

pnn16004, 2021-12-14:
Okay I don't know what's going on but it's working now. Somehow it only works if I first start the agent, then start either PuTTY or Command Shell Console, then restart the board.

If I don't start PuTTY or Console it doesn't work. I can disconnect the monitoring after connecting to the agent and it still works, and I can even restart the board and it instantly connects, but if I stop the agent I have to do the steps again.

It's easier to do if I have `rmw_uros_ping_agent` at the start, but I'm able to do it without it too.

I'm so relived I got this to work before my thesis starts, thank you so much!

If you don't mind, could you just explain what's going on, so I don't need to do these steps every time.

[full_test.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/7714688/full_test.zip)

pablogs9, 2021-12-15:
This is not a normal operation, maybe it is related to some specific configuration of your computer.
Please report if you find some solution.
I'm closing, do not hesitate to reopen if you have further questions.

closed 2021-12-15 by pablogs9

rsc9421, 2022-01-16:
hey guys, i have the same problem

pnn16004, 2022-01-16:
Which one?

rsc9421, 2022-01-16:
i followed the instructions for STM32CUBEIDE and i get these errors:

[image]

but if i followed the steps provided by @pnn16004 I can compile the program but doesn't work, I think the problem is in the 10 kB of stack used by the task, because when i run a simple print over serial port, using 1200 of stack i can see that the peripheral is working fine, but when i change the 1200 by 12000 the program desnt work
the problem is not the memory capacity, I'm using a nucleo64 F303RE
here is the memory used:
[image]

pnn16004, 2022-01-16:
If it works with 1200, why aren't you using that? It says in the instructions to use 10 kB stack but I think that's outdated [#35](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/35) . I managed to do fine with 700 * 4 stack size.

What do you mean the program isn't working?

rsc9421, 2022-01-16:
the simple serial:
```
while(1){
	HAL_UART_Transmit(&huart2, "HOLA", 4, 1000);
	osDelay(500);
	}
```
inside the task works (using 1200) but if change the task by the task provided by the example (the ros publisher) and run the ros agent using this command

docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyACM0 -v6

no ros2 topic appears, that's why I said it doesn't work

pnn16004, 2022-01-17:
Does it successfully connect to the agent or is it just that the topic isn't appearing?

rsc9421, 2022-01-17:
This is the response of the agent
[image]

I get the same response if I run the micro-ros example or a simple led blink

My microcontroller gets freezes, I made some "debug" to see where is the problem and everything works fine before the function
 rclc_support_init(&support, 0, NULL, &allocator);
after this function I can't turn on a led, the microcontroller just freezes

This should not happen since the code was compiled correctly

pnn16004, 2022-01-17:
I had the same problem. Check the replies in this thread below the following comment:

> Hi @pablogs9,
>
> After a lot of struggling I've finally gotten `printf` to work, I just had to switch to USART1. I get a clean output from miniterm now.
>
> The problem now is that it gets stuck on [...]

I solved this issue using PuTTY, but you can also do it with the Command Shell Console inside STM32CubeIDE.

tianrking, 2023-01-11:
> 

hi Same problem here.  Thanks for ur replay. Could u plz teach me how to solve it /