micro-ROS/micro_ros_stm32cubemx_utils #41: micro_ros without FreeRTOS

rscova, 2021-12-17:
Hi,

I am trying to use stm32CubeIDE configuring my board without FreeRTOS. Is this possible?

I am getting these errors:
```
../Core/Src/custom_memory_manager.c:13:10: fatal error: FreeRTOS.h: No such file or directory
   13 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:46: Core/Src/custom_memory_manager.o] Error 1
make[1]: *** Waiting for unfinished jobs....
../Core/Src/dma_transport.c:6:10: fatal error: cmsis_os.h: No such file or directory
    6 | #include "cmsis_os.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:46: Core/Src/dma_transport.o] Error 1
make: *** [makefile:77: all] Error 2
"make -j8 all" terminated with exit code 2. Build might be incomplete
```

pablogs9, 2021-12-17:
Try to not include `custom_memory_manager.c` and avoid setting custom allocators [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/8fb2233e24f071507908c5f89e8a5608c6e5cc89/sample_main.c#L381)

Let us know if this works.

rscova, 2021-12-17:
Thanks @pablogs9 

Still having errors with `dma_trasnport.c `,  `microros_allocators.c ` and  `microros_time.c `

```
../Core/Src/microros_time.c:3:10: fatal error: cmsis_os.h: No such file or directory
    3 | #include "cmsis_os.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:43: Core/Src/microros_time.o] Error 1
make[1]: *** Waiting for unfinished jobs....
../Core/Src/microros_allocators.c:3:10: fatal error: cmsis_os.h: No such file or directory
    3 | #include "cmsis_os.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[1]: *** [Core/Src/subdir.mk:43: Core/Src/microros_allocators.o] Error 1
../Core/Src/dma_transport.c: In function 'cubemx_transport_write':
../Core/Src/dma_transport.c:40:13: warning: implicit declaration of function 'osDelay' [-Wimplicit-function-declaration]
   40 |             osDelay(1);
      |             ^~~~~~~
../Core/Src/dma_transport.c: In function 'cubemx_transport_read':
../Core/Src/dma_transport.c:59:17: error: 'portTICK_RATE_MS' undeclared (first use in this function)
   59 |         osDelay(portTICK_RATE_MS);
      |                 ^~~~~~~~~~~~~~~~
../Core/Src/dma_transport.c:59:17: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [Core/Src/subdir.mk:43: Core/Src/dma_transport.o] Error 1
make: *** [makefile:77: all] Error 2
```

pablogs9, 2021-12-17:
Ok, so not possible with the current status of this repo. For sure it is possible to run micro-ROS bare metal using CubeMX but it will need a major refactor of this repo.

We should remove dependencies with FreeRTOS and CMSIS_OS and move then to HAL/libc functionality.

pablogs9, 2021-12-17:
In general:
- `microros_allocators.c` can be removed and rely on libc `malloc`
- `dma_transports.c` can be used without the delays functions as you will be using a single-thread app. Ideas here @Acuadros95 ?
- `microros_time.c` will need a complete refactor to retrieve time from a bare timer configured in the HAL.

Acuadros95, 2021-12-17:
We could also use `HAL_Delay(ms)` as an alternative for transport delays.

pablogs9, 2021-12-17:
We should have a conditional compilation option because AFAIK `HAL_Delay` does not work with RTOS. Maybe we can check if CMSIS is enabled and use `osDelay` or `HAL_Delay` appropriately.

pablogs9, 2022-01-28:
Any update on this?

eranroll, 2022-02-05:
Hi,

As I am sure that this modification can be done locally, allowing this project to support MicroROS without FreeRTOS will be a huge benefit.

Thanks in advance!

closed 2022-02-07 by pablogs9

bjsowa, 2022-02-25:
Why was this closed?