micro-ROS/micro_ros_stm32cubemx_utils #45: Use of VFD arguments

s-waguespack, 2022-01-25:
Hello,

I am trying to implement micro ros on a custom board based on the STM32L071C8T6 however I am running into this error: 

`Description	Resource	Path	Location	Type
failed to merge target specific data of file /home/spencer/SoftwareSource/STM32Source/STM32L071C8T6-MicroRosDriver/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-allocator.c.obj)	STM32L071C8T6-MicroRosDriver		 	C/C++ Problem
`

Some googling lead me to several forum posts from users of various arm based micro controllers with the problem being related to the floating point abi flag being set to soft or hard however with the STM32L071C8T6 chip I cannot change to hard (there isnt a floating point unit so it uses software floats)

pablogs9, 2022-01-26:
Could you check that when you build the library the flags reported with `Found CFLAGS:...` are ok?
Are you using CubeIDE or CubeMX? 
Could you share a basic reproducible scenario?

pablogs9, 2022-01-26:
Which compiler are you using in CubeIDE? Notice that the the micro-ROS library is being built using [`arm-none-eabi-gcc` version available in Ubuntu 20.04](https://launchpad.net/ubuntu/focal/+package/gcc-arm-none-eabi).

Make sure that the compiler you are using is the same as this. 

We are working on solving this limitation in order to be able to build micro-ROS with the same default compiler of CubeMX/IDE

s-waguespack, 2022-01-26:
the GNU tools for STM32 (9-2020-q2-update)

This is the options flag from cubeide 

`-mcpu=cortex-m0plus -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L071xx -c -I../Core/Inc -I"/home/spencer/SoftwareSource/STM32Source/STM32L071C8T6-MicroRosDriver/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include" -I../Drivers/STM32L0xx_HAL_Driver/Inc -I../Drivers/STM32L0xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32L0xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM0 -Os -ffunction-sections -fdata-sections -Wall -fstack-usage --specs=nano.specs -mfloat-abi=soft -mthumb`

pablogs9, 2022-01-26:
I guess that compiler versions are not compatible.

LegoYoda112, 2023-09-26:
I'm running into the same issue attempting to compile for an STM32G431 using humble.

To minimise chances of compiler mismatches, I'm using the same arm gcc within the docker image that was used to build the static library (10.3.1)
[image]

Using `-mfloat-abi=hard` results in `failed to merge target specific data` errors.
Using `-mfloat-abi=soft` results in `Error: selected FPU does not support instruction -- vstmdbeq r0!,{s16-s31}` errors.

Since this MCU has an FPU, I'd assume that `hard` is the preferred option, but am not sure where to look next in my troubleshooting.

pablogs9, 2023-09-27:
When you build micro-ROS which build flags does the build system print?

LegoYoda112, 2023-09-27:
Aha! Looks like it was not picking up the CFLAGS correctly from my Makefile; I've successfully compiled now.

The readme contained spaces not tabs in 
```
print_cflags:
	@echo $(CFLAGS)
```

Which caused a missing separator error that was not visible during the static library generation.

[image]

pablogs9, 2023-09-27:
Working now?