micro-ROS/micro_ros_stm32cubemx_utils #46: embeddedRTPS example with NUCLEO-F767ZI

pnn16004, 2022-02-05:
## Issue template

- Hardware description: NUCLEO-F767ZIT6
- RTOS: FreeRTOS
- Installation type: STM32CubeIDE, micro_ros_stm32cubemx_utils
- Version or commit hash: galactic

#### Steps to reproduce the issue
Follow the embeddedRTPS instructions [md](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/galactic/embeddedrtps.md).

#### Expected behavior
Visible with either `ros2 topic list` or Wireshark, or respond to listener on ROS2 side.

#### Actual behavior
No evidence that it works, but no error on the microcontroller side when monitored with PuTTY.

#### Additional information
I've double checked that I followed the instructions correctly. The only step I'm unsure about is 5. LwIP -> General Settings -> IP Address Settings (Set here the board address and mask). I used the following settings:
> IP: 192.168.001.086
> Netmask: 255.255.255.000
> Gateway: 192.168.001.001

With `ifconfig` I see that my own IP and netmask are 192.168.1.84 and 255.255.255.0, respectively for enp3s0. With `ip r` I see that my gateway is 192.168.1.1. Not sure if I'm supposed to mimic the settings like this.

I made sure to disable WiFi and export FAST-DDS for `ros2 topic list` and listener. I don't have the board and the computer in an isolated network, but shouldn't I at least be able to see packets from 192.168.1.86 with Wireshark?

[ping_test_f7.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/8007017/ping_test_f7.zip)

pablogs9, 2022-02-07:
Could you debug step by step in order to provide more information about where is the system failing?

pnn16004, 2022-02-07:
What do you mean specifically? When I checked the return values of all the `rcl` and `rclc` functions it's 0. `rcl_publish()` also returns 0, so it doesn't feel like there's anything wrong there.

pablogs9, 2022-02-07:
Maybe check if embeddedRTPS threads are being created and if any package is being sent at LwIP level. Which kind of traffic do you see in Wireshark? Do you have any RTPS messages from the board IP?

pnn16004, 2022-02-07:
I don't see any RTPS messages unless I export Fast DDS for ROS2. I don't see any messages from the board, 192.168.1.86. I just started using Wireshark, so I'm not if I'm looking at it correctly. How would I check if embeddedRTPS threads are created? I only know how to check threads on the computer.

Wireshark logs: [test2.zip](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/8013274/test2.zip)

pablogs9, 2022-02-07:
You should be able to debug thread status with STM32CubeIDE and STLink probe. Please check ST documentation.

ZhenshengLee, 2022-04-06:
@pablogs9  

I think the https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/galactic/embeddedrtps.md is not enough to do a complete test with ertps.

for example, `ros2 topic/node list` outputs what when stm32 board is running.

I thought the output may be like what @pnn16004 said 

> Expected behavior
> Visible with either ros2 topic list or Wireshark, or respond to listener on ROS2 side.

But after seeing https://github.com/micro-ROS/rmw_embeddedrtps/blob/756e867f88f882a8db0c18e198fb6b9bbf22a8c2/CMakeLists.txt#L34

> `option(RMW_ERTPS_GRAPH "Allows to perform graph-related operations to the user" OFF)`

I need to guess the meaning of this option, guess the expected bahavior of `ros2 node list` and  wonder if I should make this open manually.

pablogs9, 2022-04-07:
@ZhenshengLee As estated in the `rmw_embeddedrtps` README.md this package is **experimental**. That means that it has a minimal set of functionality for making the middleware interoperable with ROS 2. This functionality is basically a basic pub/sub implementation on top of micro-ROS and embeddedRTPS. Graph management (required for `ros2 node list`) is not handled in this RMW, nor many other features. As you can see in the documentation, the default middleware for micro-ROS is Micro XRCE-DDS, which has a much more complete list of features.

Said that, `rmw_embeddedrtps` is an open source experimental contribution to the micro-ROS ecosystem, mostly in order to show an example on how to implement differents middlewares in micro-ROS. Feel completely free to modify the codebase and contribute your improvements via pull requests. Also, if you need consulting services and/or feature acceleration services do not hesitate to contact eProsima's commercial department at mariamerlan@eprosima.com.

ZhenshengLee, 2022-04-07:
> Said that, `rmw_embeddedrtps` is an open source experimental contribution to the micro-ROS ecosystem, mostly in order to show an example on how to implement differents middlewares in micro-ROS. Feel completely free to modify the codebase and contribute your improvements via pull requests

Understood, thanks.