micro-ROS/micro_ros_stm32cubemx_utils #51: Behavior of `microros_allocate(0)`?

yushijinhun, 2022-03-09:
I found the behavior of `microros_allocate()` is inconsistent with RCL's default allocator implementation `malloc()`. When allocating zero size memory area, `microros_allocate()` returns `NULL` while `malloc()` (on Linux) returns a valid memory address.

I encounter this problem when trying to turn on `RCL_COMMAND_LINE_ENABLED`. I pass an empty commandline argument into `rclc_support_init()` (`argc = 1, argv = {""}`), and then ROS crashes at [`rcl_init`](https://github.com/ros2/rcl/blob/54339461b8b2aecf836bf8681a061c2f601a582f/rcl/src/rcl/init.c#L117-L121) because it tries to allocate 0 byte memory.

The workaround is quite simple: just allocate at least 1 byte in `microros_allocate()` if requested size is 0. However, in [the manual of `malloc()`](https://man7.org/linux/man-pages/man3/malloc.3.html), it says:
> NULL may also be returned by a successful call to malloc() with a size of zero, or by a successful call to calloc() with nmemb or size equal to zero.

I'm not sure whether it's a micro-ROS's bug or RCL's bug.

pablogs9, 2022-03-09:
I guess that we comply with `malloc()` functionality. As far as we try to save as much memory as possible in the micro-ROS stack, maybe is better to fix the RCL code to handle the `NULL` return in the CLI parsing.

yushijinhun, 2022-03-15:
I think the root of the problem is that RCL misses the terminating null byte when copying strings. I opened https://github.com/ros2/rcl/pull/969 to solve this.

closed 2022-03-15 by yushijinhun