micro-ROS/micro_ros_stm32cubemx_utils #60: micro-ROS agent does not initialize any communication

jdgg98, 2022-05-11:
Hello, I've been trying to setup micro-ROS on a STM32F407VGT6 microcontroller. I'm using ROS2's foxy distro, FreeRTOS and micro_ros_stm32cubemx_utils. I though I was almost done with a simple publisher example, but when I start the micro-ros-agent it seems like it is not able to initialize any communication with the publisher on my microcontroller. This is the result a I get:

[image]

Does anyone have a clue about what might be wrong?

pablogs9, 2022-05-12:
Can you check in your MCU UART port if something is being written using a tool such as miniterm or minicom.

If your micro-ROS client is behaving correctly, some bytes should be being written in `/dev/ttyACM0`

pablogs9, 2022-05-17:
Any update on this? @jdgg98

jdgg98, 2022-05-18:
Hi! I used minicom as you told me and it seems that nothing is being written in `/dev/ttyACM0`. I also found out that I might have made some mistakes in the FreeRTOS and UART configurations in STMCubeMX, so I will fix them and try again.

jdgg98, 2022-06-15:
At the end it turns out that microROS is not compatible with the firmware library I am using, so I had to discard the use of microROS.

closed 2022-06-16 by pablogs9

do-coding-J, 2022-10-18:
I've got something on minicom with /dev/ttyACM0 but micro ros agent can't find my board