micro-ROS/micro_ros_stm32cubemx_utils #62: Session with microros agent is constantly re-established

leoborgnino, 2022-06-09:
**Issue template**
_____________________
    Hardware description: STMF407
    RTOS: FreeRTOS
    Installation type: STM32CubeIDE, micro_ros_stm32cubemx_utils
    Version or commit hash: galactic

**Steps to reproduce the issue**

Follow [micro_ros_stm32cubemx_utils](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils). I'm using UART and DMA approach. From PC side I'm using _docker run -it --rm --net=host -v /dev:/dev --privileged microros/micro-ros-agent:galactic serial --dev /dev/ttyUSB0 -b 115200 -v6_

**Expected behavior**

Handshake between MCU and MicroROS agent. Discover topics on ROS2.

**Actual behavior**

Session is constantly re-established. DDS log never comes up. Serial data seems to be sent and received. (Log added in Additional information).

**Additional information**

This behaviour is constantly repeated through time, this is only the first lines. In MCU side, code hangs in rclc_support_init().

[image]

If I use minicom I can see MCU sending data at 115200b.

[image]

pablogs9, 2022-06-10:
@Acuadros95 can you take a look?

Acuadros95, 2022-06-10:
Hi @leoborgnino:
- Are you using minicom at the same time as the agent?
- Can you start the Agent before the board is connected? In `rclc_support_init()` The MCU will try to connect with the Agent for 10 seconds before it gives up. This could lead to your problem is the agent is not available when the board is connected.
- You can use the ping functionality to avoid this, check this example: [micro-ros_reconnection_example](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino)

leoborgnino, 2022-06-10:
> Hi @leoborgnino:
> 
> * Are you using minicom at the same time as the agent?
> * Can you start the Agent before the board is connected? In `rclc_support_init()` The MCU will try to connect with the Agent for 10 seconds before it gives up. This could lead to your problem is the agent is not available when the board is connected.
> * You can use the `rmw_uros_ping_agent` functionality to avoid this, check this example: [micro-ros_reconnection_example](https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino)

Hi Pablo and Antonio, thanks for the fast reply!

* I only used minicom to test if MCU was sending data at 115200b but I was not using it with the agent.
* I start the agent and then I connect the MCU, but I got the same behaviour
* If I try rmw_uros_ping_agent(100,1) function, code hangs there with no log in agent. If I open minicom I see MCU sending data.

Acuadros95, 2022-06-13:
Could you share your code, so we can replicate on our side?

leoborgnino, 2022-06-13:
> Could you share your code, so we can replicate on our side?

Hi Antonio,

The code is in: https://github.com/leoborgnino/microros-stm32f407

Steps (I think you guys know this but can be an error for my side in the steps):

1. clone the repo 
2. in repo directory: git clone https://github.com/micro-ROS/micro_ros_stm32cubemx_utils
3.  docker pull microros/micro_ros_static_library_builder:foxy && docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library microros/micro_ros_static_library_builder:foxy
4. make in root folder generates .elf
5. I use CubeIDE to import elf and debug
6. docker run -it --rm --net=host -v /dev:/dev --privileged microros/micro-ros-agent:foxy serial --dev /dev/ttyUSB0 -b 115200 -v6

Thanks for the help!

leoborgnino, 2022-06-19:
> Hi, I have just tested the DMA transport on a F746 and its looking good.
> 
> Some ideas based on your instructions:
> 
> * You are missing this step `Set the DMA priotity to Very High for Tx and Rx`
> * Watch-out for branches, you are using foxy for the builder and agent, but the main branch of `micro_ros_stm32cubemx_utils` is humble
> * Did you try the regular IT transport? To discard problems on the Agent side.
> * Also try with the repo [sample_main](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/humble/sample_main.c) app.
> * This PR should help with the `session re-established` problem, rebuild the library once its merged:
>   [Fix wait_session_status listen timeout eProsima/Micro-XRCE-DDS-Client#322](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/322)
> 
> Please check this points and give feedback. If the problem persist, give info about the stm Firmware Version you are using. Example (Project Manager tab on CubeIDE): [image]

Hi Antonio, I'm off on holidays until 12 Jul. I will test that and write you back, thanks!

Acuadros95, 2022-07-01:
Closing, reopen if the problem persist when you come back!

closed 2022-07-01 by Acuadros95

shreyavsanoj7, 2024-08-14:
I hv the same issue any idea how to fix it?

leoborgnino, 2024-08-14:
> I hv the same issue any idea how to fix it?

I reinstalled everything in another machine, and it worked, so I’m not really sure what the issue was.

shreyavsanoj7, 2024-08-25:
it was working previously for me. Then after reinstallation of ubuntu it stopped working..