micro-ROS/micro_ros_stm32cubemx_utils #71: Large messages get dropped with best effort publisher.

roni-kreinin, 2022-06-21:
Setup: micro-ROS Foxy running on a STM32F407 with a custom USB serial transport.

When I publish a large message such as `sensor_msgs/Imu` with a best effort publisher I notice that a percentage of the messages get dropped on the ROS2 side. For example I publish the IMU message at 50hz and I run `ros2 topic hz /imu` I will see the topic at about 35-40hz. If I increase publish rate to 100hz I will see the topic at about 70hz. 

I created a custom IMU message that only publishes the data and not the covariance arrays and that message publishes consistently at 50 or 100hz. 

Is there a message size limit for the best effort publisher? Or is there some configuration I can change to publish these larger messages?

pablogs9, 2022-06-22:
It probably will be a matter of bandwidth in your serial port and/or your application.

Could you share code for replicating this issue?

roni-kreinin, 2022-06-22:
I can't share the application code but here is how I set up the transport.

I configure the STM32 as a Full Speed OTG device using the CubeMx configurator like this: 
[image]

I then created a `usb_transport.c` file and wrote the open/close/read/write transport functions. This is the write function:
```
size_t usb_transport_write(
  struct uxrCustomTransport * transport, uint8_t * buf, size_t len,
  uint8_t * err)
{
  USBD_StatusTypeDef ret = CDC_Transmit_FS(buf, len);

  return ret == USBD_OK ? len : 0;
}
```

The read and write buffers are both 2048 bytes in size.

roni-kreinin, 2022-06-22:
I checked the bandwidth over the usb connection and its only using about 25 kb/s so I don't think that is the issue.

pablogs9, 2022-06-22:
We need a replicable environment in order to address this issue. Please provide a mock application that behaves like yours and also if you have developed custom transports provide them.

roni-kreinin, 2022-06-22:
The issue was with the CDC drivers. The `CDC_Transmit_FS` call would just drop the write if the Tx was already busy. I guess when writing larger packets of data it would be busy more often.

I added a simple loop to wait for Tx to not be busy before calling `CDC_Transmit_FS` and that fixed the problem.

closed 2022-06-22 by roni-kreinin