micro-ROS/micro_ros_stm32cubemx_utils #73: No service response is received by the caller

MariusGhica, 2022-06-30:
## Issue template:

- Hardware description: STM32G491CCUx
- RTOS: bare metal
- Installation type: STM32CubeMX with Makefile, micro_ros_stm32cubemx_utils
- Version or commit hash: galactic

### Steps to reproduce the issue
- Follow steps mentioned in [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils) using docker (ros:galactic)
- Build firmware and flash through ST-Link V2
- Create and run the agent with ros2 run micro_ros_agent micro_ros_agent serial -b 115200 --dev /dev/ttyUSB0 -v6
- Service call on a new terminal: ros2 service call /service test_msgs/srv/BasicTypes "{int16_value: 10}"

### Expected behavior
Increment by one

### Actual behavior
1. Server receives request but there is no response.
`waiting for service to become available...`
`requester: making request: test_msgs.srv.BasicTypes_Request(bool_value=False, byte_value=b'\x00', char_value=0, float32_value=0.0, float64_value=0.0, int8_value=0, uint8_value=0, int16_value=10, uint16_value=0, int32_value=0, uint32_value=0, int64_value=0, uint64_value=0, string_value='')`
2. Received request is always -1.
`Service request value: -1.`

### Additional information
I tried the following:
1. export RMW_IMPLEMENTATION=rmw_fastrtps_cpp before starting the agent, as suggested in https://github.com/micro-ROS/micro_ros_setup/issues/310 thread.
2. Making the [response and request variables](https://github.com/micro-ROS/freertos_apps/blob/galactic/apps/add_two_ints_service/app.c#L47) global suggested in https://github.com/micro-ROS/micro_ros_setup/issues/316 thread.

Logs from agent:
```Bash
[1656601276.917824] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1656601276.917954] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1656601280.139215] info     | Root.cpp           | create_client            | create                 | client_key: 0x5851F42D, session_id: 0x81
[1656601280.139322] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x5851F42D, address: 0
[1656601280.139491] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1656601280.152047] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 60, data: 
0000: 81 80 00 00 01 07 32 00 00 0A 00 01 01 03 00 00 23 00 00 00 00 01 C0 00 1B 00 00 00 64 65 63 72
0020: 65 61 73 65 5F 62 79 6F 6E 65 5F 63 6C 69 65 6E 74 5F 72 63 6C 63 00 00 00 00 00 00
[1656601280.157536] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x5851F42D, participant_id: 0x000(1)
[1656601280.157617] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1656601280.157640] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1656601280.165676] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1656601280.185523] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 180, data: 
0000: 81 80 01 00 01 07 AA 00 00 0B 00 08 08 03 00 00 9C 00 00 00 09 00 00 00 2F 73 65 72 76 69 63 65
0020: 00 BB 01 20 2A 00 00 00 74 65 73 74 5F 6D 73 67 73 3A 3A 73 72 76 3A 3A 64 64 73 5F 3A 3A 42 61
0040: 73 69 63 54 79 70 65 73 5F 52 65 71 75 65 73 74 5F 00 00 00 2B 00 00 00 74 65 73 74 5F 6D 73 67
0060: 73 3A 3A 73 72 76 3A 3A 64 64 73 5F 3A 3A 42 61 73 69 63 54 79 70 65 73 5F 52 65 73 70 6F 6E 73
0080: 65 5F 00 01 12 00 00 00 72 71 2F 73 65 72 76 69 63 65 52 65 71 75 65 73 74 00 01 20 10 00 00 00
00A0: 72 72 2F 73 65 72 76 69 63 65 52 65 70 6C 79 00 00 01 00 00
[1656601280.186195] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x5851F42D, requester_id: 0x000(7), participant_id: 0x000(1)
[1656601280.186241] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 08 00 00
[1656601280.186263] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1656601280.195667] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1656601280.198706] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 24, data: 
0000: 81 80 02 00 08 01 10 00 00 0C 00 08 80 00 00 01 FF FF 00 00 00 00 00 00
[1656601280.198846] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1656601288.898058] debug    | Replier.cpp        | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 96, data: 
0000: CD 55 10 01 D2 0D A2 A8 4A 22 9B 62 00 00 14 03 00 00 00 00 01 00 00 00 7B 21 99 14 99 9F 11 9B
0020: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0A 00 00 00 00 00
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00
[1656601288.898231] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 108, data: 
0000: 81 80 02 00 09 01 64 00 00 0C 00 08 CD 55 10 01 D2 0D A2 A8 4A 22 9B 62 00 00 14 03 00 00 00 00
0020: 01 00 00 00 7B 21 99 14 99 9F 11 9B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040: 00 00 00 00 00 00 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060: 00 00 00 00 01 00 00 00 00 00 00 00
[1656601288.913482] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1656601288.924850] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 92, data: 
0000: 81 80 03 00 07 01 51 00 00 0D 00 08 CD 55 10 01 D2 0D A2 A8 4A 22 9B 62 00 00 14 03 00 00 00 00
0020: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF 00 00 00 00 00 00 00 00
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1656601288.925025] debug    | Replier.cpp        | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 77, data: 
0000: CD 55 10 01 D2 0D A2 A8 4A 22 9B 62 00 00 14 03 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00
[1656601288.925067] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
```

[microros_projectMain.txt](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/9021254/microros_projectMain.txt)

pablogs9, 2022-07-01:
Please share your micro-ROS client code

pablogs9, 2022-07-01:
The micro-ROS client is answering to the service request. Could you make sure that you switch the middleware to Fast DDS with `RMW_IMPLEMENTATION=rmw_fastrtps_cpp` in the console where you are running your `ros2 service call...` ?

MariusGhica, 2022-07-01:
It did indeed work. Many thanks for the advice.

closed 2022-07-01 by MariusGhica