micro-ROS/micro_ros_stm32cubemx_utils #79: How to increase a published topic frame rate faster than best effort publisher

Puyu2934, 2022-08-08:


pablogs9, 2022-08-08:
Which transport are you using https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/tree/humble/extra_sources/microros_transports ?

Puyu2934, 2022-08-08:
dma_transport
https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/humble/extra_sources/microros_transports/dma_transport.c

pablogs9, 2022-08-08:
Can you try removing this lines: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/58b4c8fe26c5cd695b7a33120c754a8a78d71300/extra_sources/microros_transports/dma_transport.c#L39-L41

I guess that maybe this transport is not optimized to throughput since it is setting a DMA transfer and then is waiting until it is completed:

```c
HAL_StatusTypeDef ret;
    if (uart->gState == HAL_UART_STATE_READY){  // CHECK IF READY
        ret = HAL_UART_Transmit_DMA(uart, buf, len); // MAKE DMA TX

// PROBABLY NOT REQUIRED:
//        while (ret == HAL_OK && uart->gState != HAL_UART_STATE_READY){
//            osDelay(1);
//        }

        return (ret == HAL_OK) ? len : 0;
    }else{
        return 0;
    }
```

Puyu2934, 2022-08-08:
Yes, I tried to remove the while:
```
size_t cubemx_transport_write(struct uxrCustomTransport* transport, const uint8_t *buf, size_t len, uint8_t * err){
    UART_HandleTypeDef * uart = (UART_HandleTypeDef*) transport->args;

    HAL_StatusTypeDef ret;
    if (uart->gState == HAL_UART_STATE_READY){
        ret = HAL_UART_Transmit_DMA(uart, buf, len);
//        while (ret == HAL_OK && uart->gState != HAL_UART_STATE_READY){
//            osDelay(1);
//        }
        return (ret == HAL_OK) ? len : 0;
    }else{
        return 0;
    }
}
```

But if remove that, the program will stuck at :
```
    // create init_options
    RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
```

This is the agent output with flag `-v6`:
```
[1659947027.821985] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1659947027.822239] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1659947031.371890] info     | Root.cpp           | create_client            | create                 | client_key: 0x5851F42D, session_id: 0x81
[1659947031.372040] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x5851F42D, address: 0
[1659947031.372234] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1659947031.483376] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947031.483834] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947031.595445] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947031.595755] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947031.707365] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947031.707663] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947031.819310] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947031.819599] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947031.931195] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947031.931346] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947032.043333] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947032.043638] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947032.155342] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947032.155637] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947032.267143] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947032.267469] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[1659947032.379204] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1659947032.379510] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80

```

Interactics, 2022-08-08:
I've experienced something similar to the problem you've experienced.

In my case, it was to raise the IMU Topic frequency by 250Hz.
I checked the IMU Topic publishing function's time with oscilloscope and concluded that the IMU message type is too large to publish more fast.
So I created a smaller form of message(removing msg's headers), and this solved the problem.

Here is my issue.
[How can I increase the publishing hertz?](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/72)

Puyu2934, 2022-08-08:
> I've experienced something similar to the problem you've experienced.
>
> In my case, it was to raise the IMU Topic frequency by 250Hz. I checked the IMU Topic publishing function's time with [...]

Thank you for your advice.

and I have tried to use a lighter msg in [this reply](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/79#issuecomment-1207692977)

And the rate still unsatisfying. I wondering what makes my device so slow :sob:

DrMarkusKrug, 2023-08-02:
> Yes, I tried to remove the while:
>
> ```
> size_t cubemx_transport_write(struct uxrCustomTransport* transport, const uint8_t *buf, size_t len, uint8_t * err){
>     UART_HandleTypeDef * uart = [...]

Well, deleting that lines is really very dangerous because you risk that successive DMA calls run into each other. I fixed that problem by introducing an additional Semaphore.
So my transport write looks like this (kept the original lines and just comment them out):
size_t cubemx_transport_write(struct uxrCustomTransport* transport, uint8_t * buf, size_t len, uint8_t * err){
    UART_HandleTypeDef * uart = (UART_HandleTypeDef*) transport->args;

    HAL_StatusTypeDef ret;
    //if (uart->gState == HAL_UART_STATE_READY){
    osSemaphoreAcquire(UART_Tx_readyHandle, osWaitForever);
        ret = HAL_UART_Transmit_DMA(uart, buf, len);
        __HAL_DMA_DISABLE_IT(uart->hdmatx, DMA_IT_HT );
        osSemaphoreAcquire(UART_Tx_readyHandle, osWaitForever);
        osSemaphoreRelease(UART_Tx_readyHandle);
        //while (ret == HAL_OK && uart->gState != HAL_UART_STATE_READY){
        //    osDelay(1);
     //   }

        return (ret == HAL_OK) ? len : 0;
    //}else{
    //    return 0;
    //}
}
For my application that works fine and improves the throughput. However, I have not such high demands as you have. Additionally I made a minor change for the read function by inserting a break in the first loop. This helps avoiding to wait one millisecond even if the message already arrived. It's a small chance of corrupting the message because receiving something doesn't say you received the complete message. So please be aware of this. In my application if works because I run the serial line with 921600 baud and only receive short messages.

size_t cubemx_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err){
    UART_HandleTypeDef * uart = (UART_HandleTypeDef*) transport->args;

    int ms_used = 0;
    do
    {
        __disable_irq();
        dma_tail = UART_DMA_BUFFER_SIZE - __HAL_DMA_GET_COUNTER(uart->hdmarx);
        __enable_irq();
        ms_used++;
        if(dma_head != dma_tail)
        	break;
        osDelay(portTICK_RATE_MS);
    } while (dma_head == dma_tail && ms_used < timeout);
    
    size_t wrote = 0;
    while ((dma_head != dma_tail) && (wrote < len)){
        buf[wrote] = dma_buffer[dma_head];
        dma_head = (dma_head + 1) % UART_DMA_BUFFER_SIZE;
        wrote++;
    }
    
    return wrote;
}