micro-ROS/micro_ros_stm32cubemx_utils #80: Connection disconnects after a random period when publishing from agent to client

Bernard-Lichaa-CC, 2022-08-08:
Hello,

So I setup the micro-ros agent on my Nvidia Jetson running ubuntu 20 to communicate with micro-ros on stm32f767ZI FreeRTOS mounted on a nucleo board.

I was able to open the connection between them and successfully publishing and subscribing to different topics. However, I noticed a strange behavior. Whenever I start publishing from my agent (best effort), the subscriber on the client stops the callback after a random period of time and won't continue reading the data sent from the agent, but the publisher on the client continues publishing to the topic and I am able to see the published data on the agent.

This behavior is consistent. However, the period of time that takes the subscriber to stop the callback is random. It can stop in a couple of seconds or in a matter of minutes.

All the publishers and subscribers are using best effort reliability. I noticed that the rmw_uros_ping_agent returns an error when the subscriber stops. To avoid this issue, I followed the ping pong example where I ping the agent and spin the executor some period of time only when the agent is connected and destroy then reinitialize everything whenever it is disconnected.

However, a new issue comes up that I think is related to the first issue. Whenever I start publishing from the agent, the agent disconnects after a certain period of time and instead of reinitializing everything instantly whenever it disconnect, the client is pinging the agent for a certain period of time which is also random (between 1s to 20s) before receiving RMW_RET_OK and reinitializing.

Does anyone know from where this issue might come? I am ready to share any file requested.

Thank you!!!!

pablogs9, 2022-08-08:
Provide a replicable environment and replication instructions.

pablogs9, 2022-09-19:
Any update on this?

Bernard-Lichaa-CC, 2022-09-19:
Hello,
Sorry for not updating you. We couldn't find a solution so we will not use micro ros in our project. I will close this issue.

closed 2022-09-19 by pablogs9