micro-ROS/micro_ros_stm32cubemx_utils #84: micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include not found.

tianrking, 2023-01-04:
Thanks to the developers for providing such a project.  

But I am in some trouble . It seems to be possible to install the dependencies in this script, but the readme doesn't explain how to do it, and when I followed the steps, I found that the micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include not found.

Can someone please give me more detailed instructions on how to do this?

pablogs9, 2023-01-04:
Could you please explain which is your error and how to replicate it with enough information?

tianrking, 2023-01-11:
```bash
micro_ros_stm32cubemx_utils/extra_sources/microros_time.c: In function 'clock_gettime':
micro_ros_stm32cubemx_utils/extra_sources/microros_time.c:32:5: error: unknown type name 'TimeOut_t'
   32 |     TimeOut_t xCurrentTime = { 0 };
      |     ^~~~~~~~~
micro_ros_stm32cubemx_utils/extra_sources/microros_time.c:44:5: warning: implicit declaration of function 'vTaskSetTimeOutState' [-Wimplicit-function-declaration]
   44 |     vTaskSetTimeOutState( &xCurrentTime );
      |     ^~~~~~~~~~~~~~~~~~~~
micro_ros_stm32cubemx_utils/extra_sources/microros_time.c:48:47: error: request for member 'xOverflowCount' in something not a structure or union
   48 |     ullTickCount = ( uint64_t ) ( xCurrentTime.xOverflowCount ) << ( sizeof( TickType_t ) * 8 );
      |                                               ^
micro_ros_stm32cubemx_utils/extra_sources/microros_time.c:51:33: error: request for member 'xTimeOnEntering' in something not a structure or union
   51 |     ullTickCount += xCurrentTime.xTimeOnEntering;
      |                                 ^
make: *** [Makefile:208: build/microros_time.o] Error 1

```

tianrking, 2023-01-11:
on stm32f103c8t6  with freertos enable on stm32cubemx interface cmsis_v2

pablogs9, 2023-01-11:
It seems to be related to your include path, is FreeRTOS correctly being included in the included path?

prash85, 2023-04-17:
Same here. The "Using this package with STM32CubeIDE" guide suggests adding "libmicroros/include" folder to build settings but this folder is missing in the repo.

closed 2023-07-16 by tianrking