micro-ROS/micro_ros_stm32cubemx_utils #85: rclc_parameter_server_init_default returns RCL_RET_ERROR

ALUIS97, 2023-01-11:
[image]

I am trying to set a parameter server with a nucleo stm32f4 board , I already changed the values in colcon.meta (as said here https://micro.ros.org/docs/tutorials/programming_rcl_rclc/parameters/#callback) but the function is returning an error, although my program can continue rc =  rclc_executor_spin(&executor); also returns RCL_RET_ERROR and goes out and I can not see the parameters of my micro ros app from the command ros2 param list .
I am using the FOXY distribution

pablogs9, 2023-01-11:
Could you share your `colcon.meta` and `rmw_microxrcedds/config.h`?

ALUIS97, 2023-01-11:
[config_h.txt](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/10393818/config_h.txt)
[colcon_meta.txt](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/10393856/colcon_meta.txt)
By the way I set up my environment with static library IDE and DMA transport for serial comunication

pablogs9, 2023-01-12:
Could you share the micro-ROS Agent output with the flag `-v6`?

ALUIS97, 2023-01-12:
[micro_ros_agent_v6.txt](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/10404140/micro_ros_agent_v6.txt)
here it is

pablogs9, 2023-01-13:
An finally, can you share you micro-ROS client code?

ALUIS97, 2023-01-14:
[micro_ros_client.txt](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/files/10418391/micro_ros_client.txt)

zijian-x, 2024-11-01:
Is there a fix/update on this? I'm having this issue as well trying to add a parameter server on the f767zi nucleo board.

zijian-x, 2024-11-01:
```
[1] ⤷ cat micro_ros_stm32cubemx_utils/microros_static_library/library_generation/colcon.meta
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=8",
                "-DRMW_UXRCE_MAX_SERVICES=5",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}

[1] ⤷ cat micro_ros_stm32cubemx_utils/microros_static_library/libmicroros/microros_include/rmw_microxrcedds_c/config.h
#ifndef RMW_MICROXRCEDDS_CONFIG_H
#define RMW_MICROXRCEDDS_CONFIG_H

#include <uxr/client/config.h>

/* #undef RMW_UXRCE_TRANSPORT_UDP */
/* #undef RMW_UXRCE_TRANSPORT_TCP */
/* #undef RMW_UXRCE_TRANSPORT_SERIAL */
#define RMW_UXRCE_TRANSPORT_CUSTOM
/* #undef RMW_UXRCE_TRANSPORT_IPV4 */
/* #undef RMW_UXRCE_TRANSPORT_IPV6 */
/* #undef RMW_UXRCE_USE_REFS */
/* #undef RMW_UXRCE_ALLOW_DYNAMIC_ALLOCATIONS */
[... 89 lines cut ...]
                                cubemx_transport_read);

  rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();
  freeRTOS_allocator.allocate = microros_allocate;
  freeRTOS_allocator.deallocate = microros_deallocate;
  freeRTOS_allocator.reallocate = microros_reallocate;
  freeRTOS_allocator.zero_allocate = microros_zero_allocate;

  if (!rcutils_set_default_allocator(&freeRTOS_allocator))
    printf("Error on default allocators (line %d)\n", __LINE__);

  allocator = rcl_get_default_allocator();
  rclc_support_init(&support, 0, NULL, &allocator);
  rclc_node_init_default(&node, "micro_ros_node", "", &support);
}

```

static library is compiled via `docker` in the newest `jaazy` branch with tag `5.0.1`

`RCL_RET_ERROR` at:
```
rclc_parameter_server_t param_server;
rclc_parameter_server_init_default(&param_server, node);
```