micro-ROS/micro_ros_stm32cubemx_utils #87: [Int32MultiArray]: Help with Publisher

MingshanHe, 2023-03-01:
I have coded for msg initialization and publisher but it does not work. Can you help me with this issue?

`  rcl_publisher_t publisher;
  std_msgs__msg__Int32 msg;
  std_msgs__msg__Int32MultiArray pressure_msg;
  pressure_msg.data.capacity = 100; 
  pressure_msg.data.size = 0;
  pressure_msg.data.data = (int64_t*) malloc(pressure_msg.data.capacity * sizeof(int64_t));
  pressure_msg.layout.dim.capacity = 100;
  pressure_msg.layout.dim.size = 0;
  pressure_msg.layout.dim.data = (std_msgs__msg__MultiArrayDimension*) malloc(pressure_msg.layout.dim.capacity * sizeof(std_msgs__msg__MultiArrayDimension));
  for(size_t i = 0; i < pressure_msg.layout.dim.capacity; i++){
    pressure_msg.layout.dim.data[i].label.capacity = 20;
    pressure_msg.layout.dim.data[i].label.size = 0;
    pressure_msg.layout.dim.data[i].label.data = (char*) malloc(pressure_msg.layout.dim.data[i].label.capacity * sizeof(char));
  }
  rclc_support_t support;
  rcl_allocator_t allocator;
  rcl_node_t node;

  allocator = rcl_get_default_allocator();

  //create init_options
  rclc_support_init(&support, 0, NULL, &allocator);

  // create node
  rclc_node_init_default(&node, "cubemx_node", "", &support);

  // create publisher
//  rclc_publisher_init_default(
//    &publisher,
//    &node,
//    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
//    "cubemx_publisher");

   rclc_publisher_init_default(
     &publisher,
     &node,
	 ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32MultiArray),
     "pressure");

  msg.data = 0;

  for(;;)
  {
//    rcl_ret_t ret = rcl_publish(&publisher, &msg, NULL);
    rcl_ret_t ret = rcl_publish(&publisher, &pressure_msg, NULL);
    if (ret != RCL_RET_OK)
    {
      printf("Error publishing (line %d)\n", __LINE__); 
    }
    
    msg.data++;
    osDelay(10);
  }`

pablogs9, 2023-03-01:
- pressure_msg.data.data is a int32_t, you are allocating int64_t
- why are you doing `msg.data = 0;`?

Which error do you have?

MingshanHe, 2023-03-01:
First, I am following the tutorial on micro-ros. Yes, the allocating size needs to be changed to `int64_t`, have you some reference code on int32_multiarray to the publisher?
the `msg.data=0` is the sample code from sample_main.c

pablogs9, 2023-03-01:
Which example in micro-ROS uses `std_msgs__msg__Int32MultiArray`?

MingshanHe, 2023-03-01:
Oh, sorry, I have some mistakes to describe my mind. You are right, the int64_t needs to be int32_t

MingshanHe, 2023-03-01:
I have no reference example in micro-ROS use `std_msgs__msg__Int32MultiArray`

closed 2023-03-08 by MingshanHe