micro-ROS/micro_ros_stm32cubemx_utils #97: Need to create custom  massage/service

hishambaharom, 2023-04-14:
I would like to create custom massage/service how I can make it, what the step require.

pablogs9, 2023-04-14:
Please read the documentation: 

https://github.com/micro-ROS/micro_ros_espidf_component
https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/custom_types/custom_types.html

closed 2023-04-14 by pablogs9

hishambaharom, 2023-04-14:
Still don't get it,   I have create the custom service in my uros_ws  all ok but, what folder I need to copy to microros_static_library/library_generation/extra_packages/ so it can call when I build using stm32cube ide

pablogs9, 2023-04-17:
`ros2 pkg create --build-type ament_cmake my_custom_message` creates a new ROS 2 package where you will define your new message. This is the folder that you need to place in `extra_packages`