micro-ROS/micro_ros_stm32cubemx_utils #99: HardFault_Handler when rclc_timer_init_default is use

hishambaharom, 2023-04-20:
I have a problem when debugger reach rclc_timer_init_default function. this cause the debugger go to HardFault_Handler function. Please help.
Board: openCR(stm32f746)
IDE: stm32cubeide
OS: Freertos

pablogs9, 2023-04-20:
We need more details to be able to evaluate this issue.

hishambaharom, 2023-04-20:
This the configuration for rtos
file:///home/shahrol/Pictures/Screenshots/Screenshot%20from%202023-04-20%2022-38-31.png

This what I capture in disassembly.

file:///home/shahrol/Pictures/Screenshots/Screenshot%20from%202023-04-20%2023-34-22.png

the problem is when it execute address 0x8017de4 ( bx lr)

pablogs9, 2023-04-20:
Do you have any heap initialized?

hishambaharom, 2023-04-21:
No I did not, how I can initialized heap any ref..

pablogs9, 2023-04-21:
In Cube IDE you need to have some heap configured for your platform. This is not related to micro-ROS but to your platform usage.

hishambaharom, 2023-04-21:
I Have configure the heap to 10k  but the problem still have

hishambaharom, 2023-04-23:
What other data require for evaluate this issue..