micro-ROS/micro_ros_stm32cubemx_utils #100: micro-ros error  [ros2run]: Process exited with failure 7

WhisterKuper, 2023-05-28:
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library microros/micro_ros_static_library_builder:humble
dos2unix: converting file /project/micro_ros_stm32cubemx_utils/microros_static_library/library_generation/library_generation.sh to Unix format...
Get:1 http://packages.ros.org/ros2/ubuntu jammy InRelease [4682 B]
Get:2 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]      
Get:3 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Get:4 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages [1201 kB]
Get:5 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [360 kB]           
Get:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]                         
Get:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [108 kB]                       
Get:8 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [36.3 kB]
Get:9 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [916 kB]
Get:10 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
Get:11 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
Get:12 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [509 kB]      
Get:13 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]         
Get:14 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]                                                                                                                          
Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [360 kB]                                                                                                                  
Get:16 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [788 kB]                                                                                                                        
Get:17 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [42.2 kB]                                                                                                                 
Get:18 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [1154 kB]                                                                                                                   
Get:19 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [27.0 kB]                                                                                                                 
Get:20 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [49.4 kB]                                                                                                                     
Fetched 25.7 MB in 10s (2628 kB/s)                                                                                                                                                                        
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
33 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: http://packages.ros.org/ros2/ubuntu/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  binutils-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
Suggested packages:
  libnewlib-doc
The following NEW packages will be installed:
  binutils-arm-none-eabi gcc-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
0 upgraded, 6 newly installed, 0 to remove and 33 not upgraded.
Need to get 442 MB of archives.
After this operation, 2575 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu jammy/universe amd64 binutils-arm-none-eabi amd64 2.38-3ubuntu1+15build1 [3040 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy/universe amd64 gcc-arm-none-eabi amd64 15:10.3-2021.07-4 [47.7 MB]
Get:3 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-dev all 15:10.3-2021.07-4+17 [1049 kB]                                                                                
Get:4 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-dev all 3.3.0-1.3 [137 kB]                                                                                                          
Get:5 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-arm-none-eabi all 3.3.0-1.3 [43.2 MB]                                                                                               
Get:6 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-newlib all 15:10.3-2021.07-4+17 [347 MB]                                                                              
Fetched 442 MB in 1min 49s (4050 kB/s)                                                                                                                                                                    
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package binutils-arm-none-eabi.
(Reading database ... 61217 files and directories currently installed.)
Preparing to unpack .../0-binutils-arm-none-eabi_2.38-3ubuntu1+15build1_amd64.deb ...
Unpacking binutils-arm-none-eabi (2.38-3ubuntu1+15build1) ...
Selecting previously unselected package gcc-arm-none-eabi.
Preparing to unpack .../1-gcc-arm-none-eabi_15%3a10.3-2021.07-4_amd64.deb ...
Unpacking gcc-arm-none-eabi (15:10.3-2021.07-4) ...
Selecting previously unselected package libstdc++-arm-none-eabi-dev.
Preparing to unpack .../2-libstdc++-arm-none-eabi-dev_15%3a10.3-2021.07-4+17_all.deb ...
Unpacking libstdc++-arm-none-eabi-dev (15:10.3-2021.07-4+17) ...
Selecting previously unselected package libnewlib-dev.
Preparing to unpack .../3-libnewlib-dev_3.3.0-1.3_all.deb ...
Unpacking libnewlib-dev (3.3.0-1.3) ...
Selecting previously unselected package libnewlib-arm-none-eabi.
Preparing to unpack .../4-libnewlib-arm-none-eabi_3.3.0-1.3_all.deb ...
Unpacking libnewlib-arm-none-eabi (3.3.0-1.3) ...
Selecting previously unselected package libstdc++-arm-none-eabi-newlib.
Preparing to unpack .../5-libstdc++-arm-none-eabi-newlib_15%3a10.3-2021.07-4+17_all.deb ...
Unpacking libstdc++-arm-none-eabi-newlib (15:10.3-2021.07-4+17) ...
Setting up binutils-arm-none-eabi (2.38-3ubuntu1+15build1) ...
Setting up gcc-arm-none-eabi (15:10.3-2021.07-4) ...
Setting up libnewlib-dev (3.3.0-1.3) ...
Setting up libnewlib-arm-none-eabi (3.3.0-1.3) ...
Setting up libstdc++-arm-none-eabi-dev (15:10.3-2021.07-4+17) ...
Setting up libstdc++-arm-none-eabi-newlib (15:10.3-2021.07-4+17) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
Creating firmware for generate_lib platform generic
[ros2run]: Process exited with failure 7
[ros2run]: Process exited with failure 7

rajesh3699, 2023-06-13:
Hey, did anyone find solution to this?

WhisterKuper, 2023-06-19:
> 
I found the same result on my desktop computer, but switching to a laptop seems to solve the problem, and virtual machines are also unable to solve it！

Pansamic, 2023-09-02:
> > 
> 
> I found the same result on my desktop computer, but switching to a laptop seems to solve the problem, and virtual machines are also unable to solve it！

Distribution doesn't match local ROS2 distribution.

I've met the same error as your's. cuz the ROS2 distribution on your desktop doesn't match the docker building command "docker run ..... :iron". If you want to use this command, you are required to install ROS2:iron on your PC. If installed other distribution like ROS2:Humble, you are suggested to use "docker run -it --rm -v $(pwd):project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library microros/micro_ros_static_library_builder:humble". notice that i've changed distribution.

Pansamic, 2023-10-15:
# Reason
Network problem.

micro_ros_setup need git to build firmware but if you get some problem with network, git will report GNUTLS error. that's git uses GnuTLS and it may cause some connection error. So we've better build git with OpenSSL from source.

# Solution

execute following command on your PC, under your STM32CubeIDE project folder.

```bash
# pull docker container to local host
docker pull microros/micro_ros_static_library_builder:humble

# start container and enter terminal of container environment
docker run -it -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide --entrypoint /bin/bash microros/micro_ros_static_library_builder:humble
```

do the following in docker container:

```bash
wget https://mirrors.edge.kernel.org/pub/software/scm/git/git-2.17.0.tar.gz
sudo apt update
sudo apt install autoconf build-essential tcl-dev libssl-dev libcurl4-openssl-dev gettext
tar -xf git-2.17.0.tar.gz
cd git-2.17.0
make configure

./configure --prefix=/usr/  
make -j8
make install
git version # you will see current git version 2.17.0

bash /project/MICROROS_LIBRARY_FOLDER/library_generation/library_generation.sh
```
if everything OK, the following picture show correct building process.
[image]

building success like this:
[image]