micro-ROS/micro_ros_stm32cubemx_utils #104: Issue with Pre-Build step : can't find the specified path

rajesh3699, 2023-06-13:
I am trying to replicate the same thing:
System: Windows 11 , CubeIde (latest).

I also have a ubuntu docker running.

error description: 

10:46:58 **** Incremental Build of configuration Debug for project micro_ROS_test ****
make -j12 all 
./micro_ros_stm32cubemx_utils/library_generation//library_generation.sh
make (e=3): The system cannot find the path specified.

make[1]: *** [makefile:89: pre-build] Error 3
make: *** [makefile:62: all] Error 2
"make -j12 all" terminated with exit code 2. Build might be incomplete.

10:46:59 Build Failed. 2 errors, 0 warnings. (took 732ms)

pablogs9, 2023-06-14:
This repo is intended to be used under Linux, your problem is that `make` is not installed.

rajesh3699, 2023-06-14:
Is there a way to make in run in windows or through docker image of ubuntu?

rajesh3699, 2023-06-14:
is there a way to implement this with docker image of ubuntu?

pablogs9, 2023-06-14:
A better option is to prepare a Virtual Box machine since this tool uses docker, and it is quite hard to run docker inside docker.

In any case, you can use this tutorial to build your custom static libraries and link them directly to your STM32 project:

https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/

elgarbe, 2023-11-01:
Hi, you can use stm32cubemxide in windows 11. You will need to have docker desktop running first.
I have problem with the prebuild step, but I manged to run those lines outside the cubeIDE and everithing works ok