micro-ROS/micro_ros_stm32cubemx_utils #107: not create rclc_support_init(&support, 0, NULL, &allocator);

guillehb21, 2023-07-14:
Hello there , I have a problem when creating nodes, they are not being created or I don't know what could be happening. It's the first time I'm using ROS2 on a microcontroller. I tried using debugging to find the issue. In debug mode, the program runs fine until this line: 
[image]
. If I click on the next step, it jumps to a file called "task" and goes into a loop. 
[image]
I don't know what might be happening. My "agente_ws" looks like this: 
[image]
. The code is the same as the example, and I don't know what could be happening. 
Thank you very much in advance. I'm using ROS2 Iron, and the board is an STM32 Nucleo H743ZI2.

DrMarkusKrug, 2023-08-01:
Hi,
I have not so much experience but certainly the rclc_support_init() function seems to have serious problems at least on STM32 devices.
Is the function called inside a task? If so, make sure you have at least 3000 words heap for this task.
I assume that looping in prvCheckTasksWaitingTermination() is caused because you rclc_support_init() function does not run as expected and hangs in an endless loop. Your FreeRTOS takes over and is continously updating all other resources. So that is nothing special from my point of view.

Best Regards
Markus

toffanetto, 2024-09-30:
HI @guillehb21, I was having the same problem im my nucleo H753. In my case, I have configurated DMA as _circular_ for Tx and Rx, but Tx is configured as _normal_. Once i chanced DMA Tx mode for _normal_, everythin works!

DrMarkusKrug, 2024-10-01:
Hi toffanetto,
I haven't looked into this topic for a while. But if I remember correct the low level send/receive functions of microROS already implement a ringbuffer with an index for write and read to it. So if you add circular DMA, you very likely get in conflict with this ringbuffer concept. Therefore I'm not surprised to here that circular DMA is not working.
In the not so far future I need to come back to microROS and I will further investigate. However, my main concern is the memory management inside ROS2 that seems to be not appropriate to the limited ressources and OS functionality that we usually find for microcontrollers.
Regards
Markus