micro-ROS/micro_ros_stm32cubemx_utils #111: publishing 3rd message fails

DrMarkusKrug, 2023-08-03:
Hi,
I have setup an FreeRTOS application that is publishing 3 (multirarray) messages in 3 different tasks.
My micro ros agent is displaying all of the topics with 'ros2 topic list'. However one of them is not giving an echo. I can always echo only 2 out of the 3 and I can change the 2 in any combination (msg1+msg2, msg1+msg3 or msg2+msg3). 
The call rcl_publish() is returning with RCL_RET_OK for the call that leads to the missing echo.
I already changed in the file colcon-embeddedrtps.meta file the parameter DERTPS_MAX_PUBLISHERS=10 but with no success so far.
Any hint?

Best Regards
Markus

pablogs9, 2023-08-03:
micro-ROS by default are not thread-safe so, are you locking your micro-ROS calls between threads?

Also, could you share a micro-ROS Agent log output with the `-v6` flag enabled?

pablogs9, 2023-08-04:
Yes, it shall be enough to lock with a mutex/semaphore the access to rclc/rcl calls.

I'm closing since solved.

closed 2023-08-04 by pablogs9