micro-ROS/micro_ros_stm32cubemx_utils #112: undefined reference to `__sync_synchronize'

pdl0x0lbq, 2023-08-08:
Hi,

I Get this error:
opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.0.202305231506/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ros/CLionProjects/csmr/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `unlock_memory':
atomic_64bits.c:(.text+0xa0): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.0.202305231506/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ros/CLionProjects/csmr/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_load_8':
atomic_64bits.c:(.text+0x11c): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.0.202305231506/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ros/CLionProjects/csmr/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_store_8':
atomic_64bits.c:(.text+0x198): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.0.202305231506/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ros/CLionProjects/csmr/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_exchange_8':
atomic_64bits.c:(.text+0x214): undefined reference to `__sync_synchronize'
/opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.0.202305231506/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ros/CLionProjects/csmr/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcutils-atomic_64bits.c.obj): in function `__atomic_fetch_add_8':
atomic_64bits.c:(.text+0x2a8): undefined reference to `__sync_synchronize'
How to resolve this error

pablogs9, 2023-08-08:
You can disable building `atomic_64bits.c` file here: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library_ide/library_generation/colcon.meta#L26C19-L26C40

pdl0x0lbq, 2023-08-08:
> You can disable building `atomic_64bits.c` file here: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library_ide/library_generation/colcon.meta#L26C19-L26C40

thank you very much

pdl0x0lbq, 2023-08-08:
> You can disable building `atomic_64bits.c` file here: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library_ide/library_generation/colcon.meta#L26C19-L26C40

Another error occurred
`/opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.0.202305231506/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ros/CLionProjects/csmr/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/libmicroros.a(librcl-time.c.obj): in function `rcl_get_ros_time':
time.c:(.text.rcl_get_ros_time+0xa): undefined reference to `__atomic_load_8'`

pablogs9, 2023-08-08:
That is the point that you need to have 64 bits atomics in your platform to use micro-ROS. If your native platform does not implement them nor is able to build the provided ones, you need to provide an implementation of `__atomic_load_8`

daehwankim112, 2025-03-12:
I am also stuck here.

I've got to this point by changing floating-point ABI to Mix HW/SW implementation because I am receiving VFP register arguments errors. When I use Mix, VFP errors do not appear anymore but `atomic_64bits.c:(.text+0xa0): undefined reference to `__sync_synchronize'` error appears.

sunwen-me, 2025-03-12:
> I am also stuck here.
>
> I've got to this point by changing floating-point ABI to Mix HW/SW implementation because I am receiving VFP register arguments errors. When I use Mix, VFP errors do not [...]

I have already resolved this issue.

Before compiling the library
Modify toolchain.cmake in micro_ros_stm32cubemx_utols/microros_sttictlibrary/library-generation to add
`set(FLAGS "-O2 -ffunction-sections -fdata-sections -fno-exceptions -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -nostdlib -mthumb --param max-inline-insns-single=500 -DF_CPU=84000000L -D'RCUTILS_LOG_MIN_SEVERITY=RCUTILS_LOG_MIN_SEVERITY_NONE'" CACHE STRING "" FORCE)`
Pay attention to modifying the kernel model

daehwankim112, 2025-03-13:
Thank you soooo much @sunwen-me for the prompt response!!!

My development environment is
- Windows 11 Home
- STM32 Cube IDE
- ROS2 Humble

There 2 toolchain.cmake. One at `micro_ros_stm32cubemx_utils/microros_static_library/library_generation/extra_packages/toolchain.cmake` and `micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/extra_packages/toolchain.cmake`.

I have edited both of the cmake to include
`set(FLAGS "-O2 -ffunction-sections -fdata-sections -fno-exceptions -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -nostdlib -mthumb --param max-inline-insns-single=500 -DF_CPU=84000000L -D'RCUTILS_LOG_MIN_SEVERITY=RCUTILS_LOG_MIN_SEVERITY_NONE'" CACHE STRING "" FORCE)`.

This is my Project - > Setting -> C/C++ Build -> Setting -> MCU/MPU Setting.

[image]

However, I am receiving VFP register arguments errors.
One of the error message is:
`C:/ST/STM32CubeIDE_1.17.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.1.0.202410251130/tools/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: error: SeaPuppyV1.2.elf uses VFP register arguments, D:/Projects/2025/UVEEC/SeaPuppyV1.2/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros\libmicroros.a(librcl-publisher.c.obj) does not`

How should I fix this? What could have gone wrong? Thank you!

daehwankim112, 2025-03-13:
I fixed this issue by creating new project. I think the issue was I copied sample_main.c into main.c and compiled. However, this time, I am only copying 
```
/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <uxr/client/transport.h>
#include <rmw_microxrcedds_c/config.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int32.h>
/* USER CODE END Includes */
```
and
```
/* USER CODE BEGIN 4 */
bool cubemx_transport_open(struct uxrCustomTransport * transport);
bool cubemx_transport_close(struct uxrCustomTransport * transport);
size_t cubemx_transport_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err);
size_t cubemx_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);

void * microros_allocate(size_t size, void * state);
void microros_deallocate(void * pointer, void * state);
void * microros_reallocate(void * pointer, size_t size, void * state);
void * microros_zero_allocate(size_t number_of_elements, size_t size_of_element, void * state);
/* USER CODE END 4 */
```
from the sample_code.c and it compiles fine.