micro-ROS/micro_ros_stm32cubemx_utils #113: Static library build is missing some functions - ROS2 Humble

lycanthought, 2023-08-11:
After building the static library and then compiling I get the following errors:

`/home/dan/microROS/micro_ROS/Core/Src/freertos.c:137: undefined reference to `rmw_uros_set_custom_transport'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:145: undefined reference to `rcutils_get_zero_initialized_allocator'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:151: undefined reference to `rcutils_set_default_allocator'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:163: undefined reference to `rcutils_get_default_allocator'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:166: undefined reference to `rclc_support_init'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:169: undefined reference to `rclc_node_init_default'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:172: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:172: undefined reference to `rclc_publisher_init_default'
/home/dan/.config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /home/dan/microROS/micro_ROS/Core/Src/freertos.c:182: undefined reference to `rcl_publish'`

I believe I have followed all the build steps correctly, I have checked versions for each step

pablogs9, 2023-08-11:
Which version of the library are you using?

Could you paste here the micro-ROS library build output log?

lycanthought, 2023-08-11:
I'm using the humble version - If you need more detail than that I'm not sure how to find it....

This is the full build output:

docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library microros/micro_ros_static_library_builder:humble
dos2unix: converting file /project/micro_ros_stm32cubemx_utils/microros_static_library/library_generation/library_generation.sh to Unix format...
Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]                                                     
Get:3 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [834 kB]  
Get:4 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [44.0 kB]                                         
Get:5 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [979 kB]                                                     
Get:6 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [849 kB]                                                   
Get:7 http://packages.ros.org/ros2/ubuntu jammy InRelease [4682 B]                                            
Get:8 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]        
Get:9 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [109 kB]
Get:10 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages [1304 kB]
Get:11 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
Get:12 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
Get:13 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]           
Get:14 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]             
Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [876 kB]     
Get:16 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [49.9 kB] 
Get:17 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [1110 kB]        
Get:18 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [1236 kB]     
Get:19 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [49.2 kB]       
Get:20 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [25.6 kB]    
Fetched 27.7 MB in 8s (3482 kB/s)                                                                                                                                                                             
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
209 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: http://packages.ros.org/ros2/ubuntu/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  binutils-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
Suggested packages:
  libnewlib-doc
The following NEW packages will be installed:
  binutils-arm-none-eabi gcc-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
0 upgraded, 6 newly installed, 0 to remove and 209 not upgraded.
Need to get 442 MB of archives.
After this operation, 2575 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu jammy/universe amd64 binutils-arm-none-eabi amd64 2.38-3ubuntu1+15build1 [3040 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy/universe amd64 gcc-arm-none-eabi amd64 15:10.3-2021.07-4 [47.7 MB]
Get:3 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-dev all 15:10.3-2021.07-4+17 [1049 kB]                                                                                    
Get:4 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-dev all 3.3.0-1.3 [137 kB]                                                                                                              
Get:5 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-arm-none-eabi all 3.3.0-1.3 [43.2 MB]                                                                                                   
Get:6 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-newlib all 15:10.3-2021.07-4+17 [347 MB]                                                                                  
Fetched 442 MB in 1min 22s (5423 kB/s)                                                                                                                                                                        
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package binutils-arm-none-eabi.
(Reading database ... 61228 files and directories currently installed.)
Preparing to unpack .../0-binutils-arm-none-eabi_2.38-3ubuntu1+15build1_amd64.deb ...
Unpacking binutils-arm-none-eabi (2.38-3ubuntu1+15build1) ...
Selecting previously unselected package gcc-arm-none-eabi.
Preparing to unpack .../1-gcc-arm-none-eabi_15%3a10.3-2021.07-4_amd64.deb ...
Unpacking gcc-arm-none-eabi (15:10.3-2021.07-4) ...
Selecting previously unselected package libstdc++-arm-none-eabi-dev.
[... cut ...] rfaces rcl_lifecycle rcl_logging_interface rcl_logging_noop rclc rclc_lifecycle rclc_parameter rcutils rmw rmw_implementation rmw_microxrcedds ros2trace rosgraph_msgs rosidl_cli rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c sensor_msgs sensor_msgs_py shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_msgs test_tracetools_launch tf2_msgs tracetools_launch tracetools_read tracetools_test tracetools_trace trajectory_msgs unique_identifier_msgs visualization_msgs

pablogs9, 2023-08-11:
It seems that the library has been built correctly. Could you paste here the output of `nm [micro-ROS library .a] | grep rmw_uros` ?

lycanthought, 2023-08-11:
Sure:

         U rmw_uros_epoch_nanos
00000001 T rmw_uros_set_continous_serialization_callbacks
00000001 T rmw_uros_options_set_custom_transport
00000001 T rmw_uros_set_custom_transport
00000001 T rmw_uros_init_options
00000001 T rmw_uros_options_set_client_key
00000001 T rmw_uros_options_set_serial_device
00000001 T rmw_uros_options_set_udp_address
00000001 T rmw_uros_ping_agent
00000001 T rmw_uros_ping_agent_options
00000001 T rmw_uros_epoch_millis
00000001 T rmw_uros_epoch_nanos
00000001 T rmw_uros_epoch_synchronized
00000001 T rmw_uros_sync_session
00000001 T rmw_uros_set_client_session_timeout
00000001 T rmw_uros_set_context_entity_creation_session_timeout
00000001 T rmw_uros_set_context_entity_destroy_session_timeout
00000001 T rmw_uros_set_publisher_session_timeout
00000001 T rmw_uros_set_service_session_timeout
         U rmw_uros_epoch_nanos

pablogs9, 2023-08-11:
Which is your local gcc arm compiler version?

lycanthought, 2023-08-11:
12.2.1-1.2.1

pablogs9, 2023-08-11:
The docker you are using to build the micro-ROS library has this version:

```
root@ad6ee037515f:/# arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

You have two options:

1. Use 10.3.1 locally in order to have the library and your local code built with the same version.
2. Try to modify the build script to install your version (12.2.1) in the docker, it should be easy to modify the build script here: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library_ide/library_generation/library_generation.sh#L26-L28
Take into account that humble is defaulted to ubuntu 22.04 and the [default ARM compiler for 22.04 is 10.3](https://launchpad.net/ubuntu/jammy/+package/gcc-arm-none-eabi)

lycanthought, 2023-08-11:
great, thanks - I will get these aligned and report back.

lycanthought, 2023-08-19:
That worked, thanks

closed 2023-08-19 by lycanthought