micro-ROS/micro_ros_stm32cubemx_utils #115: How to add other packages?

isomadinow, 2023-08-14:
Hello, I want to add the pacmod3 package, but I don't know how to add it or if it's even possible to add this package. If yes, then how can I add it? I'm new to this and I don't quite understand how to add this package.

Here are the links to the packages:
https://github.com/astuff/pacmod3_msgs/tree/main
https://github.com/astuff/pacmod3.git
[image]

I would appreciate your help, thank you in advance.

pablogs9, 2023-08-14:
You need to modify the building script, for example, here we are adding an extra package: 

https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library_ide/library_generation/library_generation.sh#L41-L43

You can modify this script to add the required package to the build folder.

isomadinow, 2023-08-15:
It seems that it didn't work out. Perhaps the "pacmod3" package can only be downloaded via "sudo" command, and cloning it from GitHub might require a different approach here. Is it possible to use "sudo" within this file? What would you recommend?

pablogs9, 2023-08-16:
Clone the actual repo into the folder, it should work

isomadinow, 2023-08-16:
It didn't work out :( 

[image]
[image]

pablogs9, 2023-08-16:
Does it works if you follow this instructions: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils#adding-custom-packages ??

isomadinow, 2023-08-18:
arm-none-eabi-gcc "../Core/Src/freertos.c" -mcpu=cortex-m7 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F767xx -c -I../Core/Inc -I/home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_ver_1/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I../Drivers/STM32F7xx_HAL_Driver/Inc -I../Drivers/STM32F7xx_HAL_Driver/Inc/Legacy -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -I../Drivers/CMSIS/Device/ST/STM32F7xx/Include -I../Drivers/CMSIS/Include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/freertos.d" -MT"Core/Src/freertos.o" --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -o "Core/Src/freertos.o"
../Core/Src/freertos.c: In function 'StartDefaultTask':
../Core/Src/freertos.c:326:15: warning: unused variable 'Period' [-Wunused-variable]
  326 |  unsigned int Period;
      |               ^~~~~~
../Core/Src/freertos.c:324:6: warning: unused variable 'dWheel' [-Wunused-variable]
  324 |  int dWheel=0;
      |      ^~~~~~
../Core/Src/freertos.c:323:6: warning: unused variable 'i' [-Wunused-variable]
  323 |  int i=0;
      |      ^
../Core/Src/freertos.c:322:6: warning: variable 'Wheel_old' set but not used [-Wunused-but-set-variable]
  322 |  int Wheel_old=0;
      |      ^~~~~~~~~
../Core/Src/freertos.c:316:8: warning: unused variable 'EncValue' [-Wunused-variable]
  316 |  short EncValue;
      |        ^~~~~~~~
../Core/Src/freertos.c:315:10: warning: unused variable 'message' [-Wunused-variable]
  315 |     char message[] = "ABCDEFGHIJK";
      |          ^~~~~~~
arm-none-eabi-gcc -o "Auto_Nucleo767ZI_ver_1.elf" @"objects.list"  -lmicroros -mcpu=cortex-m7 -T"/home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_ver_1/STM32F767ZITX_FLASH.ld" --specs=nosys.specs -Wl,-Map="Auto_Nucleo767ZI_ver_1.map" -Wl,--gc-sections -static -L/home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_ver_1/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -u _printf_float -u _scanf_float -Wl,--start-group -lc -lm -Wl,--end-group
/home/isomadinov/st/stm32cubeide_1.12.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.200.202301161003/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./Core/Src/freertos.o: in function `StartDefaultTask':
/home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_ver_1/Debug/../Core/Src/freertos.c:278: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__pacmod3_msgs__msg__SystemCmdFloat'
collect2: error: ld returned 1 exit status
make[1]: *** [makefile:70: Auto_Nucleo767ZI_ver_1.elf] Error 1
make: *** [makefile:63: all] Error 2
"make -j16 all" terminated with exit code 2. Build might be incomplete.

04:13:40 Build Failed. 4 errors, 6 warnings. (took 2s.999ms)
[image]
[image]

pablogs9, 2023-08-18:
Could you paste here the complete build log to check if the package is being built?

isomadinow, 2023-08-18:
[image]

unfortunately it doesn't work

pablogs9, 2023-08-18:
The complete build log, not your `extra_packages.repos`

pablogs9, 2023-08-18:
The error is here: `Non valid RTOS/Platform: generate_libS/generic`

This `S/` shall not be there, check here: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library_ide/library_generation/library_generation.sh#L35

Maybe this is a Linux / Windows file format thing. Go to the corresponding `library_generation.sh` (be careful there are two, [this](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/iron/microros_static_library_ide/library_generation/library_generation.sh) and [this](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/iron/microros_static_library/library_generation/library_generation.sh))

And write a correct space there, maybe Windows is taking the Linux space character as `S/` somehow...

isomadinow, 2023-08-18:
08:33:29 **** Incremental Build of configuration Debug for project Auto_Nucleo767ZI_pacmod3 ****
make -j16 all 
docker pull microros/micro_ros_static_library_builder:humble && docker run --rm -v /home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_pacmod3:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:humble
humble: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:3c13fee7e7d6814ebc87c7ccd4c5d7592310b8adc86c5ff2a1f4de179776a0fb
Status: Image is up to date for microros/micro_ros_static_library_builder:humble
docker.io/microros/micro_ros_static_library_builder:humble
dos2unix: converting file /project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/library_generation.sh to Unix format...
Found CFLAGS:
-------------
-ffunction-sections -fdata-sections -DSTM32CUBEIDE -DENOTSUP=1 -DECANCELED=1 -DEOWNERDEAD=1 -DENOTRECOVERABLE=1 -mcpu=cortex-m7 -mfloat-abi=hard -mthumb -mfpu=fpv5-d16 -O0 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F767xx --specs=nano.specs -I/project/Drivers/STM32F7xx_HAL_Driver/Inc -I/project/Drivers/STM32F7xx_HAL_Driver/Inc/Legacy -I/project/Core/Inc -I/project/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -I/home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_ver_1/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include -I/project/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I/project/Drivers/CMSIS/Include -I/project/Drivers/CMSIS/Device/ST/STM32F7xx/Include -I/project/Middlewares/Third_Party/FreeRTOS/Source/include
-------------
Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Get:3 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [44.0 kB]
Get:4 http://packages.ros.org/ros2/ubuntu jammy InRelease [4682 B]
Get:5 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [980 kB]
Get:6 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages [1304 kB]
Get:7 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [897 kB]
Get:8 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [860 kB]
Get:9 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]
Get:10 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [109 kB]
Get:11 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
Get:12 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
Get:13 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]
Get:14 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [49.8 kB]
Get:16 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [1241 kB]
Get:17 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [923 kB]
Get:18 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [1136 kB]
Get:19 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [25.6 kB]
Get:20 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [49.2 kB]
Fetched 27.8 MB in 9s (3027 kB/s)
Reading package lists...
W: http://packages.ros.org/ros2/ubuntu/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev
  libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
Suggested packages:
  libnewlib-doc
The following NEW packages will be installed:
  binutils-arm-none-eabi gcc-arm-none-eabi libnewlib-arm-none-eabi
  libnewlib-dev libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
0 upgraded, 6 newly installed, 0 to remove and 214 not upgraded.
Need to get 442 MB of archives.
After this operation, 2575 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu jammy/universe amd64 binutils-arm-none-eabi amd64 2.38-3ubuntu1+15build1 [3040 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy/universe amd64 gcc-arm-none-eabi amd64 15:10.3-2021.07-4 [47.7 MB]
Get:3 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-dev all 15:10.3-2021.07-4+17 [1049 kB]
Get:4 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-dev all 3.3.0-1.3 [137 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-arm-none-eabi all 3.3.0-1.3 [43.2 MB]
Get:6 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-newlib all 15:10.3-2021.07-4+17 [347 MB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 442 MB in 54s (8172 kB/s)
Selecting previously unselected package binutils-arm-none-eabi.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
[... 288 lines cut ...]
Finished <<< ament_uncrustify [0.46s]
--- stderr: ament_uncrustify
[... cut ...] ment_xmllint domain_coordinator
#All required rosdeps installed successfully
Cloning into 'pacmod3_msgs'...
=== ./pacmod3_msgs (git) ===
Could not clone repository 'https://github.com/astuff/pacmod3_msgs': fatal: destination path '.' already exists and is not an empty directory.
make[1]: *** [makefile:90: pre-build] Error 1
make: *** [makefile:63: all] Error 2
"make -j16 all" terminated with exit code 2. Build might be incomplete.

08:35:27 Build Failed. 20 errors, 0 warnings. (took 1m:58s.29ms)

pablogs9, 2023-08-18:
it seems that you already have the folder, please delete the actual folder and retry

isomadinow, 2023-08-18:
`08:49:14 **** Build of configuration Debug for project Auto_Nucleo767ZI_pacmod3 ****
make -j16 all 
docker pull microros/micro_ros_static_library_builder:humble && docker run --rm -v /home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_pacmod3:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:humble
humble: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:3c13fee7e7d6814ebc87c7ccd4c5d7592310b8adc86c5ff2a1f4de179776a0fb
Status: Image is up to date for microros/micro_ros_static_library_builder:humble
docker.io/microros/micro_ros_static_library_builder:humble
dos2unix: converting file /project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/library_generation.sh to Unix format...
Found CFLAGS:
-------------
-ffunction-sections -fdata-sections -DSTM32CUBEIDE -DENOTSUP=1 -DECANCELED=1 -DEOWNERDEAD=1 -DENOTRECOVERABLE=1 -mcpu=cortex-m7 -mfloat-abi=hard -mthumb -mfpu=fpv5-d16 -O0 -DSTM32F767xx -DUSE_HAL_DRIVER -DDEBUG --specs=nano.specs -I/project/Core/Inc -I/project/Middlewares/Third_Party/FreeRTOS/Source/include -I/project/Drivers/STM32F7xx_HAL_Driver/Inc/Legacy -I/project/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I/project/Drivers/STM32F7xx_HAL_Driver/Inc -I/project/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -I/project/Drivers/CMSIS/Include -I/project/Drivers/CMSIS/Device/ST/STM32F7xx/Include -I/home/isomadinov/STM32CubeIDE/workspace_1.12.1/Auto_Nucleo767ZI_ver_1/micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include
-------------
Get:1 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Get:2 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Get:3 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [897 kB]
Get:4 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [109 kB]
Get:6 http://packages.ros.org/ros2/ubuntu jammy InRelease [4682 B]
Get:7 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
Get:8 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [980 kB]
Get:9 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages [1304 kB]
Get:10 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [860 kB]
Get:11 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [44.0 kB]
Get:12 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
Get:13 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
Get:14 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]
Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [1241 kB]
Get:16 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [923 kB]
Get:17 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [49.8 kB]
Get:18 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [1136 kB]
Get:19 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [25.6 kB]
Get:20 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [49.2 kB]
Fetched 27.8 MB in 4s (6357 kB/s)
Reading package lists...
W: http://packages.ros.org/ros2/ubuntu/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev
  libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
Suggested packages:
  libnewlib-doc
The following NEW packages will be installed:
  binutils-arm-none-eabi gcc-arm-none-eabi libnewlib-arm-none-eabi
  libnewlib-dev libstdc++-arm-none-eabi-dev libstdc++-arm-none-eabi-newlib
0 upgraded, 6 newly installed, 0 to remove and 214 not upgraded.
Need to get 442 MB of archives.
After this operation, 2575 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu jammy/universe amd64 binutils-arm-none-eabi amd64 2.38-3ubuntu1+15build1 [3040 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy/universe amd64 gcc-arm-none-eabi amd64 15:10.3-2021.07-4 [47.7 MB]
Get:3 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-dev all 15:10.3-2021.07-4+17 [1049 kB]
Get:4 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-dev all 3.3.0-1.3 [137 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libnewlib-arm-none-eabi all 3.3.0-1.3 [43.2 MB]
Get:6 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libstdc++-arm-none-eabi-newlib all 15:10.3-2021.07-4+17 [347 MB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 442 MB in 50s (8863 kB/s)
Selecting previously unselected package binutils-arm-none-eabi.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
[... 298 lines cut ...]
Starting >>> ament_index_cpp
Finished <<< ament_cmake_ros [0.36s]
Finished <<< ament_index_cpp [3.36s]

[... cut ...]
Using provided meta: /project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/colcon.meta
[0.293s] ERROR:colcon:colcon build: Duplicate package names not supported:
- pacmod3_msgs:
  - extra_packages/pacmod3_msgs
  - pacmod3_msgs
[ros2run]: Process exited with failure 1
make[1]: *** [makefile:90: pre-build] Error 1
make: *** [makefile:63: all] Error 2
"make -j16 all" terminated with exit code 2. Build might be incomplete.

08:50:58 Build Failed. 20 errors, 0 warnings. (took 1m:44s.144ms)

`

pablogs9, 2023-08-18:
Please check the errors in your log, it is clearly notifying that there are a duplicity in your folder:

```
[0.293s] ERROR:colcon:colcon build: Duplicate package names not supported:
- pacmod3_msgs:
  - extra_packages/pacmod3_msgs
  - pacmod3_msgs
```

isomadinow, 2023-08-18:
Everything worked out, thank you very much. I have an unrelated question: Do you have any examples of creating 3-4 subscribers?

```c
rclc_subscription_init_default(
    &subscriberSteer,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32),
    "/stm/receive/data/steer");

rclc_subscription_init_default(
    &subscriberThrottle,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32),
    "/stm/receive/data/throttle");

rclc_subscription_init_default(
    &subscriberBreak,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32),
    "/stm/receive/data/break");

// Executor
rclc_executor_t executor;
rclc_executor_init(&executor, &support.context, 3, &allocator);
rclc_executor_add_subscription(&executor, &subscriberThrottle, &throttleMsg, &GetDataThrottle, ON_NEW_DATA);
rclc_executor_add_subscription(&executor, &subscriberSteer, &steerMsg, &GetDataSteer, ON_NEW_DATA);
rclc_executor_add_subscription(&executor, &subscriberBreak, &breakMsg, &GetDataBreak, ON_NEW_DATA);

for (;;) {
    rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
    osDelay(10);
}
```

It is normal?.

pablogs9, 2023-08-18:
I'm closing since initial issue is solved

closed 2023-08-18 by pablogs9

isomadinow, 2023-08-18:
As I understand, the file specifies the number of subscribers that can be added, indicating a value of 5. I added 3, and it should work. Regarding the executor, if I created 3 subscribers, then I should create 3 executors, correct? Thank you very much!

pablogs9, 2023-08-18:
You can add the 3 subs to the same executor, if you set the parameter correctly: rclc_executor_init(&executor, &support.context, **3**, &allocator);