micro-ROS/micro_ros_stm32cubemx_utils #116: rclc_support_init() Function doesnt work and give me an RCL_RET_ERROR back

28marcel28, 2023-08-16:
Hello Microros or ROS2 Community,

Im using Ubuntu 22.04 LTS on an Raspberry Pi 4.0 (Compute Module). Im using a STM32F767ZI with the STM32CubeIDE Version 1.7 with microros stack (micro_ros_stm32cubemx_utils). I can compile and flash the MCU with docker. It still works.

The problem is, that the rclc_support_init() function doesn`t initialize rcl and doesn´t create some support data structures. Cause They give me a RCL_RET_ERROR back.

Thats the reason i cant create my Publisher.

And so, the communication between both devices doesn´t work with the Agent and the XRCE-DDS-Client on stm32.

In the github Issues I found 3 times the same problem, but they have no solution.

Issue 1: [https://github.com/micro-ROS/micro_ro...](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/107)

Issue 2: [https://github.com/micro-ROS/micro_ro...](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/82)

Issue 3: [https://github.com/micro-ROS/micro_ro...](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/85)

Can anyone help me with this problem? Or get anyone a solution for this problem?

Thank you for your help,

Your Microros Developer Kurwaboy.

pablogs9, 2023-08-16:
Hello, which transport are you using?

28marcel28, 2023-08-16:
I am utilizing USART3 in combination with DMA Transport. The configuration in my .ioc file is as follows: DMA Rx and Tx are both set to a very high priority. Rx is configured to operate in circular mode. Additionally, I have enabled the global interrupt for USART3 in the NVIC settings.

For the implementation of the MicroROS Stack in STM32CubeIDE, I am referring to this video: [MicroROS Stack Implementation: https://www.youtube.com/watch?v=bn-P3fxtTF4

However, I am encountering an issue with the rclc_support_init() function, as it fails to generate and consequently triggers the hardfault handler. I did review your previously closed issue regarding a similar problem, but I do not seem to be experiencing the same issues that were identified in that case.

pablogs9, 2023-08-16:
Could you check if the MCU is sending bytes over the serial port? Do not connect the agent for testing.

28marcel28, 2023-08-16:
Can you help me, i dont know at the moment how to force to send or to see bytes over the DMA Transport system.

Here is my main.c file: 

/* USER CODE BEGIN Header */
/**
 ******************************************************************************
 * @file           : main.c
 * @brief          : Main program body
 ******************************************************************************
 * @attention		: Marcel Kotzyba, Alper Bagci, Saleh Azizpour
 *
 * <h2><center>&copy; Copyright (c) 2023 STMicroelectronics.
 * All rights reserved.</center></h2>
 *
 * This software component is licensed by ST under BSD 3-Clause license,
 * the "License"; You may not use this file except in compliance with the
 * License. You may obtain a copy of the License at:
 *                        opensource.org/licenses/BSD-3-Clause
 *                        docker pull microros/micro_ros_static_library_builder:humble && docker run --rm -v /home/gs1/STM32CubeIDE/workspace_1.7.0/frost_ros2_micro_ros/Nucleo-F767ZI_ROS2_alternative:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:humble
 *
 ******************************************************************************
 */
#if 0
/* USER CODE END Header */
/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "cmsis_os.h"
#include "lwip.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */

#else

/**** No FreeRTOS Wrapper Functions ****/
#include "main.h"
#include "lwip.h"
#include "FreeRTOS.h"
#include "task.h"
#include "queue.h"
#include "timers.h"

#endif

/**** Micro-ROS Libraries ****/
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <uxr/client/transport.h>
#include <rmw_microxrcedds_c/config.h>
#include <rmw_microros/rmw_microros.h>

/**** ROS Message Typen ****/
#include <std_msgs/msg/int32.h>
#include <geometry_msgs/msg/twist.h>

/**** Standard Libraries ****/
#include "stdio.h"
#include "stdlib.h"
#include "string.h"

/**** Ethernet TCP Socket Librearies ****/
#include "sys/socket.h"
#include "sys/types.h"

/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
[... 604 lines cut ...]
	//__disable_irq();
	while (1)
	{
	}
	/* USER CODE END Error_Handler_Debug */
}

#ifdef  USE_FULL_ASSERT
/**
 * @brief  Reports the name of the source file and the source line number
 *         where the assert_param error has occurred.
 * @param  file: pointer to the source file name
 * @param  line: assert_param error line source number
 * @retval None
 */
void assert_failed(uint8_t *file, uint32_t line)
{
	/* USER CODE BEGIN 6 */
	/* User can add his own implementation to report the file name and line number,
     ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
	/* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

/************************ (C) COPYRIGHT STMicroelectronics *****END OF FILE****/

28marcel28, 2023-08-16:
So, i understand, that the problem can be a bad Allocation.
Does that mean, i can use the normal alloc malloc, free functions to replace following:
`	rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();
	freeRTOS_allocator.allocate = microros_allocate;
	freeRTOS_allocator.deallocate = microros_deallocate;
	freeRTOS_allocator.reallocate = microros_reallocate;
	freeRTOS_allocator.zero_allocate =  microros_zero_allocate;`

I couldnt see any wrapper functions to solve the problem in the mentioned issue: #110

danielallstar, 2023-09-08:
Hi @28marcel28, did you find a solution for this problem?

28marcel28, 2023-09-12:
We didnt find a solution for the STM32F767ZI. But we changed the board to the STM32F401re and it works fine! 
But we assume that the Transport Layer has an Hardware problem via USART and UART. This board isnt compatible for the present Microros-Stack!

zp1915, 2024-05-27:
@danielallstar @28marcel28   I have the same problem as you using the STM32F103ZE. finally,I find a solution from the suggetion of pablogs9. the stack of microros task is too small.
const osThreadAttr_t defaultTask_attributes = {
  .name = "defaultTask",
  .stack_size = 128 * 16,       //.stack_size = 128 * 4
  .priority = (osPriority_t) osPriorityNormal,
};

Scrath1, 2024-11-13:
I had the same issue with rclc_support_init and managed to fix it though I don't know why it works for me now.

I am using a Creality 4.2.2 3D printer mainboard which has an STM32F103RET6 controller. I tried establishing a connection through 2 GPIO pins with the USART3 using interrupts without any success.

What worked for me instead was to use USART1 with the exact same settings. The main difference with USART1 (which technically shouldn't make a difference) on this board is that USART1 is connected to a CH340G Serial-to-USB chip. Attaching a USB cable and establishing the connection over that instead of using jumper wires from GPIO pins did the trick for me.

Some additional information about my setup in case someone wants to try and replicate it:
FreeRTOS-Kernel version: v11.1.0
Allocators used: microros specific allocators as specified in the examples
Micro-ros task stack size: 5000 32bit words (I could probably lower that though)