micro-ROS/micro_ros_stm32cubemx_utils #117: I can't create the libmicroros folder and the libmicroros.a

zxa1095696200, 2023-08-17:
hardware: stm32f429
env: cubeMX+ ubuntu20.04 + ros2-foxy

issue detail:  
after the step of cmds below:
1、
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ sudo docker pull microros/micro_ros_static_library_builder:foxy
[sudo] password for allen: 
foxy: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:bda0dcca059534cfc39f1fb2744bf58807acb8c6215a33f17fed2a9af15501a7
Status: Image is up to date for microros/micro_ros_static_library_builder:foxy
docker.io/microros/micro_ros_static_library_builder:foxy

2、
llen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library microros/micro_ros_static_library_builder:foxy
dos2unix: converting file /project/micro_ros_stm32cubemx_utils/microros_static_library/library_generation/library_generation.sh to Unix format...
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]                                           
Get:2 http://packages.ros.org/ros2/ubuntu focal InRelease [4685 B]                          
Get:3 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1189 kB]               
Get:4 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                                            
Get:5 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]                          
Get:6 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]                      
Get:7 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]                    
Get:8 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]                              
Get:9 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]                                                                                                                                
Get:10 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [1091 kB]                                                                                                                 
Get:11 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]                                                                                                                           
Get:12 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [29.3 kB]                                                                                                               
Get:13 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [2648 kB]                                                                                                               
Get:14 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [2938 kB]                                                                                                                     
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1391 kB]                                                                                                                   
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [3420 kB]                                                                                                                       
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [32.0 kB]                                                                                                                 
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [2789 kB]                                                                                                                 
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [55.2 kB]                                                                                                                     
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [28.6 kB]                                                                                                                 
Fetched 29.0 MB in 33s (888 kB/s)                                                                                                                                                                         
Reading package lists... Done
Building dependency tree       
Reading state information... Done
253 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  binutils-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev libstdc++-arm-none-eabi-newlib
Suggested packages:
  libnewlib-doc
The following NEW packages will be installed:
  binutils-arm-none-eabi gcc-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev libstdc++-arm-none-eabi-newlib
0 upgraded, 5 newly installed, 0 to remove and 253 not upgraded.
Need to get 271 MB of archives.
After this operation, 2340 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu focal/universe amd64 binutils-arm-none-eabi amd64 2.34-4ubuntu1+13ubuntu1 [2571 kB]
Get:2 http://archive.ubuntu.com/ubuntu focal/universe amd64 gcc-arm-none-eabi amd64 15:9-2019-q4-0ubuntu1 [31.6 MB]
Get:3 http://archive.ubuntu.com/ubuntu focal/universe amd64 libnewlib-dev all 3.3.0-0ubuntu1 [131 kB]                                                                                                     
Get:4 http://archive.ubuntu.com/ubuntu focal/universe amd64 libnewlib-arm-none-eabi all 3.3.0-0ubuntu1 [34.5 MB]                                                                                          
Get:5 http://archive.ubuntu.com/ubuntu focal/universe amd64 libstdc++-arm-none-eabi-newlib all 15:9-2019-q4-0ubuntu1+12build2 [202 MB]                                                                    
Fetched 271 MB in 3min 11s (1419 kB/s)                                                                                                                                                                    
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package binutils-arm-none-eabi.
(Reading database ... 59185 files and directories currently installed.)
Preparing to unpack .../binutils-arm-none-eabi_2.34-4ubuntu1+13ubuntu1_amd64.deb ...
Unpacking binutils-arm-none-eabi (2.34-4ubuntu1+13ubuntu1) ...
Selecting previously unselected package gcc-arm-none-eabi.
Preparing to unpack .../gcc-arm-none-eabi_15%3a9-2019-q4-0ubuntu1_amd64.deb ...
Unpacking gcc-arm-none-eabi (15:9-2019-q4-0ubuntu1) ...
Selecting previously unselected package libnewlib-dev.
Preparing to unpack .../libnewlib-dev_3.3.0-0ubuntu1_all.deb ...
Unpacking libnewlib-dev (3.3.0-0ubuntu1) ...
Selecting previously unselected package libnewlib-arm-none-eabi.
Preparing to unpack .../libnewlib-arm-none-eabi_3.3.0-0ubuntu1_all.deb ...
Unpacking libnewlib-arm-none-eabi (3.3.0-0ubuntu1) ...
Selecting previously unselected package libstdc++-arm-none-eabi-newlib.
Preparing to unpack .../libstdc++-arm-none-eabi-newlib_15%3a9-2019-q4-0ubuntu1+12build2_all.deb ...
Unpacking libstdc++-arm-none-eabi-newlib (15:9-2019-q4-0ubuntu1+12build2) ...
Setting up binutils-arm-none-eabi (2.34-4ubuntu1+13ubuntu1) ...
Setting up gcc-arm-none-eabi (15:9-2019-q4-0ubuntu1) ...
Setting up libnewlib-dev (3.3.0-0ubuntu1) ...
Setting up libnewlib-arm-none-eabi (3.3.0-0ubuntu1) ...
Setting up libstdc++-arm-none-eabi-newlib (15:9-2019-q4-0ubuntu1+12build2) ...
Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
Creating firmware for generate_lib platform generic

then, i can't create the libmicroros folder and the libmicroros.a, and it doesn't report any errors. 

allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test/micro_ros_stm32cubemx_utils/microros_static_library$ ls
library_generation

we can see , only the library_generation folder in the microros_static_library but we lost the libmicroros folder and the libmicroros.a, why?

pablogs9, 2023-08-17:
Could you paste te complete log of the generation step?

zxa1095696200, 2023-08-17:
Okey. Here is the  complete log of the generation step with a new terminal below ：

allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ source /opt/ros/$ROS_DISTRO/setup.bash
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ l
build/  Core/  Drivers/  install/  log/  Makefile*  MDK-ARM/  micro_ros_stm32cubemx_utils/  microros_test.ioc*  Middlewares/  startup_stm32f429xx.s*  STM32F429ZITx_FLASH.ld*
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ ls
build  Core  Drivers  install  log  Makefile  MDK-ARM  micro_ros_stm32cubemx_utils  microros_test.ioc  Middlewares  startup_stm32f429xx.s  STM32F429ZITx_FLASH.ld
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ rm -rf micro_ros_stm32cubemx_utils
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ git clone https://github.com/micro-ROS/micro_ros_stm32cubemx_utils.git
Cloning into 'micro_ros_stm32cubemx_utils'...
remote: Enumerating objects: 420, done.
remote: Counting objects: 100% (282/282), done.
remote: Compressing objects: 100% (103/103), done.
remote: Total 420 (delta 204), reused 225 (delta 159), pack-reused 138
Receiving objects: 100% (420/420), 565.61 KiB | 21.00 KiB/s, done.
Resolving deltas: 100% (247/247), done.
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ cd micro_ros_stm32cubemx_utils
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test/micro_ros_stm32cubemx_utils$ git checkout foxy
Branch 'foxy' set up to track remote branch 'foxy' from 'origin'.
Switched to a new branch 'foxy'
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test/micro_ros_stm32cubemx_utils$ cd ..
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ docker pull microros/micro_ros_static_library_builder:foxy
permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/images/create?fromImage=microros%2Fmicro_ros_static_library_builder&tag=foxy": dial unix /var/run/docker.sock: connect: permission denied
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ sudo docker pull microros/micro_ros_static_library_builder:foxy
[sudo] password for allen: 
foxy: Pulling from microros/micro_ros_static_library_builder
Digest: sha256:bda0dcca059534cfc39f1fb2744bf58807acb8c6215a33f17fed2a9af15501a7
Status: Image is up to date for microros/micro_ros_static_library_builder:foxy
docker.io/microros/micro_ros_static_library_builder:foxy
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test$ sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library microros/micro_ros_static_library_builder:foxy
dos2unix: converting file /project/micro_ros_stm32cubemx_utils/microros_static_library/library_generation/library_generation.sh to Unix format...
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                   
Get:2 http://packages.ros.org/ros2/ubuntu focal InRelease [4685 B]                                                
Get:3 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]                                                       
Get:4 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1189 kB]             
Get:5 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [2938 kB]                
Get:6 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]                 
Get:7 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]                          
Get:8 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]                  
Get:9 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]                                                                                                                                
Get:10 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]                                                                                                                           
Get:11 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [1091 kB]                                                                                                                 
Get:12 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [2648 kB]                                                                                                               
Get:13 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]                                                                                                                         
Get:14 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [3430 kB]                                                                                                                       
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [32.0 kB]                                                                                                                 
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1392 kB]                                                                                                                   
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [2795 kB]                                                                                                                 
[... cut ...]
3rd-party-licenses.txt  CHANGELOG.rst  CONTRIBUTING.md  extra_sources  LICENSE  microros_static_library  microros_static_library_ide  NOTICE  package.xml  README.md  sample_main.c  sample_project.ioc
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test/micro_ros_stm32cubemx_utils$ cd microros_static_library
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test/micro_ros_stm32cubemx_utils/microros_static_library$ ls
library_generation
allen@ubuntu:~/Desktop/STM32F429ZIT6_uROS/microros_test/micro_ros_stm32cubemx_utils/microros_static_library$

pablogs9, 2023-08-18:
This is weird... Could you debug the library generation script and check if it [reach this line](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library/library_generation/library_generation.sh#L19) ?

cocaston, 2024-06-22:
Have you successfully resolved this issue.
I also have the same problem. After debugging, I found that the issue lies in 
`ros2 run micro_ros_setup create_firmware_ws.sh generate_lib`
 There is no relevant output, and I only get 
'Creating firmware for generate_lib platform generic.' 
Moreover, all subsequent commands are not executed.

cocaston, 2024-06-22:
> This is weird... Could you debug the library generation script and check if it [reach this line](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/microros_static_library/library_generation/library_generation.sh#L19) ?

The normal output of `ros2 run micro_ros_setup create_firmware_ws.sh generate_lib` should be as shown in the following figure:
[image]

My environment setting:
ubuntu20.04
ros2 foxy

In order to reproduce. I have uploaded the project created by STM32CubeMX to [GitHub](https://github.com/cocaston/stm32_micro_ros_demo.git).

cocaston, 2024-06-28:
> The normal output of `ros2 run micro_ros_setup create_firmware_ws.sh generate_lib` should be as shown in the following figure: [...]

Resolved by set a http proxy for the docker command.