micro-ROS/micro_ros_stm32cubemx_utils #118: rcl_publish() failed

rusuanjun007, 2023-09-01:
Hi,

I am new to micro-ros and I have a problem with rcl_publish(). rcl_publish() failed and I cannot receive data sent back to my PC.  
Here is the detail of my setting.

**Microcontroller board:**
Nucleo-f401re

**PC:**
ROS humble
STM32CubeIDE

**Code is the example code:**
```
void StartDefaultTask(void *argument)
{
	 rmw_uros_set_custom_transport(
	    true,
	    (void *) &huart2,
	    cubemx_transport_open,
	    cubemx_transport_close,
	    cubemx_transport_write,
	    cubemx_transport_read);

	  rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();
	  freeRTOS_allocator.allocate = microros_allocate;
	  freeRTOS_allocator.deallocate = microros_deallocate;
	  freeRTOS_allocator.reallocate = microros_reallocate;
	  freeRTOS_allocator.zero_allocate =  microros_zero_allocate;

	  if (!rcutils_set_default_allocator(&freeRTOS_allocator)) {
	      printf("Error on default allocators (line %d)\n", __LINE__);
	  }

	  // micro-ROS app

	  rcl_publisher_t publisher;
	  std_msgs__msg__Int32 msg;
	  rclc_support_t support;
	  rcl_allocator_t allocator;
	  rcl_node_t node;

	  allocator = rcl_get_default_allocator();

	  //create init_options
	  rclc_support_init(&support, 0, NULL, &allocator);

	  // create node
	  rclc_node_init_default(&node, "cubemx_node", "", &support);

	  // create publisher
	  rclc_publisher_init_default(
	    &publisher,
	    &node,
	    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
	    "cubemx_publisher");

	  msg.data = 0;

	  for(;;)
	  {
	    rcl_ret_t ret = rcl_publish(&publisher, &msg, NULL);
	    if (ret != RCL_RET_OK)
	    {
	      printf("Error publishing (line %d)\n", __LINE__);
	    }

	    msg.data++;
	    osDelay(10);
	  }
}
```

**Fault behavior:**

- rcl_publish(&publisher, &msg, NULL) returns 300.
[image]

- datawriter is not created.
[image]

- topic  /cubemx_publisher can not be found. 
[image]

Does anyone have any suggestions? Thanks for your great help!

pablogs9, 2023-09-04:
- Could you share the full agent log using the flag `-v6`?
- Does your code freeze or does it just return an error code?
- Could you test this code by adding a timer and an executor? You can take this code as a reference: https://github.com/micro-ROS/micro_ros_arduino/blob/iron/examples/micro-ros_publisher/micro-ros_publisher.ino

rusuanjun007, 2023-09-04:
Hi @pablogs9,

Thanks for your suggestion. I tried the following.

- Add flag -v6

  The command is
  `ros2 run micro_ros_agent micro_ros_agent serial -b 115200 --dev /dev/ttyACM0 -v6`
  
  The log shows the following
  ```
  rusuanjun@rusuanjun-MS-7D30:~$ ros2 run micro_ros_agent micro_ros_agent serial -b 115200 --dev /dev/ttyACM0 -v6
  [1693816636.111053] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
  [1693816636.111163] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
  [1693816639.367885] info     | Root.cpp           | create_client            | create                 | client_key: 0x5851F42D, session_id: 0x81
  [1693816639.367940] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x5851F42D, address: 0
  [1693816639.368081] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 19, data: 
  0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
  [1693816639.583568] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 44, data: 
  0000: 81 80 00 00 01 07 22 00 00 0A 00 01 01 03 00 00 14 00 00 00 00 01 A5 A5 0C 00 00 00 63 75 62 65
  0020: 6D 78 5F 6E 6F 64 65 00 00 00 00 00
  [1693816639.589402] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x5851F42D, participant_id: 0x000(1)
  [1693816639.589455] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data: 
  0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
  [1693816639.589486] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
  [1693816639.714047] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
  [1693816639.791803] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
  [1693816639.820514] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 84, data: 
  0000: 81 80 01 00 01 07 4A 00 00 0B 00 02 02 03 00 00 3C 00 00 00 14 00 00 00 72 74 2F 63 75 62 65 6D
  0020: 78 5F 70 75 62 6C 69 73 68 65 72 00 00 01 01 08 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73
  0040: 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 00 00 01 00 00
  [1693816639.820687] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x5851F42D, topic_id: 0x000(2), participant_id: 0x000(1)
  [1693816639.820832] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 14, data: 
  0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
  [1693816639.820845] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
  [1693816639.831631] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0B 01 05 00 00 00 01 00 80
  [1693816639.831869] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
  [1693816639.914238] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
  [1693816640.114460] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
  [1693816640.236736] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
  [1693816640.246726] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
  [1693816640.246956] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
  [1693816640.314629] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
  [1693816640.454715] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 13, data: 
  0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
  [1693816640.465472] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x5851F42D, len: 24, data: 
  0000: 81 80 02 00 01 07 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
  [1693816640.465616] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x5851F42D, publisher_id: 0x000(3), participant_id: 0x000(1)
[... cut ...]
	  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
	  RCCHECK(rclc_executor_add_timer(&executor, &timer));
  
	  msg.data = 0;
  
	  for (;;) {
		  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
		  msg.data++;
		  osDelay(10);
	  }
  
	  /* USER CODE END StartDefaultTask */
  }
  ```
  I might find the problem. I noticed the rclc_publisher_init_default() failed with temp_rc=1. The publisher creation failed and it caused rcl_publish() to fail. But not sure how to solve it.

pablogs9, 2023-09-04:
Definitely, the problem is that the datawriter is not being created because `rclc_publisher_init_default` is failing.

This may be due to many causes, could you debug the publisher creation to check where it is failing?

My first thought is do you have enough heap (for mallocing memory) and stack configured for the FreeRTOS task?

rusuanjun007, 2023-09-04:
Hi @pablogs9,

It seems there is something wrong with the DMA setting. I changed the transport configuration from U(S)ART with DMA to U(S)ART with Interrupts, and it works. Appreciate your help!

closed 2023-09-04 by rusuanjun007