micro-ROS/micro_ros_stm32cubemx_utils #119: Issue with rclc_support_init() on STM32H743ZI using FreeRTOS 10.3.1

danielallstar, 2023-09-08:
Hello,

I'm encountering a similar issue as described in ticket #116 while working with an STM32H743ZI2. Specifically, the problem lies with the rclc_support_init() function, which does not appear to be functioning correctly. I've configured the settings in line with those mentioned in ticket #116.

The issue manifests as follows: after calling the rclc_support_init() function, the MCU exits the Task function but fails to return. When halting the debugger at various points, I've observed that the FreeRTOS functions vTaskSwitchContext() and vPortExitCritical() are being invoked.

It's worth noting that on another MCU using an older version of FreeRTOS (10.2.1), there were no issues. However, the STM32H743ZI MCU comes with FreeRTOS 10.3.1 as the default from MXCube.

This problem appears to be a recurring issue that has been reported multiple times. The most recent suggested solution involves modifying the allocator functions, but this solution lacks clarity and hasn't resolved the problem for me.

I would greatly appreciate any assistance or suggestions that could help me address this issue conclusively. Thank you in advance for your help.

Hao-Lion-ZJU, 2023-12-07:
hello ， I'm having the same problem as you. It seems that the connect with agent is not well established yet. You can try a few more times on rclc_support_init(). That's what I did.
```
//broke unitl success
   do{
        ret =  rclc_support_init(&support, 0, NULL, &allocator);
        osDelay(1000);
    } while (ret != RCL_RET_OK);
```

nikitax75, 2024-02-04:
Hello there @danielallstar  encountering the exact same problem with the exact same MCU. Wondering if you fixed it.

Thanks

danielallstar, 2024-02-06:
Hey, I did not solve that problem. You could try the solution of @Hao-Lion-ZJU, but I did not try that myself.

nikitax75, 2024-02-06:
Thanks! Yeah, did try it, does not work and could never stop the problem because the default task never resumes once it has entered `rclc_support_init`. 

Also, tried using default allocators as suggested in [](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/110) but does not fix the problem. Seems that micro-ros support for the stm32h743 is essentially broken. I wished we didn't design a custom PCB thinking it would work...

nikitax75, 2024-02-06:
Wondering if the maintainer @pablogs9 can help. I'd love to help but not sure in which direction to go there. I'd also be really happy to pay for a bug bounty or some support if it can unlock the issue for everyone

pablogs9, 2024-02-07:
Hello @nikitax75 could you open a PR for fixing the issue you mentioned?