micro-ROS/micro_ros_stm32cubemx_utils #121: Can't understand build

JorgePerC, 2023-09-22:
Hey, newcomer here. 
I'm trying to incorporate this into a new CubeIDE project, but can't seem to understand how to do it. Specially with steps 3 and 4, I clone this repo into my project, yet, I fail at _building?_ libmicroros. I guess this is done through docker, but then how can I find the absolute path at step 4?
I've tried to run the library_generation.sh in my local machine but fail at doing so. 
I've completed the tutorial at https://micro.ros.org/docs/tutorials/core/first_application_linux/

pablogs9, 2023-09-25:
Hello, we need more details about your failure. In which step it fails?

You mention problems with steps 3 and 4:

[image]

Could you elaborate on this?

JorgePerC, 2023-10-11:
So... I had an initial error due to working on a board with two micros.
Tried again with a single one,  and copypasted the paths as the instructions said. 
//question: should step 3 be repeated for MCU GCC and MCU G++ compiler?
It eventually got to the build process. 
However, it failed. It is a missing file _<uxr/client/transport.h>_
[image]

Then I found a similar [issue](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/113)
But I got question, if I want to modify it, should I just edit the file locally on the clone of this repo? or is it dependent on the Docker image?

Btw, thanks for the help :D

pablogs9, 2023-10-13:
Yes, if you are using C and C++ you will need to add the include paths for both compilers