micro-ROS/micro_ros_stm32cubemx_utils #124: rcl_publish() returns RCL_RET_ERROR

Pansamic, 2023-10-12:
## issue description
publishing nav_msgs/msg/odometry failed. rcl_publish() returns RCL_RET_ERROR. 
but there's no error when i publish sensor_msgs/msg/Imu.
successfully initialize support, node, publisher and subcriber, but just failed to publish odometry.
using USB and haven't try UART.

## brief config
* microros cross compile PC: Ubuntu22.04 + ROS2 Humble
* stm32cubemx_utils branch distro: Humble
* microros communication interface: 8MHz USB CDC
* MCU: STM32F407ZGT6

## trials

**1. increase FreeRTOS task stack size**
[image]

**2. increase USB CDC transmission buffer**
increase buffer size from 2048 to 4096
[image]

**3. disable imu publishing and publish odom only**

**4. publish constant odometry message(don't update it with new data)**
```c
nav_msgs__msg__Odometry msg_odometry={0};
void StartDefaultTask(void const * argument){
	......
	/***************************************/
	/*      Odometry publisher init        */
	/***************************************/
	nav_msgs__msg__Odometry__init(&msg_odometry);
	ret = rclc_publisher_init_default(&pub_odometry,&node,ROSIDL_GET_MSG_TYPE_SUPPORT(nav_msgs, msg, Odometry),"odom/unfiltered");
	if (ret == RCL_RET_OK)
	{
		printf("[INFO] odometry publisher init OK!\r\n");
  		msg_odometry.header.frame_id.data = "odom";
  		msg_odometry.header.frame_id.capacity = 4;
  		msg_odometry.header.frame_id.size = 4;
	}
	else
	{
		printf("[ERROR]odometry publisher init failed!\r\n");
		MicroROS_SysInit = 0;
	}
        ......
}
```
shows in code snippet below.

**5. increase value of RMW_UXRCE_MAX_HISTORY**
micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/colcon.meta

i increase value of DRMW_UXRCE_MAX_PUBLISHERS from default to 6.
i increase value of DRMW_UXRCE_MAX_HISTORY from default 4 to 6.
```
{
    "names": {
        ...,
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=6",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=6",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
[... 299 lines cut ...]
	{
		vTaskDelayUntil( &xLastWakeTime, xFrequency );
//		Odometry_Update(&msg_odometry, (float)xFrequency/1000.0f, Car.CurrentXVelocity, Car.CurrentYVelocity, Car.CurrentAngularVelocity);
		rcl_ret_t ret = rcl_publish(&pub_odometry, &msg_odometry, NULL);
		if(ret == RCL_RET_ERROR)
		{
			printf("[ERROR]publish odometry failed:publish error.\r\n");
		}
		else if(ret == RCL_RET_INVALID_ARGUMENT)
		{
			printf("[ERROR]publish odometry failed:invalid argument.\r\n");
		}
		else if(ret == RCL_RET_PUBLISHER_INVALID)
		{
			printf("[ERROR]publish odometry failed:publisher invalid.\r\n");
		}
	}
}
```

## phenomenon
UART debug message as follows. it prints publish odometry failed. it's in code snippet "PublishOdom()" FreeRTOS task function.
[image]
micro-ROS-Agent terminal shows that it is receiving IMU topic but none of odometry topic data.
[image]

pablogs9, 2023-10-13:
Do not rely on functions like `nav_msgs__msg__Odometry__init`, just initialize the memory yourself.

You can read some more details here: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html#message-memory

For example for your nav_msgs/Odom:

1. You need to init `header.frame_id` string
2. You need to init `child_frame_id` string
3. You need to take into account that strings are NULL terminated, so capacity shall be bigger than size
4. Also assigning a literal to a pointer (`msg_odometry.header.frame_id.data = "odom"`) may present problems on some platforms. It is recommended to assign heap/stack/static memory to that pointer and then copy the string.

Pansamic, 2023-10-13:
> Do not rely on functions like `nav_msgs__msg__Odometry__init`, just initialize the memory yourself.
>
> You can read some more details here: [...]

thank you for your reply. I tried to initialize odometry message myself without `nav_msgs__msg__Odometry__init`, and the error remained the same. I even created a new project to publish only odometry message and unfortunately failed. the following is my new project code (main part of it).

**main.c**
```c
#include "main.h"
#include "cmsis_os.h"
#include "usb_device.h"

#include <stdio.h>

#include "usbd_cdc_if.h"
#include "usbd_cdc.h"

/* micro ros kernel */
#include "rcl/rcl.h"
#include "rclc/executor.h"
#include "rclc/rclc.h"
#include "rmw_microros/rmw_microros.h"

/* micro ros messages */
#include "sensor_msgs/msg/imu.h"
#include "sensor_msgs/msg/time_reference.h"
#include "sensor_msgs/msg/joint_state.h"
#include "geometry_msgs/msg/twist.h"
#include "nav_msgs/msg/odometry.h"

UART_HandleTypeDef huart1;

osThreadId defaultTaskHandle;

/*****************************/
/* micro ros core components */
/*****************************/
rclc_support_t support={0};
rcl_allocator_t allocator={0};
rcl_allocator_t freeRTOS_allocator={0};
rclc_executor_t executor={0};

/*****************************/
/*  micro ros communication  */
/*****************************/
rcl_node_t node={0};
rcl_publisher_t pub_odometry={0};

/*****************************/
/*     micro ros messages    */
/*****************************/
nav_msgs__msg__Odometry msg_odometry={0};

void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_USART1_UART_Init(void);
void StartDefaultTask(void const * argument);

/*****************************/
/*    port comm interface    */
/*****************************/
bool cubemx_transport_open(struct uxrCustomTransport * transport);
bool cubemx_transport_close(struct uxrCustomTransport * transport);
size_t cubemx_transport_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err);
size_t cubemx_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);

/*****************************/
/*     memory management     */
/*****************************/
void * microros_allocate(size_t size, void * state);
[... 160 lines cut ...]
	const TickType_t xFrequency = 500;
	xLastWakeTime = xTaskGetTickCount();
	/* Infinite loop */
	for(;;)
	{
		vTaskDelayUntil( &xLastWakeTime, xFrequency );
//		Odometry_Update(&msg_odometry, (float)xFrequency/1000.0f, Car.CurrentXVelocity, Car.CurrentYVelocity, Car.CurrentAngularVelocity);
		rcl_ret_t ret = rcl_publish(&pub_odometry, &msg_odometry, NULL);
		if(ret == RCL_RET_ERROR)
		{
			printf("[ERROR]publish odometry failed:publish error.\r\n");
		}
		else if(ret == RCL_RET_INVALID_ARGUMENT)
		{
			printf("[ERROR]publish odometry failed:invalid argument.\r\n");
		}
		else if(ret == RCL_RET_PUBLISHER_INVALID)
		{
			printf("[ERROR]publish odometry failed:publisher invalid.\r\n");
		}
	}
  /* USER CODE END 5 */
}

```

pablogs9, 2023-10-13:
You need to take into account that strings are NULL terminated, so capacity shall be bigger than size

Pansamic, 2023-10-13:
yes I've tried to set capacity bigger than size, but many trials all failed. I even change the code like the following but failed to publish:
```c
static char odometry_frame_id[5] = {'o','d','o','m',0};
static char odometry_child_frame_id[10] = {'b','a','s','e','_','l','i','n','k',0};
msg_odometry.header.frame_id.data = odometry_frame_id;
msg_odometry.header.frame_id.capacity = 5;
msg_odometry.header.frame_id.size = 4;
msg_odometry.child_frame_id.data = odometry_child_frame_id;
msg_odometry.child_frame_id.capacity = 10;
msg_odometry.child_frame_id.size = 9;
```
btw, IMU message publish successfully with initial configuration like this:
```c
	/***************************************/
	/*        IMU publisher  init          */
	/***************************************/
	// create publisher
	ret = rclc_publisher_init_best_effort(&pub_imu,&node,ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, Imu),"imu/data");
	if (ret == RCL_RET_OK)
		printf("[INFO] imu publisher init OK!\r\n");
	else
	{
		printf("[ERROR]imu publisher init failed!\r\n");
		MicroROS_SysInit = 0;
	}
	msg_imu.header.frame_id.data = "base_link";
	msg_imu.header.frame_id.size = 9;
	msg_imu.header.frame_id.capacity = 9;
	for(uint8_t i=0 ; i<9 ; i++)
		msg_imu.angular_velocity_covariance[i]=-1;
	for(uint8_t i=0 ; i<9 ; i++)
		msg_imu.linear_acceleration_covariance[i]=-1;
	for(uint8_t i=0 ; i<9 ; i++)
		msg_imu.orientation_covariance[i]=-1;
```
so i think frame_id strings are not the culprit.

pablogs9, 2023-10-16:
Another issue that may be happening is that XRCE does not allow fragmentation in best effort and nav_msgs/Odometry is quite big.

- Does it work if you configure the publisher as reliable?
- Does it work if you rebuild the library using a bigger MTU by means of configuring [this parameter](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/97175304425c5bee87c6fddd99de1ef8d0c394dc/CMakeLists.txt#L65C5-L65C33) to a size that can fit the message size?

Pansamic, 2023-10-16:
I publish sensor_msgs/msg/Odometry **successfully** by increasing this parameter from 512 to 1024. Micro-ROS-Agent shows that odometry message is 732Bytes which is larger than default MTU 512 bytes.

pablogs9, 2023-10-16:
So, that was the problem. Thanks for reporting back @Pansamic.