micro-ROS/micro_ros_stm32cubemx_utils #130: Cannot publish several messages with reliable QoS.

Pansamic, 2023-11-17:
# Background & Configuration
* MCU: STM32F407ZGT6
* repo-distro: humble
* agent machine: Jetson Nano 4GB
* agent ROS2 distro: Foxy

# Detailed Description
I've realized multi-publisher with FreeRTOS after #125 issue solved. I use semaphore to protect each of micro-ros API.
But it publishes messages with best-effort QoS, which are kind of denied by many ROS2 packages such as robot_state_publisher and slam_toolbox.
So I change publisher init function from `rclc_publisher_init_best_effort()` to `rclc_publisher_init_default()`. Unfortunately, all of my publishers failed to publisher and return `RCL_RET_ERROR`.

# Thinking
I've tried to subscribe what Micro-ROS publishes as ROS2 documentation c++ minimal subscriber does, But it cannot correctly call callback function, so it must be not receiving any message. Then I tried to change QoS of my subscriber to best-effort and it can receive messages.

# Question
1. Why all of Micro-ROS publishers failed after changing QoS from best-effort to reliable?
2. Does Micro-ROS for Arduino or ESP32 has the same problem?
3. Is it FreeRTOS's fault?
4. Please tell me how to support multi-thread feature by setting Cmake option of Micro-XRCEDDS of Micro-ROS.

pablogs9, 2023-11-17:
Could you provide the code you are using?

Pansamic, 2023-11-17:
Two files: 1.microros.c ; 2.freertos.c.
freertos.c calls `microros_init()` in microros.c to initialize micro-ros, `microros_init()` init rclc kernel, publishers and subscribers.
First part
```c
/**
 * @file microros.c
 * @author pansamic (pansamic@foxmail.com)
 * @brief micro-ROS related functions
 * @version 0.1
 * @date 2023-11-15
 * 
 * @copyright Copyright (c) 2023
 * 
 */
/******************************************************/
/*                      INCLUDES                      */
/******************************************************/
#include <time.h>

#include <microros.h>

/* STM32 Peripherals */
#include "rtc.h"
#include "tim.h"

/* Board Peripheral drivers */
#include "motor.h"
#include "led.h"
#include "icm20602.h"
#include "mecanum.h"

/******************************************************/
/*                      DEFINES                       */
/******************************************************/

/******************************************************/
/*                       MACROS                       */
/******************************************************/
#define RCCHECK(fn) \
{\
    rcl_ret_t temp_rc = fn;\
    if (RCL_RET_OK != temp_rc) {\
        printf("Failed status on line %d: %d. Aborting.\n", __LINE__, (int)temp_rc);\
        return 1;\
    }\
}

#define RCSOFTCHECK(fn) \
{\
    rcl_ret_t temp_rc = fn;\
    if(RCL_RET_OK != temp_rc) {\
        printf("Failed status on line %d: %d. Continuing.\n", __LINE__, (int)temp_rc);\
    }\
}

/******************************************************/
/*                     VARIABLES                      */
/******************************************************/
/*****************************/
/* micro ros core components */
/*****************************/
rclc_support_t support={0};
rcl_allocator_t allocator={0};
rcl_allocator_t freeRTOS_allocator={0};
rclc_executor_t executor={0};
rclc_parameter_server_t param_server={0};

/* micro ros parameter server config */
const rclc_parameter_options_t param_server_options = 
{
[... 1052 lines cut ...]
    odom_msg->pose.pose.orientation.y = (double) q[2];
    odom_msg->pose.pose.orientation.z = (double) q[3];
    odom_msg->pose.pose.orientation.w = (double) q[0];

    odom_msg->pose.covariance[0] = 0.001;
    odom_msg->pose.covariance[7] = 0.001;
    odom_msg->pose.covariance[35] = 0.001;

    //linear speed from encoders
    odom_msg->twist.twist.linear.x = linear_vel_x;
    odom_msg->twist.twist.linear.y = linear_vel_y;
    odom_msg->twist.twist.linear.z = 0.0;

    //angular speed from encoders
    odom_msg->twist.twist.angular.x = 0.0;
    odom_msg->twist.twist.angular.y = 0.0;
    odom_msg->twist.twist.angular.z = angular_vel_z;

    odom_msg->twist.covariance[0] = 0.0001;
    odom_msg->twist.covariance[7] = 0.0001;
    odom_msg->twist.covariance[35] = 0.0001;
}

/* USER CODE END Application */
```

pablogs9, 2023-11-17:
Does it works if you use the micro-ROS Agent version according to your client distro? Instead of `agent ROS2 distro: Foxy` does it works if you use `agent ROS2 distro: Humble`?

Pansamic, 2023-11-17:
I connect MCU to my PC(Ubuntu 22.04 ROS-Humble) and still the same - All publishers fail.

pablogs9, 2023-11-17:
Could you debug inside this creation function in order to see which internal call is causing the error?

Pansamic, 2023-11-17:
But this repo create a `.a` library to link. I cannot see micro-ros API inside. 
publisher and subscriber all init succeed. Just publish fails.

pablogs9, 2023-11-17:
You are using osSemaphoreWait on MicroROSSemaphoreHandle with a finite value and not checking the return code. This may lead to continue the execution without the ownership of the semaphore and probably calling rcl_publish from multiple threads simultaneously. Try to use an infinite value or checking the return code of the wait operation.

Pansamic, 2023-11-17:
I set all to `osSemaphoreWait(MicroROSSemaphoreHandle,0)` to wait forever but still the same, all publishers fail.
Micro-ROS-Agent -v5 level output is like:
```bash
[1700228887.378333] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228887.378484] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228887.579297] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228887.579428] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228887.813349] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228887.813468] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228888.017290] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228888.017461] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228888.221211] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228888.221408] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228888.425279] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228888.425409] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228888.629239] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228888.629433] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228888.831246] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228888.831469] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228888.962705] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 32
[1700228888.962788] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 44
[1700228888.963047] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228888.963094] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228889.163264] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228889.163403] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228889.360307] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228889.360447] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
[1700228889.563251] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x71F88A29, len: 13
[1700228889.563390] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x71F88A29, len: 13
```
Idon't know what this 13bytes length data is, but it seems that agent is still communicating with micro-ros but publishers fail. If all publishers work well, there are all data with length 320(IMU) or 720(Odometry), etc.

pablogs9, 2023-11-17:
Is 0 wait forever or a no timeout operation?

Pansamic, 2023-11-17:
Thanks! I set all to `osSemaphoreWait(MicroROSSemaphoreHandle,0XFFFFFFFF)`. Publishers works well in default ( reliable ) QoS now.
Here is the `osSemaphoreWait()` description.
```c
/**
* @brief Wait until a Semaphore token becomes available
* @param  semaphore_id  semaphore object referenced with \ref osSemaphore.
* @param  millisec      timeout value or 0 in case of no time-out.
* @retval  number of available tokens, or -1 in case of incorrect parameters.
* @note   MUST REMAIN UNCHANGED: \b osSemaphoreWait shall be consistent in every CMSIS-RTOS.
*/
int32_t osSemaphoreWait (osSemaphoreId semaphore_id, uint32_t millisec)
```

closed 2023-11-17 by Pansamic