micro-ROS/micro_ros_stm32cubemx_utils #131: Cannot sync accurate time

Pansamic, 2023-11-17:
# Detailed Description

I use `rmw_uros_sync_session()` to sync time with Micro-ROS-Agent. But when I use `rmw_uros_epoch_millis()` and `rmw_uros_epoch_nanos()` to get time, I got seconds starting from 0, not 170xxxxxx (Linux timestamp seconds) as expected.

# Thinking

This result was no surprise because I cannot find any API that can change local time of MCU, while there is an API to get time called `clock_gettime()` in `micro_ros_stm32cubemx_utils/extra_sources/microros_time.c`. 

I want Micro-ROS to set local time that high accuracy hardware timer ( DWT, TIM or RTC ) holds, rather than FreeRTOS ticks.

it's said that `rmw_uros_sync_session()` uses NTP to sync time so it must has high accurate and less latency, but it cannot function properly, pity.

# Trials

My `clock_gettime()` is now modified to set RTC time. Micro-ROS subscribes `/time_reference` topic from Micro-ROS-Agent and transform timestamp to RTC Binary format. I use the following code to get current timestamp and it works well (superficially). Actually this method introduces huge latency.

```c
rcutils_time_point_value_t current_time;
rcutils_ret_t ret;
ret = rcutils_system_time_now(&current_time);
if(ret != RCUTILS_RET_OK)
    printf("[ERROR]read systime failed!\r\n");
msg->header.stamp.sec = RCUTILS_NS_TO_S(current_time);
msg->header.stamp.nanosec = (uint32_t)current_time;
```

# Question
1. How does Micro-ROS set time?
2. Can maintainers expose some API to set local time?

pablogs9, 2023-11-20:
> How does Micro-ROS set time?

`rmw_uros_sync_session` will not change your local time it just stores a time offset of the XRCE session that is used when `rmw_uros_epoch_nanos` and `rmw_uros_epoch_millis` are called.

> Can maintainers expose some API to set local time?

Out of micro-ROS scope.

bnbhat, 2024-10-01:
Hi @Pansamic,

Were you able to find a solution for the time synchronization issue? I'm facing a similar challenge, as I need to timestamp all the messages published by a micro-ROS node running on an MCU.

@pablogs9, could you please advise on the recommended approach for timestamping messages within a micro-ROS node?

Best Regards

bnbhat, 2024-10-02:
If anyone stumbles upon this issue looking for a way to timestamp messages in micro-ROS, here's a solution that worked for me:

1. Synchronize the Time with the Agent
```C
#include <rmw_microros/rmw_microros.h>

// Sync timeout
const int timeout_ms = 1000;

// Synchronize time with the agent
rmw_uros_sync_session(timeout_ms);
```

2. Get the Synchronized Time
```C
#include <rmw_microros/rmw_microros.h>

// Get corrected epoch time 
int64_t time_ns = rmw_uros_epoch_nanos();

// Convert to required format
msg.header.stamp.sec = (int32_t)(time_ns / 1000000000);
msg.header.stamp.nanosec = (uint32_t)(time_ns % 1000000000);
```

l4es, 2024-11-21:
Hi @bnbhat,

Does one need to call _rmw_uros_sync_session()_ everytime prior _rmw_uros_epoch_nanos()_ for filling the timestamp of the topic being published ?

In the following example, _rmw_uros_sync_session()_ is called once when the agent is available : https://github.com/adityakamath/akros2_firmware/blob/akros2_humble/akros2_firmware.ino#L331

And the time offset is calculated for being used later (filling the timestamp of published topics) : https://github.com/adityakamath/akros2_firmware/blob/akros2_humble/akros2_firmware.ino#L554

bnbhat, 2024-11-21:
Hi @l4es 

You don't need to call  `rmw_uros_sync_session()` every time as it stores the offset. But you can call it once in a while to sync time if you want to. 

When you call `rmw_uros_epoch_nanos()`, it adds the stored offset to the system time and gives you the synced time.

flux0-0, 2024-11-28:
hi @bnbhat , If I pio run my code same you about rmw_uros_sync_session(timeout_ms) and  int64_t time_ns = rmw_uros_epoch_nanos();, I have error "collect2: error: ld returned 1 exit status" . Have you get similar problem before?

bnbhat, 2024-11-28:
Hi @l4es , could you share your code? or more details on your setup?

have you synced the time before calling `rmw_uros_epoch_nanos()` ?

flux0-0, 2024-11-28:
This is my code " https://github.com/flux0-0/test_microros_time/blob/main/main.cpp " . I tried "pio run " in platform IO and I have " collect2: error: ld returned 1 exit status
*** [.pio/build/esp32doit-devkit-v1/firmware.elf] Error 1"