micro-ROS/micro_ros_stm32cubemx_utils #138: MicroRos subscriber callback stops getting called after receiving a couple messages

Nivant11, 2024-01-15:
I am using microRos on an STM32 device which is connected to a host agent over serial. I am attempting to do a simple function first. Basically, I have the STM32 subscribing to a "heartbeat" topic. The message type (.msg file) is simply "uint8 heartbeat" The host agent publishes a "1" on this topic.

The STM32 has a callback function which I have setup. Ideally, everytime it sees a "1", an LED will toggle.

What is currently happening though, is that the LED toggles a couple times (about 11 times) before it stops toggling, even though the host agent is still publishing. The details of my STM32 implementation are below:
`
void mros_init(void *argument) {
HAL_GPIO_TogglePin(GPIOC, LED_DEBUG_Pin);

if (rmw_uros_set_custom_transport(
true, (void*) &huart1, cubemx_transport_open, cubemx_transport_close,
        cubemx_transport_write, cubemx_transport_read) != RMW_RET_OK){
    NVIC_SystemReset();
}

//allocate freeRtos
freeRTOS_allocator = rcutils_get_zero_initialized_allocator();
freeRTOS_allocator.allocate = microros_allocate;
freeRTOS_allocator.deallocate = microros_deallocate;
freeRTOS_allocator.reallocate = microros_reallocate;
freeRTOS_allocator.zero_allocate = microros_zero_allocate;

if (!rcutils_set_default_allocator(&freeRTOS_allocator)) {
    printf("Error on default allocators (line %d)\n", __LINE__);
    NVIC_SystemReset();
}

// micro-ROS app

rcl_allocator_t allocator = rcl_get_default_allocator();

//create init_options
rcl_ret_t ret3 = rclc_support_init(&support, 0, NULL, &allocator);

if (ret3 != RCL_RET_OK) {
    NVIC_SystemReset();
}

// create node
rcl_ret_t ret0 = rclc_node_init_default(&node, "ccb", "", &support);`

Above I have shown the initialization function for micro ros. You can see that I am creating the subscriber and then attaching the callback function. The callback function is simple, shown below:

`void heartbeat_sub_callback(const void *msgin) {
HAL_GPIO_TogglePin(GPIOC, LED_DEBUG_Pin);}`

As you can see, I am just toggling an LED. The host agent publishes data on this topic every 5ms, so I expect that this callback should be invoked continuously. Instead, it gets invoked about 11 times and then stops. Does anyone have any clues as to what could be the problem?

pablogs9, 2024-01-15:
Could you share the whole code?

Nivant11, 2024-01-15:
`/*
 * cmr_micro_ros.c
 *
 *  Created on: Feb 25, 2023
 *      Author: ajsco
 */
#include "cmr_micro_ros.h"
#include "main.h"
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <uxr/client/transport.h>
#include <rmw_microxrcedds_c/config.h>
#include <rmw_microros/rmw_microros.h>
#include <micro_ros_utilities/type_utilities.h>
#include <std_msgs/msg/int32.h>

#include "cmr_msgs/msg/motor_write_batch.h"
#include "cmr_msgs/msg/sensor_read_batch.h"
#include "cmr_msgs/msg/power.h"
#include "cmr_msgs/msg/error.h"
#include "cmr_msgs/msg/heartbeat.h"
#include "boards.h"

//global variables

#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

extern UART_HandleTypeDef huart1;
extern TIM_HandleTypeDef htim6;
extern uint8_t MROS_REINIT_FLAG;

rcl_allocator_t freeRTOS_allocator;

// uROS definitions
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rclc_executor_t executor;

// error publisher definitions
rcl_publisher_t error_pub;
cmr_msgs__msg__Error error_pub_msg;

// sensor publisher definitions
rcl_publisher_t sensor_pub;
cmr_msgs__msg__SensorReadBatch sensor_pub_msg;
uint8_t sensor_ids[] = { BLDC_D1, BLDC_D2, BLDC_D3, BLDC_D4, BLDC_D5, BLDC_D6,
						BLDC_AR1, BLDC_AR2, BLDC_AR3, BLDC_AR4, BLDC_AR5, BLDC_AR6,
						SENSOR1, SENSOR2 };
size_t num_sensors = sizeof(sensor_ids) / sizeof(uint8_t);
//cmr_msgs__msg__SensorReadBatch sensor_msg_buf;  // buffer to transiently store enqueued sensor messages

// heartbeat subscriber definitions
rcl_subscription_t heartbeat_sub;
cmr_msgs__msg__Heartbeat heartbeat_sub_msg;

// motor subscriber definitions
rcl_subscription_t motor_sub;
cmr_msgs__msg__MotorWriteBatch motor_sub_msg;
//cmr_msgs__msg__MotorWriteBatch motor_msg_buf;  // buffer to transiently store enqueued motor messages

// power subscriber definitions
rcl_subscription_t power_sub;
cmr_msgs__msg__Power power_sub_msg;
//cmr_msgs__msg__Power power_msg_buf;  // buffer to transiently store enqueued power messages

static micro_ros_utilities_memory_conf_t conf = {0};

[... 346 lines cut ...]
}

int RCCHECK(rcl_ret_t fn) {
	rcl_ret_t temp_rc = fn;
	if((temp_rc != RCL_RET_OK)){

		printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);
		NVIC_SystemReset();
//		dealloc_msgs();
//		RCCHECK(rcl_subscription_fini(&motor_sub, &node));
//		RCCHECK(rcl_subscription_fini(&power_sub, &node));
//		RCCHECK(rcl_publisher_fini(&error_pub, &node));
//		RCCHECK(rcl_publisher_fini(&sensor_pub, &node));
//		RCCHECK(rcl_subscription_fini(&heartbeat_sub, &node));
//		RCCHECK(rcl_node_fini(&node));
//		RCCHECK(rclc_support_fini(&support));
//		mros_init(NULL);
		return -1;
	}
	return 1;
}

`

Here is all the code. I have commented out a lot of other publishers/subscribers because I was trying to isolate the issue to only the "heartbeat." In this case, the heartbeat is my own custom topic that I am just trying to get to work the way I expect before I move onto other things.