micro-ROS/micro_ros_stm32cubemx_utils #139: micro-ROS on STM32H743 established session with micro-ros-agent quickly ends

nikitax75, 2024-02-06:
Hello,

First of all, thanks for the very detailed tutorials and explanations. Thanks to all the information available online, I was able to sucessfully build a firmware on our stm32h743 board that runs and attemps a connection micro-ros agent (there was an issue with the udp transport example resulting in a deadlock on `LOCK_TCPIP_CORE()` but that's another story.

It feels like I am already there, but I am encountering a new problem when the board attemps to connect to the micro-ros-agent. It seems that the board is sending unexpected data which results in micro-ros-agent shutting fown the session.

Please see debug below:

[image]

`ros2 run micro_ros_agent micro_ros_agent udp4 --port 457879 -v6
[1707258505.140859] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 64663
[1707258505.140997] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1707258509.430965] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 2A 2E D9 75 81 00 FC 01
[1707258509.431336] info     | Root.cpp           | create_client            | create                 | client_key: 0x2A2ED975, session_id: 0x81
[1707258509.431443] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2A2ED975, address: 192.168.10.210:38908
[1707258509.431593] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2A2ED975, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1707258509.539450] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2A2ED975, len: 36, data: 
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00
[1707258509.539573] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x2A2ED975
[1707258509.539626] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x2A2ED975, address: 192.168.10.210:38908
[1707258509.539636] warning  | Root.cpp           | create_client            | invalid client key     | client_key: 0x00000000
[1707258509.539873] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x00000000, len: 23, data: 
0000: 00 00 00 00 00 00 00 00 04 01 0B 00 85 00 00 00 00 00 00 00 00 00 00
[1707258509.643521] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258509.742010] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258509.842342] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258509.943190] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258510.042678] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258510.143645] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258510.243912] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258510.343683] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: E8 03 00 00 00 CA 9A 3B C0 CB 17 07 00
[1707258510.444295] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
`

nikitax75, 2024-02-06:
Also, here is the callstack for this second 36-byte packet that I suspect to be the closing the session on micro-ros-agent side:

[image]

I've tried to find the code for `send_custom_msg` to better understand what's happenning but sadly cound'nt find it anywhere.

Thanks

pablogs9, 2024-02-07:
Hello @nikitax75 which is the main application you are using?

nikitax75, 2024-02-07:
Hello @pablogs9 , thanks a lot for your very prompt reply! 

Here is the application code :

```c
/* USER CODE BEGIN Header_StartDefaultTask */
/**
  * @brief  Function implementing the defaultTask thread.
  * @param  argument: Not used
  * @retval None
  */
/* USER CODE END Header_StartDefaultTask */
void StartDefaultTask(void *argument)
{
  /* init code for LWIP */
  MX_LWIP_Init();
  /* USER CODE BEGIN 5 */
  /* ETH_CODE: Adding lwiperf to measure TCP/IP performance.
   * iperf 2.0.6 (or older?) is required for the tests. Newer iperf2 versions
   * might work without data check, but they send different headers.
   * iperf3 is not compatible at all.
   * Adding lwiperf.c file to the project is necessary.
   * The default include path should already contain
   * 'lwip/apps/lwiperf.h'
   */

  printf("[i] Locking TCP/IP core\n");
  /* USER CODE BEGIN 5 */

  // micro-ROS configuration
  printf("[i] Setting custom transport\n");
  rmw_uros_set_custom_transport(
    false,              //Framing disable here. Udp should Use Packet-oriented mode.
    "192.168.10.121",    //your Agent's ip address
    cubemx_transport_open,
    cubemx_transport_close,
    cubemx_transport_write,
    cubemx_transport_read);

  // micro-ROS app
  rcl_publisher_t publisher;
  std_msgs__msg__Int32 msg;
  rclc_support_t support;
  rcl_allocator_t allocator;
  rcl_node_t node;

  // sett-up default allocator (heap allocated rather than FreeRTOS allocated)
  printf("[i] Setting default allocator\n");
  allocator = rcl_get_default_allocator();
  rcl_ret_t ret;

  // initialize micro-ROS (in a loop as it sometimes fail)
  printf("[i] RCLC support init... start\n");
   do{
        ret =  rclc_support_init(&support, 0, NULL, &allocator);
        osDelay(100);
        printf("[i] RCLC support init failed... retry\n");
    } while (ret != RCL_RET_OK);
  printf("[+] RCLC support done\n");

  // // create node
  rclc_node_init_default(&node, "stm32", "", &support);
  printf("[+] RCLC node init done\n");

  osDelay(100);

  // create publisher
  rclc_publisher_init_default(
    &publisher,
    &node,
[... 594 lines cut ...]
     ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

```

I am using the default allocators

Also, about the extra `LOCK_TCPIP_CORE` thing, it owe an apology as it doesn't seem to be from the main examples, it seems that I took this from somewhere else, will re-read all the examples I used to understand were that error came from this afternoon.

Finally, please note that I am getting an incompatbile point type compilation warning for the `cubemx_transport_write` function 

`Core/Src/main.c: In function 'StartDefaultTask':
Core/Src/main.c:413:5: warning: passing argument 5 of 'rmw_uros_set_custom_transport' from incompatible pointer type [-Wincompatible-pointer-types]
     cubemx_transport_write,
     ^~~~~~~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/nucleo_h743zi/micro_ros_platformio/libmicroros/include/rmw_microros/rmw_microros.h:35:0,
                 from .pio/libdeps/nucleo_h743zi/micro_ros_platformio/platform_code/micro_ros_platformio.h:13,
                 from Core/Src/main.c:27:
.pio/libdeps/nucleo_h743zi/micro_ros_platformio/libmicroros/include/rmw_microros/custom_transport.h:54:11: note: expected 'write_custom_func {aka unsigned int (*)(struct uxrCustomTransport *, const unsigned char *, unsigned int,  unsigned char *)}' but argument is of type 'size_t (*)(struct uxrCustomTransport *, uint8_t *, size_t,  uint8_t *) {aka unsigned int (*)(struct uxrCustomTransport *, unsigned char *, unsigned int,  unsigned char *)}'
 rmw_ret_t rmw_uros_set_custom_transport(
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

Thanks again

nikitax75, 2024-02-07:
The udp transport function are as per the standard ` udp_transport.c`:

```c

#include <uxr/client/transport.h>

#include <rmw_microxrcedds_c/config.h>

#include "main.h"
#include "cmsis_os.h"

#include <unistd.h>
#include <stdio.h>
#include <string.h>
#include <stdbool.h>

// --- LWIP ---
#include "lwip/opt.h"
#include "lwip/sys.h"
#include "lwip/api.h"
#include <lwip/sockets.h>

#ifdef RMW_UXRCE_TRANSPORT_CUSTOM

// --- micro-ROS Transports ---
#define UDP_PORT        9999
static int sock_fd = -1;

bool cubemx_transport_open(struct uxrCustomTransport * transport){
    sock_fd = socket(AF_INET, SOCK_DGRAM, 0);
    struct sockaddr_in addr;
    addr.sin_family = AF_INET;
    addr.sin_port = htons(UDP_PORT);
    addr.sin_addr.s_addr = htonl(INADDR_ANY);
    
    if (bind(sock_fd, (struct sockaddr *)&addr, sizeof(addr)) == -1)
    {
        return false;
    }

    return true;
}

bool cubemx_transport_close(struct uxrCustomTransport * transport){
    if (sock_fd != -1)
    {
        closesocket(sock_fd);
        sock_fd = -1;
    }
    return true;
}

size_t cubemx_transport_write(struct uxrCustomTransport* transport, uint8_t * buf, size_t len, uint8_t * err){
    if (sock_fd == -1)
    {
        return 0;
    }
    const char * ip_addr = (const char*) transport->args;
    struct sockaddr_in addr;
    addr.sin_family = AF_INET;
    addr.sin_port = htons(UDP_PORT);
    addr.sin_addr.s_addr = inet_addr(ip_addr);
    int ret = 0;
    ret = sendto(sock_fd, buf, len, 0, (struct sockaddr *)&addr, sizeof(addr));
    size_t writed = ret>0? ret:0;

    return writed;
}

size_t cubemx_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err){

    int ret = 0;
    //set timeout
    struct timeval tv_out;
    tv_out.tv_sec = timeout / 1000;
    tv_out.tv_usec = (timeout % 1000) * 1000;
    setsockopt(sock_fd, SOL_SOCKET, SO_RCVTIMEO,&tv_out, sizeof(tv_out));
    ret = recv(sock_fd, buf, len, MSG_WAITALL);
    size_t readed = ret > 0 ? ret : 0;
    return readed;
}

#endif

```

With the prototypes declared in a custom header:

```c

#ifndef MICROROS_EXTRA_H
#define MICROROS_EXTRA_H

#include <unistd.h>
#include "cmsis_os.h"
#include <uxr/client/transport.h>

#include <rmw_microxrcedds_c/config.h>

#include "main.h"
#include "cmsis_os.h"

#include <unistd.h>
#include <stdio.h>
#include <string.h>
#include <stdbool.h>

// --- LWIP ---
#include "lwip/opt.h"
#include "lwip/sys.h"
#include "lwip/api.h"
#include <lwip/sockets.h>

#include <stdlib.h>
#include <string.h>

#include "FreeRTOS.h"
#include "task.h"

bool cubemx_transport_open(struct uxrCustomTransport * transport);

bool cubemx_transport_close(struct uxrCustomTransport * transport);

size_t cubemx_transport_write(struct uxrCustomTransport* transport, uint8_t * buf, size_t len, uint8_t * err);

size_t cubemx_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);

#endif // MICROROS_EXTRA_H

```

pablogs9, 2024-02-07:
Try to use an executor with a timer and also follow an approach where you control the client-agent connection status. More info here: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/handle_reconnections/handle_reconnections.html#state-machine-approach

pablogs9, 2024-02-07:
Example here: https://github.com/micro-ROS/micro_ros_arduino/blob/iron/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino

pablogs9, 2024-02-08:
You can find more information about XRCE-DDS [here](https://micro-xrce-dds.docs.eprosima.com/en/latest/) and [here](https://www.omg.org/spec/DDS-XRCE/1.0/About-DDS-XRCE)

nguyenlam185, 2024-06-04:
Hello guys,
Have anybody made any progress on this issue? I got the same problem on Nucleo H723ZG. On the terminal of Nucleo, the `publish` function reports error code 300

```
void StartDefaultTask(void *argument)
{
    /* init code for LWIP */
    MX_LWIP_Init();
    /* USER CODE BEGIN 5 */

    // micro-ROS configuration
    rmw_uros_set_custom_transport(
                    false,  //Framing disable here. Udp should Use Packet-oriented mode.
                    "192.168.0.111",  //your Agent's ip address
                    cubemx_transport_open, cubemx_transport_close, cubemx_transport_write,
                    cubemx_transport_read);

    rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();
    freeRTOS_allocator.allocate = microros_allocate;
    freeRTOS_allocator.deallocate = microros_deallocate;
    freeRTOS_allocator.reallocate = microros_reallocate;
    freeRTOS_allocator.zero_allocate = microros_zero_allocate;

    if (!rcutils_set_default_allocator(&freeRTOS_allocator))
    {
        log_e("Error on default allocators");
    }

    // micro-ROS app
    rcl_publisher_t publisher;
    std_msgs__msg__Int32 msg;
    rclc_support_t support;
    rcl_allocator_t allocator;
    rcl_node_t node;

    allocator = rcl_get_default_allocator();

    //create init_options
    rclc_support_init(&support, 0, NULL, &allocator);

    // create node
    rclc_node_init_default(&node, "cubemx_node", "", &support);

    // create publisher
    rclc_publisher_init_default(&publisher, &node,
                                ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
                                "cubemx_publisher");

    msg.data = 0;

    for (;;)
    {
        rcl_ret_t ret = rcl_publish(&publisher, &msg, NULL);
        if (ret != RCL_RET_OK)
        {
            log_e("Error (code: %d) publishing", ret);
        }

        msg.data++;
        osDelay(1000);
    }
    /* USER CODE END 5 */
}

```

zijian-x, 2025-03-07:
got the same exact error right after just enabling the d-cache for a already functioning micro-ros firmware with ethernet transport via dma. i guess its an issue with cache coherency as my firmware was functioning stably until i tried to increase the performance and enabled the d-cache.

nikitax75, 2025-03-07:
Hi guys, 

He did solve the issue since then and are running micro-ros in production on our STM32H743. It was indeed a cache coherence issue, I'll ask one of our engineers to post the fix last week.

zijian-x, 2025-03-07:
appreciate the lightening speed of a response, and would be really great if you would post the fix as im combating with this issue rn, thank you in advance!

nikitax75, 2025-03-10:
Thanks, it took us a few good weeks to solve this issue entirely so we're happy if someone else can benefit from that work: I initially deactivated d-cache entirely (as you suggested) to get micro-ros to work, but someone from my team (author of this commit) later found a much more elegant solution, see the change below:

```commit cf6f04341ab2c538f4e4b705d7c2ce7b82a4f437
Author: Thierry Baldo <thierry.baldo@joinboxco.com>
Date:   Thu Mar 21 10:42:18 2024 +0100

    nucleo-h743zi2/basic_freertos: flush dcache line before transmitting eth frame
    
    DCache must be flushed (aka "cleaned") before transmitting ethernet frame,
    as DMA is used and it can access RAM only, not DCache
    
    Address must be aligned to 32 bytes

diff --git a/nucleo-h743zi2/basic_freertos/LWIP/Target/ethernetif.c b/nucleo-h743zi2/basic_freertos/LWIP/Target/ethernetif.c
index b1d40aa..89d002a 100644
--- a/nucleo-h743zi2/basic_freertos/LWIP/Target/ethernetif.c
+++ b/nucleo-h743zi2/basic_freertos/LWIP/Target/ethernetif.c
@@ -34,6 +34,11 @@
 /* Within 'USER CODE' section, code will be kept by default at each generation */
 /* USER CODE BEGIN 0 */
 
+/* address has to be aligned to 32 bytes */
+#define ALIGN_ADDR(addr)                  ((uint32_t*)((uint32_t)(addr) & ~0x1F))
+#define ALIGN_SIZE(addr, size)            ((size) + ((uint32_t)(addr) & 0x1f))
+#define FLUSH_CACHE_BY_ADDR(addr, size)   SCB_CleanDCache_by_Addr(ALIGN_ADDR(addr), ALIGN_SIZE(addr, size))
+
 /* USER CODE END 0 */
 
 /* Private define ------------------------------------------------------------*/
@@ -412,6 +417,8 @@ static err_t low_level_output(struct netif *netif, struct pbuf *p)
     Txbuffer[i].buffer = q->payload;
     Txbuffer[i].len = q->len;
 
+    FLUSH_CACHE_BY_ADDR(Txbuffer[i].buffer, Txbuffer[i].len);
+
     if(i>0)
     {
       Txbuffer[i-1].next = &Txbuffer[i];
```

zijian-x, 2025-03-10:
patched it and it also works FLAWLESSLY on the nucleo f7. you just saved me tons of work trying to figure this out myself for my thesis, THANK YOU!

pablogs9, 2025-03-10:
Hello @zijian-x @nikitax75, I've taken a look at this issue and researched a bit about DCACHE. 
As far as I see it, it is a data cache to speed up read/write access to RAM, right?

I see that the patch that solves it modifies LwIP code, so my concern now is: is this specifically related to micro-ROS or is there some sort of low-level bug in the LwIP bindings of ST code?

If I write an application that just writes UDP packets, is this bug also going to be a problem, or is there a specific reason to make it only appear under micro-ROS operation?

Thanks a lot for the solution guys, this one seems to be one of the hard ones 😉.

zijian-x, 2025-03-10:
yea imo its not micro-ROS related, but its ST who messed up in their lwip HAL and didn't consider the case when both d-cache and dma is enabled. it just so happened that @nikitax75 found out about this issue while using micro-ROS and then managed to solved it so that when i too encountered this last week, i could fortunately just ask for the fix, which tbh should also be sent to ST just to help them out... but what do i know