micro-ROS/micro_ros_stm32cubemx_utils #140: lib microros cannot be founded

kartman88, 2024-02-20:
Hi I'm working on Windows 11 with STM32CubeIde and I have a problem while building that is the following:
`C:/ST/STM32CubeIDE_1.14.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lmicroros: No such file or directory
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:69: micro_ros.elf] Error 1
"make -j20 all" terminated with exit code 2. Build might be incomplete.`
I think this is related to the step 4. I double checked the path to see if it was correct and it seems right.

totolerest, 2024-02-21:
you may follow this first then will got the lib microros folder
[image]

kartman88, 2024-02-22:
I did that and followed everything exactly as in the readme but still have this error only on windows

totolerest, 2024-02-22:
Could you share the screen capture of GCC Linker on no.4?

kartman88, 2024-02-22:
I have ../ because was working on Linux but I've tried with the absolute path too and got the same error.
[image]

totolerest, 2024-02-22:
Hmmm... Weird.
BTW, when I build the project I did not put this in prebuild step.
[image]

kartman88, 2024-02-22:
Same, I skipped to step 4. Can I ask you what you have put in the docker command. The absolute path like something/workspace/name_of_project ?

totolerest, 2024-02-22:
I just skip no.2 but I follow the rest.
Step 3 is add path that need to add
I also checkout the humble project.
[image]
For command I follow the instructions 
<ABSOLUTE_PATH_TO_PROJECT> this I put the folder that I checkout humble 
At the end of 2nd command is humble which match with the version that I checkout.
/micro_ros_static_library_builder**:humble**

I case you checkout iron you may try to change to iron and run command again.

kartman88, 2024-02-22:
I'm already working on humble. About the step number 3 did you put what was written in the guide:

`micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include`

Or did you put something different like:

`../micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros/include`

Or absolute path

totolerest, 2024-02-22:
No, I browse it manually to the destination folder which could be absolute path.

daehwankim112, 2025-03-12:
https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/140#issuecomment-1959064349
Thank you @totolerest. Menaullly setting destination folder to absolute path worked.

But I am getting new errors. It seems like this issue also appear in [this post](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/45). However, the post fix it by modifying makefile generated by CubeMX. I am using CubeIDE which does not generate Makefile.

\AppData\Local\Temp\cc8Jz80h.s:833: Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'
\AppData\Local\Temp\cc8Jz80h.s:855: Error: selected FPU does not support instruction -- `vldmiaeq r0!,{s16-s31}'
make: *** [Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F/subdir.mk:19: Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F/port.o] Error 1

totolerest, 2025-03-13:
Hi @daehwankim112  could you share how do you setup the system?
What is the board or microcontroller that you select for gen the code?

daehwankim112, 2025-03-14:
@totolerest I fixed the issue!

I fixed this issue by creating new project. I made it as C project not C++ project this time. I think the issue was I copied sample_main.c into main.c and compiled leading to compiler errors. However, this time, I am only copying

```
/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <uxr/client/transport.h>
#include <rmw_microxrcedds_c/config.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int32.h>
/* USER CODE END Includes */
```
and
```
/* USER CODE BEGIN 4 */
bool cubemx_transport_open(struct uxrCustomTransport * transport);
bool cubemx_transport_close(struct uxrCustomTransport * transport);
size_t cubemx_transport_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err);
size_t cubemx_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);

void * microros_allocate(size_t size, void * state);
void microros_deallocate(void * pointer, void * state);
void * microros_reallocate(void * pointer, size_t size, void * state);
void * microros_zero_allocate(size_t number_of_elements, size_t size_of_element, void * state);
/* USER CODE END 4 */
```
from the sample_code.c to main.c and it compiles fine.

This is my unchanged toolchain.c

```
SET(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_CROSSCOMPILING 1)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)

set(CMAKE_C_COMPILER $ENV{TOOLCHAIN_PREFIX}gcc)
set(CMAKE_CXX_COMPILER $ENV{TOOLCHAIN_PREFIX}g++)

SET(CMAKE_C_COMPILER_WORKS 1 CACHE INTERNAL "")
SET(CMAKE_CXX_COMPILER_WORKS 1 CACHE INTERNAL "")

set(FLAGS $ENV{RET_CFLAGS} CACHE STRING "" FORCE)
set(MICROROSFLAGS "-DCLOCK_MONOTONIC=0 -D'__attribute__(x)='" CACHE STRING "" FORCE)

set(CMAKE_C_FLAGS_INIT "-std=c11 ${FLAGS} ${MICROROSFLAGS} " CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_INIT "-std=c++14 ${FLAGS} -fno-rtti ${MICROROSFLAGS} " CACHE STRING "" FORCE)

set(__BIG_ENDIAN__ 0)
```
 
This is my C/C++ Build setting.

[image]

daehwankim112, 2025-03-14:
@totolerest 
Nevermind.. The error still exists.

Here is how I setup the project.

1. I am using NUCLEO-F446ZE. MCU is STM32F446ZETx
2. New STM32 project. Using C and executable default option.
3. System Core -> RCC -> High Speed Clock (HSE) -> Crystal/Ceramic Resonator
4. System Core -> SYS -> Timebase Source -> TIM1
5. Connectivity -> USART3 -> Configuration -> DMA Settings -> Add RX and TX
- RX: Circular, High priority
- TX: Normal, High priority
6. Connectivity -> USART3 -> Configuration -> NVIC Settings -> enable USART3 global interrupt
7. Middleware and Software Packs -> FREERTOS -> Mode -> Interface -> CMSIS_V2
8. Middleware and Software Packs -> FREERTOS -> Configuration -> Tasks and Queues -> Edit Task
- Stack Size (words) 3000
- Allocatoin: Static
9. Generate Code
10.  I left the Clock Configuration as default
11. Save and Generate Code
12. Git pull this repo at the project directory (humble)
> Project name
> \Includes
> \Core
> \Drivers
> \Middlewares
> \Debug
> \micro_ros_stm32cubemx_utils
13. Follow "[Using this package with STM32CubeIDE](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils?tab=readme-ov-file#using-this-package-with-stm32cubeide)" step 1-5. The project builds fine at this point. `Src` folder sould look like this at this point.
[image]
14. Add following code snippets from the sample_code.c to main.c.
```
/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <uxr/client/transport.h>
#include <rmw_microxrcedds_c/config.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int32.h>
/* USER CODE END Includes */
```
and
```
/* USER CODE BEGIN 4 */
bool cubemx_transport_open(struct uxrCustomTransport * transport);
bool cubemx_transport_close(struct uxrCustomTransport * transport);
size_t cubemx_transport_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err);
size_t cubemx_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);

void * microros_allocate(size_t size, void * state);
void microros_deallocate(void * pointer, void * state);
void * microros_reallocate(void * pointer, size_t size, void * state);
void * microros_zero_allocate(size_t number_of_elements, size_t size_of_element, void * state);
/* USER CODE END 4 */
```
My MCU/MPU setting and toolchain.c can be found above.
15. at `/* USER CODE BEGIN 5 */`, add
```
// micro-ROS configuration

  rmw_uros_set_custom_transport(
    true,
    (void *) &huart3,
    cubemx_transport_open,
    cubemx_transport_close,
    cubemx_transport_write,
    cubemx_transport_read);

  rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();
  freeRTOS_allocator.allocate = microros_allocate;
  freeRTOS_allocator.deallocate = microros_deallocate;
  freeRTOS_allocator.reallocate = microros_reallocate;
  freeRTOS_allocator.zero_allocate =  microros_zero_allocate;

  if (!rcutils_set_default_allocator(&freeRTOS_allocator)) {
      printf("Error on default allocators (line %d)\n", __LINE__); 
  }

  // micro-ROS app

  rcl_publisher_t publisher;
  std_msgs__msg__Int32 msg;
  rclc_support_t support;
  rcl_allocator_t allocator;
  rcl_node_t node;

  allocator = rcl_get_default_allocator();
```
The project builds fine until this point.
16. Add this code snippet right after previous step.
```
//create init_options
  rclc_support_init(&support, 0, NULL, &allocator);
```
17. Build
18. This compiler error shows up.
```
C:/ST/STM32CubeIDE_1.17.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.13.3.rel1.win32_1.0.0.202411081344/tools/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: error: SeaPuppyV1.3.elf uses VFP register arguments, ../micro_ros_stm32cubemx_utils/microros_static_library_ide/libmicroros\libmicroros.a(librclc-init.c.obj) does not
`atomic_64bits.c:(.text+0xa0): undefined reference to `__sync_synchronize'`
```
19. If I change Floating point ABI to `Software Implementation`, this error shows up
```
C:\Users\DAEHWA~1\AppData\Local\Temp\ccm48r0d.s:833: Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'
```

This is how I got here... PLEASE help!

daehwankim112, 2025-03-20:
I fixed it.

I followed the exact instruction again but in different Windows PC. I am sorry I wasn't able to locate what was causing the issue. It is probably WSL2 or Docker because when I run Docker on the not working PC, I receive this error.

```
dos2unix: /project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/library_generation.sh: No such file or directory
dos2unix: Skipping /project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/library_generation.sh, not a regular file.
```

But somehow it works well in my other PC. What am I missing here?