micro-ROS/micro_ros_stm32cubemx_utils #144: STM32 and ROS 2 Humble connection problem

omerdurmus61, 2024-05-21:
I am using the STM32F411 Discovery board and working with STM32CubeIDE. I followed the STM32CubeIDE configuration steps provided in this repository. However, when I run the command ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6, the output is as follows

```
[1716275194.803444] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x4127EE6D, address: 0
[1716275194.803583] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4127EE6D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1716275194.805337] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4127EE6D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 41 27 EE 6D 81 00 FC 01
[1716275194.805493] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x4127EE6D, address: 0
[1716275194.805735] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4127EE6D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1716275194.808237] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4127EE6D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 41 27 EE 6D 81 00 FC 01
[1716275194.808364] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x4127EE6D, address: 0
[1716275194.808506] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4127EE6D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1716275194.811310] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4127EE6D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 41 27 EE 6D 81 00 FC 01
[1716275194.811435] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x4127EE6D, address: 0
[1716275194.811570] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4127EE6D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1716275194.813338] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x4127EE6D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 41 27 EE 6D 81 00 FC 01
[1716275194.813518] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x4127EE6D, address: 0
[1716275194.813770] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x4127^C[ros2run]: Interrupt

```

MehmetZekiSildir, 2024-05-21:
the same as my problem.

pablogs9, 2024-05-21:
Your configuration is sending messages to DDS, check `[** <<DDS>> **]` logs. How are you checking the reception on the ROS 2 side?