micro-ROS/micro_ros_stm32cubemx_utils #148: Dual ucHeap memory regions.

Dmivaka, 2024-07-18:
Hi!

I'm making a project based on FreeRTOS in STM32CubeIDE.
Build Analyzer shows two memory regions named ucHeap in RAM. I believe these two objects come from the custom_memory_manager.c and FreeRTOS's heap4.c. Is it normal these two objects coexist? Why are their sizes defined in FreeRTOS config? Why do they have the same name?

Thanks!
[image]
[image]

DrMarkusKrug, 2025-01-15:
Hi,
no having two times ucHeap, that's not correct and also a little bit surprising that the whole thing is compiling and propably even running.
In the file file custom_memory_manager.c there should be a 'extern' statement declaring ucHeap. So it should look like 
extern unsigned char ucHeap[];
In the file heap_4.c the ucHeap[] should be declared.

Best Regards
Markus

Dmivaka, 2025-02-03:
Hi @DrMarkusKrug,
Seems like a correct way to declare heap is to delete "static" attribute of ucHeap in custom_memory_manager.c:
`uint8_t ucHeap[ configTOTAL_HEAP_SIZE ];`
and then define 
`configAPPLICATION_ALLOCATED_HEAP=1`
in project settings. This way FreeRTOS's heap_4.c is kept unmodified and can be re-generated by CubeMX without deleting changes.
Thanks a lot!