micro-ROS/micro_ros_stm32cubemx_utils #152: STM32H753 microros (humble) publisher does not publish anything

ugurxun, 2024-08-16:
Hello,
I am trying to work with microros on STM32H753ZIT6U .

System: WSL Ubuntu22.04
Ros Distro: Humble
Microcontroller: STM32H753ZIT6U
Transport: USART with DMA
USART baudrate: 115200

**What I have done**
---------------------------------------------------------------------------------------------------------------------------------
I followed the instrunctions on microros tutorials and I created microros workspace, cloned micro_ros_setup, updated dependencies, installed python3-pip, builded, sourced. Then, I created firmware, builded it and crated microros agent. 

I created a stm32cubeide project and arranged peripherals as HSE to crystal/seramic, SYS to TIM1. I enabled the USART3 (PD8, PD9) which is directly connected STLINKs microusb. I adjusted usart3_rx and usart3_tx for DMA with very high priority. I selected circular buffer for RX, I enabled the global interrupt for that. Finally, I  enabled the FreeRTOS and I gave 12kb storage for stack of default task.

I cloned the micro_ros_stm32cubemx_utils and moved the files in extra_sources to /Core/Src . I deleted the files except dma_transport.c in microros_transports. Then, I added prebuid steps, micro-ROS include directory and Add the micro-ROS precompiled library  as described. I sourced ros2 humble, micro ros workspace and gave the permissions via "sudo chmod 666 /var/run/docker.sock ". I found a basic projects ,compiled it and flashed succesfully. I runned the microros agent with "ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0". I also tried the run it with "ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888". However the publisher does not publish anything.
[image]

I tried to see the publisher with "ros2 topic list" but it is not seen.

**Code**
--------------------------------------------------------------------------------------------------------------------------------------------------
/* USER CODE BEGIN Header */
/**
  ******************************************************************************
  * @file           : main.c
  * @brief          : Main program body
  ******************************************************************************
  * @attention
  *
  * Copyright (c) 2024 STMicroelectronics.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  */
/* USER CODE END Header */
/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "cmsis_os.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <uxr/client/transport.h>
#include <rmw_microxrcedds_c/config.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int32.h>
/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN PTD */

/* USER CODE END PTD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */

/* USER CODE END PD */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN PM */

[... 412 lines cut ...]

#ifdef  USE_FULL_ASSERT
/**
  * @brief  Reports the name of the source file and the source line number
  *         where the assert_param error has occurred.
  * @param  file: pointer to the source file name
  * @param  line: assert_param error line source number
  * @retval None
  */
void assert_failed(uint8_t *file, uint32_t line)
{
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
     ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

--------------------------------------------------------------------------------------------------------------------------------------------------
**What I expect to see**
--------------------------------------------------------------------------------------------------------------------------------------------------
[image]
Then, I need to see my topic in ros2 topic list. The publisher must publish increasing counter data.

Do you have any idea to solve my problem? Thank you all.

handsome7729, 2024-10-16:
have you solved the problem?

ugurxun, 2024-10-16:
No, unfortunately i couldn't.

MinchangSung0223, 2024-11-27:
Press the reset button on your board

yuliang2, 2025-03-21:
I am using STM32H723 and facing the same issue...