micro-ROS/micro_ros_stm32cubemx_utils #155: included `control_msgs` package still on `galactic-devel`

zijian-x, 2024-11-09:
In the [extra_packages.json](microros_static_library/library_generation/extra_packages/extra_packages.repos) file under `library_generation`:

Does it make sense to bump the `control_msgs` version from `galatic-devel` to `master` for the jazzy branch which I'm currently using? Because their master branch serves all three distros(iron/jazzy/rolling) at the moment and I'd like to use the newer msg types from that repo which don't exist in the older `galactic-devel`.

I'll send a pull request if this makes sense.

pablogs9, 2024-11-11:
Yes, this makes sense. Thanks for pointing this out. Would you mind changing it in both folders?

zijian-x, 2024-11-11:
Thank you and yes of course Ill send a merge request updating the version in both `microros_static_library` and `microros_static_library_ide`.