micro-ROS/micro_ros_stm32cubemx_utils #161: Can't use MicroRos Custom Messages on STM32H723ZG

antoniocaso, 2024-12-02:
Hi everyone,

after months of excruciating pain and panic attacks and many kilos gained by stress eating, I managed to have microros running on my H723ZG nucleo: this however is true only with standard messages. Countless hours have now been wasted on trying to make custom messages work, to no avail. I already have the custom messages definitions ready, can even create a ros2 pub node to send them and I can even see them through a quick ro2 topic echo, but I can't for the life of me get this onto my stm32 board and I have lost all hope completely.

Do you maybe happen to know some insights, or maybe a procedure I could follow? Really, anything would help.

I just want to clarify that, since I have to use the .ioc, my only way is using the Stm32CubeIDE to create a project, and then find some sort of satanic ritual to bring custom messages to the project.

pablogs9, 2024-12-02:
Hello @antoniocaso, sorry for the big inconvenience micro-ROS is causing, usually is preferable to not deal with embedded libraries nor cmake/python mixtures of build system... xd

Now seriously, I assume that you have tried the following: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils?tab=readme-ov-file#adding-custom-packages

Normally this [part of the build script](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/1a1a8a9b8c3ba793e773a9947f25d2d36b6523af/microros_static_library_ide/library_generation/library_generation.sh#L46-L57) is in charge of copying the ROS 2 message packages that you have in the `library_generation/extra_packages` folder. Once they are copied along with the rest of the micro-ROS packages they are built together. It seems that you are building correctly the builtin set of packages, so this shall be a matter of checking if this part is copying your custom packages to the correct folder.

As far as this procedure normally run inside a docker, make sure that volumes are being shared correctly:

> docker pull microros/micro_ros_static_library_builder:jazzy && docker run --rm **-v ${workspace_loc:/${ProjName}}:/project** --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:jazzy

Let me know if you find something.

antoniocaso, 2024-12-02:
Hi Pablo, and thank you so much for your help! Sorry for not replying earlier!
I’ll try to be as clear as possible, and forgive me if I lack expertise on the subject.

Yes, I confirm that I tried taking the package with the messages and copying it into the "extra_packages" folder located inside the CubeIDE project folder. I’m using this project creation method because I not only need the IOC file but I’m also working on the Nucleo H723ZG as my platform. Here’s the tree starting from the ros2_ws/src/ folder, which contains all the messages. 

antonio@antonio:~/prova/ros2_ws/src/custom_microros_msgs$ tree
.
├── CMakeLists.txt
├── include
│   └── custom_microros_msgs
├── msg
│   ├── StmToUbuntuMessage.msg
│   ├── UbuntuToStm32Arbiter.msg
│   └── UbuntuToStm32Info.msg
├── package.xml
└── src

4 directories, 5 files

I copy this in the extra_packages folder, then try to run, and then nothing happens at all, like nothing was ever moved there. Here, this is the path 

/home/antonio/STM32CubeIDE/workspace_1.16.1/h723_microros_pt6/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/extra_packages

├── custom_microros_msgs
│   ├── CMakeLists.txt
│   ├── include
│   │   └── custom_microros_msgs
│   ├── msg
│   │   ├── StmToUbuntuMessage.msg
│   │   ├── UbuntuToStm32Arbiter.msg
│   │   └── UbuntuToStm32Info.msg
│   ├── package.xml
│   └── src
└── extra_packages.repos

And I also copy the message package in the other extra_packages under the microros_static_library (not _ide) folder just to be sure, but alas nothing. 

Unfortunately, I am forced to do this blend because I will need to interface an SRR520DI sensor with this board, use its FDCAN port to retrieve data from it, process the data, and then send custom messages to the onboard computer of an autonomous vehicle using micro-ROS. On the PC, there will, of course, be a standard ROS 2 setup, which the board will interface with thanks to the essential agent.

So, as I was saying, I confirm that I tried pasting it into that folder and building it using the command you provided, modified for Humble:

docker pull microros/micro_ros_static_library_builder:humble && docker run --rm -v ${workspace_loc:/${ProjName}}:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:humble VERBOSE=1

Actually, I’ve been dedicating myself entirely to trying to get all this to work for several days now, trying every possible method, starting with simply taking the headers and .a files (and unfortunately, even with .a, I’ve encountered endless problems since the simple colcon build only gives me .so files. To get .a files, I had to struggle quite a bit, and I ended up with FastRTPS files, which I’m not even sure are suitable. As expected, I ran into even more issues there).

By doing this I actually even managed to get things to build, but only when no "elements" of the library where used in the code, but if I dared to use the library it would tell me the type support .c was missing, even though the headers were all being seen correctly.

Unfortunately, all my efforts have been unsuccessful, and I’m now out of tears to shed. This is just a small part of my thesis project, which will require months and months of more struggles like this… and this one alone seems never-ending.
If you need anything—any clarification, anything at all—I’m entirely at your disposal and immensely grateful for any time you dedicate to helping me.

pablogs9, 2024-12-03:
Could you please: 

1. Remove the `micro_ros_stm32cubemx_utils` folder from your repo, to start from a clean start point.
2. Ensure there is no `libmicroros.a` or micro-ROS `include` folder anywhere in the project.
3. Clone the repo in the project as stated [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/tree/jazzy?tab=readme-ov-file#using-this-package-with-stm32cubemx)
4. Copy your custom message package to the required folder
5. Continue [the instructions](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/tree/jazzy?tab=readme-ov-file#using-this-package-with-stm32cubemx)

And paste here the output of the console so I can look at what is happening.

antoniocaso, 2024-12-03:
Hello @pablogs9 ! Thank you for your VERY PRECIOUS help!! Sorry if I took some time but I wanted to make everything right and clean and so I created a new project and a new custom message folder and all. Well, let's get to it. 
This is tree command done to the project's folder in order to check if nothing regarding microros was present at all. As this was just after creating the project as one normally does when using Stm32CubeIde. 

antonio@antonio:~/STM32CubeIDE/workspace_1.16.1/h723_microros_pt7$ tree
.
├── Core
│   ├── Inc
│   │   ├── eth.h
│   │   ├── fdcan.h
│   │   ├── FreeRTOSConfig.h
│   │   ├── gpio.h
│   │   ├── main.h
│   │   ├── memorymap.h
│   │   ├── stm32h7xx_hal_conf.h
│   │   ├── stm32h7xx_it.h
│   │   ├── usart.h
│   │   └── usb_otg.h
│   ├── Src
│   │   ├── eth.c
│   │   ├── fdcan.c
│   │   ├── freertos.c
│   │   ├── gpio.c
│   │   ├── main.c
│   │   ├── memorymap.c
│   │   ├── stm32h7xx_hal_msp.c
│   │   ├── stm32h7xx_hal_timebase_tim.c
│   │   ├── stm32h7xx_it.c
│   │   ├── syscalls.c
│   │   ├── sysmem.c
│   │   ├── system_stm32h7xx.c
│   │   ├── usart.c
│   │   └── usb_otg.c
│   └── Startup
│       └── startup_stm32h723zgtx.s
├── Drivers
│   ├── CMSIS
│   │   ├── Device
│   │   │   └── ST
│   │   │       └── STM32H7xx
│   │   │           ├── Include
│   │   │           │   ├── stm32h723xx.h
│   │   │           │   ├── stm32h7xx.h
│   │   │           │   └── system_stm32h7xx.h
│   │   │           ├── LICENSE.txt
│   │   │           └── Source
│   │   │               └── Templates
│   │   ├── Include
│   │   │   ├── cmsis_armcc.h
│   │   │   ├── cmsis_armclang.h
│   │   │   ├── cmsis_armclang_ltm.h
│   │   │   ├── cmsis_compiler.h
│   │   │   ├── cmsis_gcc.h
│   │   │   ├── cmsis_iccarm.h
│   │   │   ├── cmsis_version.h
│   │   │   ├── core_armv81mml.h
│   │   │   ├── core_armv8mbl.h
│   │   │   ├── core_armv8mml.h
│   │   │   ├── core_cm0.h
│   │   │   ├── core_cm0plus.h
│   │   │   ├── core_cm1.h
│   │   │   ├── core_cm23.h
│   │   │   ├── core_cm33.h
│   │   │   ├── core_cm35p.h
│   │   │   ├── core_cm3.h
│   │   │   ├── core_cm4.h
│   │   │   ├── core_cm7.h
│   │   │   ├── core_sc000.h
│   │   │   ├── core_sc300.h
│   │   │   ├── mpu_armv7.h
│   │   │   ├── mpu_armv8.h
[... 204 lines cut ...]
5 directories, 3 files

And, after doing so, this is the tree from inside the microros_static_library folder:
antonio@antonio:~/STM32CubeIDE/workspace_1.16.1/h723_microros_pt7/micro_ros_stm32cubemx_utils/microros_static_library$ tree
.
└── library_generation
    ├── colcon.meta
    ├── extra_packages
    │   ├── custom_msg
    │   │   ├── CMakeLists.txt
    │   │   ├── include
    │   │   │   └── custom_msg
    │   │   ├── msg
    │   │   │   └── MyCustomMessage.msg
    │   │   ├── package.xml
    │   │   └── src
    │   └── extra_packages.repos
    ├── library_generation.sh
    └── toolchain.cmake

7 directories, 7 files

Having already set up the build settings, and having already pasted to the freertos.c the contents of the simple_main.c in order to get microros actually running, I started the "build project" command, and attached to this reply is the build_output plain text file which is what the console on Stm32CubeIDE gave me when trying to build.

Also I want to clarify that trying to do #include "custom_msg/..." results in build error as the .h can't be seen or found, and it's especially mind-breaking the fact that the build didn't even give an error, it just... ignored completely the package I copied, didn't even give me the honour of refusing, just ignored me.

pablogs9, 2024-12-03:
> Having already set up the build settings, and having already pasted to the freertos.c the contents of the simple_main.c in order to get microros actually running, I started the "build project" [...]

Nothing attached

antoniocaso, 2024-12-03:
> Nothing attached

Sorry @pablogs9 , what was dumb of me. I edited the reply, it should be at the top of my message now.

pablogs9, 2024-12-03:
It seems that you folder is not being copied, could you add some debug echo commands [here](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/1a1a8a9b8c3ba793e773a9947f25d2d36b6523af/microros_static_library/library_generation/library_generation.sh#L28-L37) and rerun the whole procedure to check what is happening? Something like:

```bash
# Import user-defined packages
echo "Starting to import user-defined packages..."
mkdir extra_packages
echo "Created 'extra_packages' directory."

pushd extra_packages > /dev/null
echo "Switched to 'extra_packages' directory."

USER_CUSTOM_PACKAGES_DIR=$BASE_PATH/../../microros_component/extra_packages
echo "USER_CUSTOM_PACKAGES_DIR set to: $USER_CUSTOM_PACKAGES_DIR"

if [ -d "$USER_CUSTOM_PACKAGES_DIR" ]; then
    echo "Copying contents of $USER_CUSTOM_PACKAGES_DIR to current directory..."
    cp -R $USER_CUSTOM_PACKAGES_DIR/* .
else
    echo "Directory $USER_CUSTOM_PACKAGES_DIR does not exist. Skipping copy."
fi

if [ -f $USER_CUSTOM_PACKAGES_DIR/extra_packages.repos ]; then
    echo "Found extra_packages.repos in $USER_CUSTOM_PACKAGES_DIR. Importing..."
    vcs import --input $USER_CUSTOM_PACKAGES_DIR/extra_packages.repos
else
    echo "No extra_packages.repos found in $USER_CUSTOM_PACKAGES_DIR. Skipping vcs import."
fi

echo "Copying library generation packages..."
cp -R $BASE_PATH/library_generation/extra_packages/* .

if [ -f extra_packages.repos ]; then
    echo "Found extra_packages.repos in the current directory. Importing..."
    vcs import --input extra_packages.repos
else
    echo "No extra_packages.repos found in the current directory. Skipping vcs import."
fi

popd > /dev/null
echo "Returned to the previous directory."
echo "Completed user-defined packages import."
```

antoniocaso, 2024-12-05:
Ok @pablogs9 , I'm back with some... interesting news on all fronts: sorry for not replying these past few days, but a lot of interesting things have popped up.

First of all, before I tried implementing what you suggested, I tried to create an online repo ad put my custom messages package there, you know, just for shits and giggles cause I had seen that also "control_msgs" was being included this way, and so I gave it a try with absolutely zero hope it would work... well fuck, it did. 

At that point then I tried to create another project from scratch, same Ide same workspace same everything and same procedure, I did this like the same day so everything in my mind was absolutely the same. This time, instead of using the extra_packages.repo and the online ad-hoc created repo, I just copied and pasted the packages like I had tried (and failed) the first time, always sticking to both your advices and the official procedure on the repo.

This time, with no online repo, worked as well! At that point I thought "maybe it was the fact I did not enable the REENTRANT option for the freertos middleware in the ioc", and so I created yet ANOTHER project, from scratch, and did the exact same completely identical setup and procedure as the first time where things where not working the build process was ignoring the extra_packages: this time it ALSO worked.

I mean, maybe you won't believe all of this, but I swear to God this is completely nuts and I have no idea why, and its part of the reason why I haven't replied in the past few days, I've been trying to uncover this mistery, alas with no success. 

Still though, I added the code you told me to add to the library_generation.sh, so I will still provide the build output as attached. Since I didn't change anything at all, both the CMakeLists.txt and package.xml of the custom message package are still the same, and the folder's paths and all are still all the same. I will also provide the main.c and freertos.c files though, maybe something there has been getting things to work.

[build_output.txt](https://github.com/user-attachments/files/18023635/build_output.txt)
[freertos.c.txt](https://github.com/user-attachments/files/18023661/freertos.c.txt)
[main.c.txt](https://github.com/user-attachments/files/18023662/main.c.txt)

On a footnote, many other warfronts have sadly opened up, all regarding microROS I'm afraid (like the code working absolutely fine in flash mode, but debug mode completely makes the board invisible to the microROS agent on my pc): do I open another thread or do I just stick with this one?

pablogs9, 2024-12-05:
Nice to know that you managed to solve it.

Please open new issues for new problems you have with the library.

I'm closing this as solved.

closed 2024-12-05 by pablogs9

daehwankim112, 2025-09-03:
Hello! After spending about 2 months of searching and stressful nights I have came across this corner of Github. I am trying to build a custom interface using microROS and STM32F446ZETx on NUCLEO-F446ZE. This is officially microROS supported STM32 dev board. I am new to CUBEIDE.

I have custom interface built using ROS2 Humble. I have imported the folder into extra_packages. The CubeIDE project and the custom interface is very simple so far.

This is what I have at extra_packages under microros_static_library_ide
```
└───extra_packages
    └───uveec_custom_interfaces
        │   CMakeLists.txt
        │   LICENSE
        │   package.xml
        │
        └───msg
                RaspberrySensorsInterface.msg
                StmSensorsInterface.msg
```

This is my CMakeLists.txt
```
cmake_minimum_required(VERSION 3.8)
project(uveec_custom_interfaces)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rosidl_default_generators REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)

find_package(rosidl_default_generators REQUIRED)
rosidl_generate_interfaces(${PROJECT_NAME}
  "msg/RaspberrySensorsInterface.msg"
  "msg/StmSensorsInterface.msg"
)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # comment the line when a copyright and license is added to all source files
  set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # comment the line when this package is in a git repo and when
  # a copyright and license is added to all source files
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

This is my package.xml
```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>uveec_custom_interfaces</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="uvicenvironment@gmail.com">student</maintainer>
  <license>Apache-2.0</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
  <buildtool_depend>rosidl_default_generators</buildtool_depend>
  <exec_depend>rosidl_default_runtime</exec_depend>
  <member_of_group>rosidl_interface_packages</member_of_group>
</package>
```

When I build the project, it builds fine. However, when I try to import the created custom interfaces, the project does not acknowledge it exists. This is how I am trying to import following what I see from @antoniocaso 's code.
```
#include <uveec_custom_interfaces/msg/raspberry_sensors_interface.h>
#include <uveec_custom_interfaces/msg/stm_sensors_interface.h>
```
Even Camel case does not work as well. Furthermore, I cannot find these generated definition files in CubeIDE. What am I doing wrong? I am following this [instruction](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils?tab=readme-ov-file#adding-custom-packages) to import the files but I find it very abstractive. Is there any more documentation how should I approach this?

Thank you so much!

daehwankim112, 2025-09-19:
Hello I found out what was the reason. I don't know why it was documented but thinking about it now, it was quite obvious.

First of all, I am using CubeIDE on Windows. Please follow the Windows installation according to the documentation. If there is an issue with rcl not getting recognized, check this issue thread. https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/140 

Once you have working project with microROS, place your custom interface package at `micro_ros_stm32cubemx_utils\microros_static_library_ide\library_generation\extra_packages`, **rebuild (regenerate) your microROS library using the docker command**.

The docker command can be found here.
https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/140

Good luck!